PX4/px4_ros_com #17: List of issues found while following the PX4  ROS2 installation instructions

mlautman, 2019-02-25:
These issue are not limited to just the PX4_ros_com repo:

- The [install script](https://dev.px4.io/en/setup/dev_env_linux_ubuntu.html) for installing eProsima/FastRTPS installs the wrong version of FastRTPS for ROS Bouncy. (1.5.0 instead of 1.6.0) which causes `fastrtpsgen` to complain that it does not recognize the command line arguement `-I` when you build the ros2 branch of px4_ros_com.

- If you have a version of openjdk newer than 1.8, the setup will fail because of ambiguous libraries called `Module` ([link](https://github.com/eProsima/Fast-RTPS/issues/206)). You must uninstall all versions of java newer than 8 and then install `sudo apt install openjdk-8-jdk openjdk-8-jre gradle`. (If you have a version of java newer than 8 installed, `gradle` will use that instead of the correct version) 

- Using the [install script](https://raw.githubusercontent.com/PX4/Devguide/master/build_scripts/ubuntu_sim_ros_gazebo.sh) is broken in a few ways. For one it installs ROS kinetic rather than melodic which is assumed [later in the documentation](https://dev.px4.io/en/middleware/micrortps.html#installing-ros-and-ros2-and-respective-dependencies). Also, the script silently makes undesirable changes to the user's `.bashrc`. This can be considered a bug as running `rossource="source /opt/ros/kinetic/setup.bash"` at the start of every terminal session causes issues when you are trying to create a pure ROS2 project.

- The command `colcon build --symlink-install --packages-skip ros1_bridge --event-handlers console_direct+` in the "Building the workspaces" section fails because `python3` fails to find `genmsg`. This is discussed [here](https://github.com/ros2/ros1_bridge/issues/168#issuecomment-465753704) but it is never resolved. I have been running `source /opt/ros/melodic/setup.bash` before `source /opt/ros/bouncy/setup.bash` to get colcon to build the ros2 workspace.

- The command `colcon build --symlink-install --packages-skip ros1_bridge --event-handlers console_direct+` in the "Building the workspaces" section will also fail unless you run `export ROS2_DISTRO="bouncy"` 

- The command `colcon build --symlink-install --packages-skip ros1_bridge --event-handlers console_direct+` in the "Building the workspaces" section will also fail with `generate_microRTPS_bridge.py: error: argument -f/--fastrtpsgen-dir: expected one argument` unless you manually set FASTRTPSGEN_DIR to the location of `fastrtpsgen`. (eg. `export FASTRTPSGEN_DIR=/usr/local/bin`)

There are still more issues getting the example to run. I still haven't figured out why uOrb messages aren't making it through to the `sensor_combined_listener`. I hope this issue can act as a TODO list for fixing the documentation.

mlautman, 2019-02-25:
Also, the instructions still reference ROS2 Ardent in https://dev.px4.io/en/middleware/micrortps.html#testing-the-px4-fastrpts-bridge-with-ros2-and-ros

TSC21, 2019-02-26:
@mlautman thanks for testing. Keep in mind that this all suffered a major update on the package structure which has not been updated on the documentation yet.

TSC21, 2019-02-26:
As soon as the documentation gets updated I will let you know. For now, keep im mind that the major change is regarding how the package was split into a msg part and a rtps bridge part. Also the scripts have suffered a major update. I am closing this for now as most of what you describe here can be handled on the coming docs.

mlautman, 2019-02-26:
I really don't think it makes sense to close out an issue before the fixes have been applied. At very least this should be left as an open issue so that new users are aware of the regressions in the documentation. I spent days finding fixes to these issues and I am still finding unresolved bugs. Let's keep this open until the issues have been resolved

mlautman, 2019-02-26:
Another issue: 
- Installing FastRTPS from source can cause your ROS installation to fail to link correctly. This will cause even `ros2 run demo_nodes_cpp talker` to fail with a linker error. I have found that I can resolve this issue on my machine if I remove the system install of FastRTPS with the exclusion of the `fastrtpsgen` executable and `.jar` files and re-installing `ros-$ROS2_DISTRO-desktop`

TSC21, 2019-02-26:
> I really don't think it makes sense to close out an issue before the fixes have been applied. At very least this should be left as an open issue so that new users are aware of the regressions in the [...]

It's ok and noted. Again, thank you very much for your time testing this. But again, this is something that is really work in progress and a lot of things that you are reporting are being addressed on the back stage already.

TSC21, 2019-02-26:
Also, keep in mind this is an effort that currently only me is moving forward with and there are a lot of edge cases that I wasn't able to solve yet.

@mlautman I do appreciate your contributions, but I rather prefer you open separate issues for each of the problems than you actually condensate all of them in the same place. I did close this issue since most of the reported issues were already being addressed on the documentation, but please bring your findings to separate issues so they can be tackled separately as unique tasks.

Again, thanks for your contributions!

TSC21, 2019-02-26:
Regarding https://github.com/PX4/px4_ros_com/issues/17#issuecomment-467642549: this can actually be avoided, in the sense that we only required the `fastrtpsgen` to generate the source files we need for the `micrortps_bridge`. Again, noted and it's another (among many others) thing to address.

mlautman, 2019-02-26:
Thanks for reopening this. I am glad these issues are being addressed. I completely understand the frustrations of maintaining a package by oneself. If you would like my feedback on any of the changes you are making feel free to tag me in your PR's. 

I will separate out these issues into individual issues so that you are able to mark them off as they are resolved :)

closed 2019-02-26 by mlautman

TSC21, 2019-02-26:
> Thanks for reopening this. I am glad these issues are being addressed. I completely understand the frustrations of maintaining a package by oneself. If you would like my feedback on any of the changes you are making feel free to tag me in your PR's.

Thanks for understanding. And many thanks for offering your help! This will help me get things moving and have also someone with experience and interest getting this tested and give feedback. I will for sure have you tagged :)

> I will separate out these issues into individual issues so that you are able to mark them off as they are resolved :)

Awesome thanks!

davetcoleman, 2019-02-27:
+1 to collaboration!