PX4/px4_ros_com #18: Issues with

mlautman, 2019-02-26:
This is copied from #17 

Another issue: 
- Installing FastRTPS from source can cause your ROS installation to fail to link correctly. This will cause even `ros2 run demo_nodes_cpp talker` to fail with a linker error. I have found that I can resolve this issue on my machine if I remove the system install of FastRTPS with the exclusion of the `fastrtpsgen` executable and `.jar` files and re-installing `ros-$ROS2_DISTRO-desktop`

Example error:
`ros2 run demo_nodes_cpp talker`
```
failed to initialized rcl init options: failed to load shared library of rmw implementation. Exception: Cannot load library: /opt/ros/crystal/lib/librmw_fastrtps_cpp.so: undefined symbol: _ZTVN8eprosima8fastrtps27DataRepresentationQosPolicyE,
```

TSC21, 2019-02-26:
I will see if this can be solved by avoiding to actually install Fast-RTPS in the first place.

TSC21, 2020-02-11:
This is not an issue that's reproducible. I have both Fast-RTPS installed from source and from ROS binaries and use it without any issues. Closing.

closed 2020-02-11 by TSC21