PX4/px4_ros_com #19: Installing the correct version of FastRTPS for ROS2

mlautman, 2019-02-26:
* The [install script](https://dev.px4.io/en/setup/dev_env_linux_ubuntu.html) for installing eProsima/FastRTPS installs the wrong version of FastRTPS for ROS Bouncy. (1.5.0 instead of 1.6.0) which causes `fastrtpsgen` to complain that it does not recognize the command line arguement `-I` when you build the ros2 branch of px4_ros_com.

For ROS2 Bouncy you should install 1.6.0 and for ROS2 Crystal you should install 1.7.0.

TSC21, 2019-02-26:
Being addressed on scripts now, but should be simplified so to only get the `fastrtpsgen` script.

tfoote, 2019-03-02:
@TSC21 @LorenzMeier I've been working on a new approach for this with @clalancette that leverages the ROS 2 typesupport packages and is working toward being more generic such that the agent does not have to hard code the topics and types.  It works with the binary installed version of fastrtps.

Could you take a look at: https://github.com/osrf/ros2_serial_example  We're currently backwards compatible with your implementation but expect to add more features and robustness are planning to evolve the protocol. For example: https://github.com/osrf/ros2_serial_example/issues/18 and add dynamic topic support: https://github.com/osrf/ros2_serial_example/issues/13 instead of just having the yaml file defining the topics as we have now.

TSC21, 2019-03-02:
@tfoote really interesting! Could it be something that could back the agent implementation on this package? Meaning a package that is added as a dependency of this package that then subscribes to the uorb data being published after serialized by CDR?

tfoote, 2019-03-04:
Yes, at the moment it should be a drop in replacement for the agent in this package.

TSC21, 2019-03-04:
Alright. I will take a look at it. Thank you

TSC21, 2020-02-11:
This is already addressed on the containers and documentation.

closed 2020-02-11 by TSC21