PX4/px4_ros_com #20: Using the correct version of Java with fastrtpsgen

mlautman, 2019-02-26:
If you have a version of openjdk newer than 1.8, the setup will fail because of ambiguous libraries called `Module` ([link](https://github.com/eProsima/Fast-RTPS/issues/206)). You must uninstall all versions of java newer than 8 and then install `sudo apt install openjdk-8-jdk openjdk-8-jre gradle`. (If you have a version of java newer than 8 installed, `gradle` will use that instead of the correct version)

TSC21, 2019-02-26:
Well, this will probably become a no-problem from the moment we drop the Fast-RTPS source build. Stay tuned.

mlautman, 2019-02-26:
fastrtpsgen uses java so unless you have the correct version installed, I don't know how you would get around this

TSC21, 2019-02-26:
> fastrtpsgen uses java so unless you have the correct version installed, I don't know how you would get around this

I wonder if a pre-compiled version would do

mlautman, 2019-02-27:
I think that would be a much better solution

closed 2020-03-13 by TSC21