PX4/px4_ros_com #22: Issues building the ros1_bridge from source

mlautman, 2019-02-26:
Following the installation instructions in the [documentation](https://dev.px4.io/en/middleware/micrortps.html#building-the-workspaces), the command `colcon build --symlink-install --packages-skip ros1_bridge --event-handlers console_direct+` in the "Building the workspaces" section fails because `python3` fails to find `genmsg`. This is discussed [here](https://github.com/ros2/ros1_bridge/issues/168#issuecomment-465753704) but it is never resolved. I have been running `source /opt/ros/$ROS1_DISTRO/setup.bash` before `source /opt/ros/$ROS2_DISTRO/setup.bash` to get colcon to build the ros2 workspace.

TSC21, 2019-02-26:
The current solution found for now is the one you suggest above (source the ROS1 env before the ROS2 env).

@mlautman btw, what is the current ROS2 release you are using?

mlautman, 2019-02-26:
I have tested with both bouncy and crystal

mlautman, 2019-02-26:
I think you will still need to set `ROS2_DISTRO` otherwise, https://github.com/PX4/px4_ros_com/blob/a8e74d344af320ba859588565f58e985775da4ea/CMakeLists.txt#L17 will fail

TSC21, 2019-02-26:
So you were able to build the `ros1_bridge` under Crystal? I am actually facing some annoying issue when building it (you can follow everything in https://github.com/ros2/ros1_bridge/issues/168.

TSC21, 2019-02-26:
> I think you will still need to set `ROS2_DISTRO` otherwise, [this line](https://github.com/PX4/px4_ros_com/blob/a8e74d344af320ba859588565f58e985775da4ea/CMakeLists.txt#L17) will fail

Yep, that's something I will rearrange so it's not required (this was assuming an automation script before).

mlautman, 2019-02-26:
I actually haven't been building ros1_bridge in Crystal since my goal is to build a pure ROS2 system. These notes were from when I was following the documentation line by line.

TSC21, 2019-02-26:
> I actually haven't been building ros1_bridge in Crystal since my goal is to build a pure ROS2 system. These notes were from when I was following the documentation line by line.

Alright fair point. Thanks

mlautman, 2019-02-26:
Actually, I lied. I totally figured out how to build it :stuck_out_tongue_closed_eyes:  

In a fresh terminal, I run these commands:
```bash         
        export FASTRTPSGEN_DIR=/usr/local/bin
        export ROS1_DISTRO="melodic"
        export ROS2_DISTRO="crystal"
        export ROS1_PATH="/opt/ros/$ROS1_DISTRO/setup.bash"
        export ROS2_PATH="/opt/ros/$ROS2_DISTRO/local_setup.bash"
        export ROS1_WS=$HOME/ws_ros1/
        export ROS2_WS=$HOME/ws_ros2/

        cd $ROS2_WS
        colcon build --symlink-install --packages-select ros1_bridge --cmake-force-configure --event-handlers console_direct+
```

Not all of the above is necessary but it works so I haven't been messing with it :man_shrugging:

TSC21, 2019-02-26:
@mlautman thanks! The only difference I actually see from what I was trying is the `local_setup.bash` instead of the `setup.bash`. Let's see where it gets me.

mlautman, 2019-02-26:
I'm 99% sure that was the fix. I'm currently stuck trying to get the topics to show up when I run     `micrortps_client start -t UDP` and `micrortps_agent -t UDP`.

TSC21, 2019-02-26:
> I'm 99% sure that was the fix. I'm currently stuck trying to get the topics to show up when I run `micrortps_client start -t UDP` and `micrortps_agent -t UDP`.

What issues are you actually facing at this point?

mlautman, 2019-02-26:
When I run `sensor_combined_listener`, all I get is: `Starting sensor_combined listener node...` which makes sense since none of the topics are present

TSC21, 2019-02-26:
> which makes sense since none of the topics are present

What does this mean exactly? Do you get "matches" when you start the micrortps client and agent?

mlautman, 2019-02-26:
When I run `ros2 topic list`, I don't see any of the PX4 topics, just `/parameter_events` and `/rosout`

TSC21, 2019-02-26:
> When I run `ros2 topic list`, I don't see any of the PX4 topics, just `/parameter_events` and `/rosout`

Well that shouldn't be the first thing you have to debug. If you don't have a connection on the micrortps bridge, you won't have data on the ROS2 side.

mlautman, 2019-02-27:
For sure. Let me know if you run into the same micrortps bridge issue

TSC21, 2019-02-27:
> For sure. Let me know if you run into the same micrortps bridge issue

How are you sure this is an micrortps bridge issue?

Last time I tried this on my system (like a week ago) this was working perfectly fine.

TSC21, 2019-02-27:
What version of PX4 are you using?

TSC21, 2019-02-27:
Better: Can you please bring this to the #messaging channel of PX4 Slack? This is already out of scope ot this issue. Thanks

mlautman, 2019-02-27:
> Better: Can you please bring this to the #messaging channel of PX4 Slack? This is already out of scope ot this issue. Thanks

For sure! I was just thinking the same thing

TSC21, 2019-02-27:
> Actually, I lied. Here is how I build it :stuck_out_tongue_closed_eyes:  
>
>In a fresh terminal, I run these commands:
>```bash         
>        export FASTRTPSGEN_DIR=/usr/local/bin
>        export ROS1_DISTRO="melodic"
>        export ROS2_DISTRO="crystal"
>        export ROS1_PATH="/opt/ros/$ROS1_DISTRO/setup.bash"
>        export ROS2_PATH="/opt/ros/$ROS2_DISTRO/local_setup.bash"
>        export ROS1_WS=$HOME/ws_ros1/
>        export ROS2_WS=$HOME/ws_ros2/
>
>        cd $ROS2_WS
>        colcon build --symlink-install --packages-select ros1_bridge --cmake-force-configure --event-handlers console_direct+
>```
>
>Not all of the above is necessary but it works so I haven't been messing with it :man_shrugging:

@mlautman evaluating the list above, I have to remove what I said yesterday. I don't really get how it is working. Firstly you are exporting the paths for the environments, but you are not sourcing them in any way. Also, you are not sourcing the workspace env, so that the `px4_msgs` become available for `ros1_bridge` to build. So you are basically building the `ros1_bridge`, but without any of the workspaces sourced.

mlautman, 2019-02-28:
My bad. I missed some lines:

Open 3 Terminals (`Terminal A`, `Terminal B`, and `Terminal C`) and set these environment variables in both:

```bash
        export FASTRTPSGEN_DIR=/usr/local/bin
        export ROS1_DISTRO="melodic"
        export ROS2_DISTRO="crystal"
        export ROS1_PATH="/opt/ros/$ROS1_DISTRO/setup.bash"
        export ROS2_PATH="/opt/ros/$ROS2_DISTRO/local_setup.bash"
        export ROS1_WS=$HOME/ws_ros1/
        export ROS2_WS=$HOME/ws_ros2/
```

Next, `Terminal A` build the ROS2 repos except `ros1_bridge`

```bash
        source $ROS1_PATH && \
        source $ROS2_PATH && \
        cd $ROS2_WS && \
        colcon build --symlink-install --packages-skip ros1_bridge --event-handlers console_direct+
```

In `Terminal B` build the ROS1 repos with colcon

```bash
        source $ROS1_PATH && \
        cd $ROS1_WS && \
        colcon build --cmake-args --symlink-install --event-handlers console_direct+
```

In `Terminal C`

```bash
        source $ROS1_WS/install/setup.bash && \
        source $ROS2_PATH && \
        source $ROS2_WS/install/local_setup.bash

        cd $ROS2_WS
        colcon build --symlink-install --packages-select ros1_bridge --cmake-force-configure --event-handlers console_direct+
```

mlautman, 2019-02-28:
Note that you shouldn't have any environments being sourced in your bashrc.

TSC21, 2019-02-28:
> Note that you shouldn't have any environments being sourced in your bashrc.

Yep I know.

From what I can see in the Terminal C, you are not sourcing the ROS1 workspace. So you are not linking the ROS1 packages of px4_ros_com and px4_msgs, and they are required.

TSC21, 2019-02-28:
> In `Terminal C`
> 
> ```shell
>         source $ROS1_WS/install/setup.bash && \
>         source $ROS2_PATH && \
>         source $ROS2_WS/install/local_setup.bash
> 
>         cd $ROS2_WS
>         colcon build --symlink-install --packages-select ros1_bridge --cmake-force-configure --event-handlers console_direct+
> ```

This still does not work to me....

```sh
CMake Error at CMakeLists.txt:70 (message):     
  Failed to find ROS 2 package 'diagnostic_msgs'

-- Configuring incomplete, errors occurred!
See also "/home/nuno/PX4/px4_ros_com_ros2/build/ros1_bridge/CMakeFiles/CMakeOutput.log".
--- stderr: ros1_bridge
CMake Error at CMakeLists.txt:70 (message):
  Failed to find ROS 2 package 'diagnostic_msgs'

---
Failed   <<< ros1_bridge	[ Exited with code 1 ]

Summary: 0 packages finished [4.96s]
  1 package failed: ros1_bridge
  1 package had stderr output: ros1_bridge
```

TSC21, 2019-03-20:
@mlautman I was finally able to build `ros1_bridge` from source so this is no longer an issue.

closed 2020-02-11 by TSC21