PX4/px4_ros_com #23: Build errror

slmat27, 2019-03-11:
I am having a problem with building px4_ros_com package, I am using Ubuntu 18.04, with ROS Melodic and ROS2 Crystal, followed these setup steps:

   1. Clone https://github.com/PX4/px4_ros_com `master` branch to a ROS2 workspace;
   2. Clone https://github.com/PX4/px4_msgs `master` branch to the same ROS2 workspace;
   3. Clone https://github.com/PX4/px4_ros_com `ros1` to a ROS1 workspace;
   4. Clone https://github.com/PX4/px4_msgs `ros1` branch to the same ROS1 workspace;
   5. Installed Fast RTPS using the steps over here https://dev.px4.io/en/setup/fast-rtps-installation.html

Steps to reproduce issue:

   * Terminal A in `<ros1_ws_dir>`:
     * `source /opt/ros/melodic/setup.bash`
     * `colcon build`

   * Terminal B in `<ros2_ws_dir>`:
     * `source /opt/ros/melodic/setup.bash`
     * `source /opt/ros/crystal/setup.bash`
     * `export ROS2_DISTRO=$ROS_DISTRO`
     * `colcon build`
       * `px4_msgs` build is completed successfully
       * `px4_ros_com` build fails and trigger the below error
         * `generate_microRTPS_bridge.py: error: argument -f/--fastrtpsgen-dir: expected one argument`

Thanks once again!

TSC21, 2019-03-11:
Where do you have `fastrtpsgen` on your system?

slmat27, 2019-03-11:
In `/usr/local/bin/fastrtpsgen`

christianrauch, 2019-03-19:
You may need to set `FASTRTPSGEN_DIR` via `export FASTRTPSGEN_DIR=/usr/local/bin/` before calling colcon. See also: https://dev.px4.io/en/middleware/micrortps.html#agent-not-builtfastrtpsgen-is-not-found.

TSC21, 2019-03-20:
@slmat27, @christianrauch gave you the correct answer.

closed 2019-03-20 by TSC21