PX4/px4_ros_com #24: Missing 'genmsg' on ROS2-only workspace

christianrauch, 2019-03-19:
I am trying to build px4_ros_com in a pure ROS2 workspace and getting the error:
```
[  1%] [34m[1mGenerating micro-RTPS agent code...[0m
python import error:  No module named 'genmsg'

Required python packages not installed.

On a Debian/Ubuntu system please run:

  sudo apt-get install python-empy
  sudo pip install catkin_pkg

On MacOS please run:
  sudo pip install empy catkin_pkg

On Windows please run:
  easy_install empy catkin_pkg

CMakeFiles/micrortps_agent.dir/build.make:188: recipe for target '/home/christian/Downloads/px4_ws/src/px4_ros_com/src/micrortps_agent/microRTPS_agent.cpp' failed
CMakeFiles/Makefile2:419: recipe for target 'CMakeFiles/micrortps_agent.dir/all' failed
Makefile:140: recipe for target 'all' failed
```

I am not sourcing ROS1 and I am not trying to build the `ros1_bridge`. I do not see why `genmsg` (from ROS1) is required here and how do I satisfy this requirement with ROS2 and without ROS1.

TSC21, 2019-03-20:
@christianrauch at this stage I am trying to make it independent of `genmsg` of ROS1 and use the `python-genmsg` package instead. I will let you know when I have updates.

christianrauch, 2019-03-20:
Why is `genmsg` actually required for this pure ROS2 workspace? I assumed that `genmsg` is only used for ROS1 message definitions.

TSC21, 2019-03-20:
> Why is `genmsg` actually required for this pure ROS2 workspace? I assumed that `genmsg` is only used for ROS1 message definitions.

This has nothing to do with ROS2. `genmsg` is used by the `micrortps_agent` code generator.

TSC21, 2019-05-16:
@christianrauch solved here: https://github.com/PX4/px4_ros_com/commit/d447a5acb21aa8c03b5f28ba0fc17c822fe7ca16

TSC21, 2019-05-16:
A note that `python3-genmsg` and `python3-gencpp` are only available for deb download in 18.10 and above so one needs to download the deb and install it.

closed 2019-05-16 by TSC21