PX4/px4_ros_com #25: px4_ros_com build failed

virtualclone, 2019-04-03:
Hi, I follow the steps in [https://github.com/ros2/ros1_bridge/issues/168](url).
When `colcon build` in ROS2 workspace, it shows the error like this

`
ncrl@ncrl-D830MT:~/px4_ros_com_ros2$ colcon build --symlink-install
Starting >>> px4_msgs
[0.166s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/opt/ros/kinetic' in the environment variable CMAKE_PREFIX_PATH seems to be a catkin workspace but it doesn't contain any 'local_setup.*' files. Maybe the catkin version is not up-to-date?
[Processing: px4_msgs]                             
Finished <<< px4_msgs [40.9s]                         
Starting >>> px4_ros_com
[Processing: px4_ros_com]                                 
--- stderr: px4_ros_com                                    
openjdk version "1.8.0_191"
OpenJDK Runtime Environment (build 1.8.0_191-8u191-b12-2ubuntu0.16.04.1-b12)
OpenJDK 64-Bit Server VM (build 25.191-b12, mixed mode)
openjdk version "1.8.0_191"
OpenJDK Runtime Environment (build 1.8.0_191-8u191-b12-2ubuntu0.16.04.1-b12)
OpenJDK 64-Bit Server VM (build 25.191-b12, mixed mode)
openjdk version "1.8.0_191"
OpenJDK Runtime Environment (build 1.8.0_191-8u191-b12-2ubuntu0.16.04.1-b12)
OpenJDK 64-Bit Server VM (build 25.191-b12, mixed mode)
openjdk version "1.8.0_191"
OpenJDK Runtime Environment (build 1.8.0_191-8u191-b12-2ubuntu0.16.04.1-b12)
OpenJDK 64-Bit Server VM (build 25.191-b12, mixed mode)
openjdk version "1.8.0_191"
OpenJDK Runtime Environment (build 1.8.0_191-8u191-b12-2ubuntu0.16.04.1-b12)
OpenJDK 64-Bit Server VM (build 25.191-b12, mixed mode)
openjdk version "1.8.0_191"
OpenJDK Runtime Environment (build 1.8.0_191-8u191-b12-2ubuntu0.16.04.1-b12)
OpenJDK 64-Bit Server VM (build 25.191-b12, mixed mode)
openjdk version "1.8.0_191"
OpenJDK Runtime Environment (build 1.8.0_191-8u191-b12-2ubuntu0.16.04.1-b12)
OpenJDK 64-Bit Server VM (build 25.191-b12, mixed mode)
openjdk version "1.8.0_191"
OpenJDK Runtime Environment (build 1.8.0_191-8u191-b12-2ubuntu0.16.04.1-b12)
OpenJDK 64-Bit Server VM (build 25.191-b12, mixed mode)
openjdk version "1.8.0_191"
OpenJDK Runtime Environment (build 1.8.0_191-8u191-b12-2ubuntu0.16.04.1-b12)
OpenJDK 64-Bit Server VM (build 25.191-b12, mixed mode)
openjdk version "1.8.0_191"
OpenJDK Runtime Environment (build 1.8.0_191-8u191-b12-2ubuntu0.16.04.1-b12)
OpenJDK 64-Bit Server VM (build 25.191-b12, mixed mode)
openjdk version "1.8.0_191"
OpenJDK Runtime Environment (build 1.8.0_191-8u191-b12-2ubuntu0.16.04.1-b12)
OpenJDK 64-Bit Server VM (build 25.191-b12, mixed mode)
openjdk version "1.8.0_191"
OpenJDK Runtime Environment (build 1.8.0_191-8u191-b12-2ubuntu0.16.04.1-b12)
OpenJDK 64-Bit Server VM (build 25.191-b12, mixed mode)
openjdk version "1.8.0_191"
OpenJDK Runtime Environment (build 1.8.0_191-8u191-b12-2ubuntu0.16.04.1-b12)
OpenJDK 64-Bit Server VM (build 25.191-b12, mixed mode)
openjdk version "1.8.0_191"
OpenJDK Runtime Environment (build 1.8.0_191-8u191-b12-2ubuntu0.16.04.1-b12)
OpenJDK 64-Bit Server VM (build 25.191-b12, mixed mode)
openjdk version "1.8.0_191"
OpenJDK Runtime Environment (build 1.8.0_191-8u191-b12-2ubuntu0.16.04.1-b12)
OpenJDK 64-Bit Server VM (build 25.191-b12, mixed mode)
openjdk version "1.8.0_191"
OpenJDK Runtime Environment (build 1.8.0_191-8u191-b12-2ubuntu0.16.04.1-b12)
OpenJDK 64-Bit Server VM (build 25.191-b12, mixed mode)
openjdk version "1.8.0_191"
OpenJDK Runtime Environment (build 1.8.0_191-8u191-b12-2ubuntu0.16.04.1-b12)
OpenJDK 64-Bit Server VM (build 25.191-b12, mixed mode)
openjdk version "1.8.0_191"
OpenJDK Runtime Environment (build 1.8.0_191-8u191-b12-2ubuntu0.16.04.1-b12)
OpenJDK 64-Bit Server VM (build 25.191-b12, mixed mode)
openjdk version "1.8.0_191"
OpenJDK Runtime Environment (build 1.8.0_191-8u191-b12-2ubuntu0.16.04.1-b12)
OpenJDK 64-Bit Server VM (build 25.191-b12, mixed mode)
openjdk version "1.8.0_191"
[... 410 lines cut ...]
                 from /home/ncrl/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/RtpsTopics.h:45,
                 from /home/ncrl/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/RtpsTopics.cpp:33:
/home/ncrl/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/IridiumsbdStatus_PubSubTypes.h:32:2: error: #error Generated IridiumsbdStatus_ is not compatible with current installed Fast-RTPS. Please, regenerate it with fastrtpsgen.
 #error Generated IridiumsbdStatus_ is not compatible with current installed Fast-RTPS. Please, regenerate it with fastrtpsgen.
  ^
In file included from /home/ncrl/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/AdcReport_Publisher.h:47:0,
                 from /home/ncrl/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/RtpsTopics.h:46,
                 from /home/ncrl/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/RtpsTopics.cpp:33:
/home/ncrl/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/AdcReport_PubSubTypes.h:32:2: error: #error Generated AdcReport_ is not compatible with current installed Fast-RTPS. Please, regenerate it with fastrtpsgen.
 #error Generated AdcReport_ is not compatible with current installed Fast-RTPS. Please, regenerate it with fastrtpsgen.
  ^
In file included from /home/ncrl/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/BatteryStatus_Publisher.h:47:0,
                 from /home/ncrl/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/RtpsTopics.h:47,
                 from /home/ncrl/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/RtpsTopics.cpp:33:
/home/ncrl/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/BatteryStatus_PubSubTypes.h:32:2: error: #error Generated BatteryStatus_ is not compatible with current installed Fast-RTPS. Please, regenerate it with fastrtpsgen.
 #error Generated BatteryStatus_ is not compatible with current installed Fast-RTPS. Please, regenerate it with fastrtpsgen.
  ^
In file included from /home/ncrl/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/RadioStatus_Publisher.h:47:0,
                 from /home/ncrl/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/RtpsTopics.h:48,
                 from /home/ncrl/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/RtpsTopics.cpp:33:
/home/ncrl/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/RadioStatus_PubSubTypes.h:32:2: error: #error Generated RadioStatus_ is not compatible with current installed Fast-RTPS. Please, regenerate it with fastrtpsgen.
 #error Generated RadioStatus_ is not compatible with current installed Fast-RTPS. Please, regenerate it with fastrtpsgen.
`

My fastrtps version is 1.7.1.

TSC21, 2019-04-07:
@virtualclone please follow the official guide on how to setup `px4_ros_com`: https://dev.px4.io/en/middleware/micrortps.html. If you have problems, then I can reopen the issue.

closed 2019-04-07 by TSC21

zhongzhw, 2019-08-21:
I have met the same problem, how do you solved it

zhongzhw, 2019-08-21:
@virtualclone