PX4/px4_ros_com #27: Issues when building and running an example

ArkadiuszNiemiec, 2019-04-04:
I am trying to run `px4_ros_com` on `Ubuntu 18.04` but have some problems. Can you tell me what's wrong? This is what I did:
1. Installed the PX4 Firmware by using [ubuntu_sim_nuttx.sh](https://raw.githubusercontent.com/PX4/Devguide/master/build_scripts/ubuntu_sim_nuttx.sh)
2. Added `export FASTRTPSGEN_DIR=/home/aniemiec/eProsima_FastRTPS-1.7.1-Linux/bin` to `~/.bashrc` and restarted the session
3. Built `ROS2 Crystal` by following [Building ROS 2 on Linux](https://index.ros.org/doc/ros2/Installation/Linux-Development-Setup/)
4. Cloned the `px4_msgs` to ~/ros2_ws/src/
5. Cloned the `px4_ros_com` to ~/ros2_ws/src/
6. Installed `genmsg` by executing `sudo apt install ros-melodic-genmsg`
7. Sourced `ros` by executing `source /opt/ros/melodic/setup.bash`
8. Sourced `ros2_ws` by executing `source ~/ros2_ws/install/local_setup.bash`
9. Built a `ros2_ws` by executing `colcon build --symlink-install`
10. Sourced `ros2_ws` by executing `source ~/ros2_ws/install/local_setup.bash`
11. Started a PX4 instance in another terminal by executing `cd ~/src/Firmware/; make px4_sitl_rtps gazebo`
12. Executed `micrortps_client start -t UDP` inside a `NuttShell`
13. Started `micrortps_agent -t UDP` in first terminal.

To test my setup I opened a third terminal, sourced `ros2_ws` and listed topics. This genertated an error:
```
/usr/bin/python3: symbol lookup error: /home/aniemiec/ros2_ws/install/rmw_fastrtps_shared_cpp/lib/librmw_fastrtps_shared_cpp.so: undefined symbol: _ZN8eprosima8fastrtps6Domain17createParticipantERNS0_21ParticipantAttributesEPNS0_19ParticipantListenerE
```

Sourcing ros2_ws by `source ~/ros2_ws/install/setup.bash` generates this message:

```
aniemiec@r2d2l:~/ros2_ws$ source ~/ros2_ws/install/setup.bash 
ROS_DISTRO was set to 'crystal' before. Please make sure that the environment does not mix paths from different distributions.
not found: "/home/aniemiec/ros2_ws/install/ament_index_cpp/local_setup.bash"
not found: "/home/aniemiec/ros2_ws/install/ament_cmake_core/local_setup.bash"
not found: "/home/aniemiec/ros2_ws/install/ament_cmake_export_definitions/local_setup.bash"
not found: "/home/aniemiec/ros2_ws/install/ament_cmake_export_include_directories/local_setup.bash"
not found: "/home/aniemiec/ros2_ws/install/ament_cmake_export_libraries/local_setup.bash"
not found: "/home/aniemiec/ros2_ws/install/ament_cmake_export_interfaces/local_setup.bash"
not found: "/home/aniemiec/ros2_ws/install/ament_cmake_export_link_flags/local_setup.bash"
not found: "/home/aniemiec/ros2_ws/install/ament_cmake_include_directories/local_setup.bash"
not found: "/home/aniemiec/ros2_ws/install/ament_cmake_libraries/local_setup.bash"
not found: "/home/aniemiec/ros2_ws/install/ament_cmake_export_dependencies/local_setup.bash"
...
not found: "/home/aniemiec/ros2_ws/install/examples_rclpy_executors/local_setup.bash"
not found: "/home/aniemiec/ros2_ws/install/examples_rclpy_minimal_publisher/local_setup.bash"
not found: "/home/aniemiec/ros2_ws/install/examples_rclpy_minimal_subscriber/local_setup.bash"
not found: "/home/aniemiec/ros2_ws/install/test_launch_ros/local_setup.bash"
not found: "/home/aniemiec/ros2_ws/install/rqt/local_setup.bash"
not found: "/home/aniemiec/ros2_ws/install/topic_monitor/local_setup.bash"
not found: "/home/aniemiec/ros2_ws/install/ament_uncrustify/local_setup.bash"
not found: "/home/aniemiec/ros2_ws/install/demo_nodes_py/local_setup.bash"
not found: "/home/aniemiec/ros2_ws/install/examples_rclpy_minimal_client/local_setup.bash"
not found: "/home/aniemiec/ros2_ws/install/examples_rclpy_minimal_service/local_setup.bash"
ROS_DISTRO was set to 'melodic' before. Please make sure that the environment does not mix paths from different distributions.
```

This sourcing allows me to run `ros2 topic list` but I got this error when I echo any topic:
```
aniemiec@r2d2l:~/ros2_ws$ ros2 topic echo /SensorCombined_topic
Traceback (most recent call last):
  File "/home/aniemiec/ros2_ws/install/ros2cli/bin/ros2", line 11, in <module>
    load_entry_point('ros2cli', 'console_scripts', 'ros2')()
  File "/home/aniemiec/ros2_ws/build/ros2cli/ros2cli/cli.py", line 69, in main
    rc = extension.main(parser=parser, args=args)
  File "/home/aniemiec/ros2_ws/build/ros2topic/ros2topic/command/topic.py", line 43, in main
    return extension.main(args=args)
  File "/home/aniemiec/ros2_ws/build/ros2topic/ros2topic/verb/echo.py", line 70, in main
    return main(args)
  File "/home/aniemiec/ros2_ws/build/ros2topic/ros2topic/verb/echo.py", line 81, in main
    node.node, args.topic_name, args.message_type, callback)
  File "/home/aniemiec/ros2_ws/build/ros2topic/ros2topic/verb/echo.py", line 124, in subscriber
    msg_module = import_message_type(topic_name, message_type)
  File "/home/aniemiec/ros2_ws/build/ros2topic/ros2topic/api/__init__.py", line 71, in import_message_type
    module = importlib.import_module(package_name + '.' + middle_module)
  File "/usr/lib/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 941, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 953, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'px4_ros_com'
```

I have also tested scripts and followed instructions in PX4 docks but ended up with same results.

TSC21, 2019-04-07:
@ArkadiuszNiemiec you have a full guide [here](https://dev.px4.io/en/middleware/micrortps.html) explaining how to setup the environment. You also have build scripts [here](https://github.com/PX4/px4_ros_com/tree/master/scripts) that do all the above job for you. Please do follow the guide or use the scripts and come back if you are not successful.

ArkadiuszNiemiec, 2019-04-07:
As I said: `I have also tested scripts and followed instructions in PX4 docks but ended up with same results.`
But I will do a clean VM with Ubuntu 18.04 and test it again.

TSC21, 2019-04-07:
Try using the `build_all.bash` script and then use the respective opened terminals to launch the respective daemons and nodes. If you are not planning to use ROS 1, then just use the `build_ros2_workspace.bash`. Any if these scripts will setup the environments properly on your system.

ArkadiuszNiemiec, 2019-04-07:
I have prepared a Dockerfile to test `px4_ros_com`: https://github.com/ArkadiuszNiemiec/px4_ros_com/tree/master/docker
Dockerfile includes your `px4-dev-ros2-crystal`, clones PX4 Firmware and do a `pxr_ros_com` setup the same way the docs and script does:

```
# --- px4_ros_com ---
RUN ["/bin/bash", "-c", "\
    cd ${HOME_DIR} && \
    mkdir -p ${HOME_DIR}/px4_ros_com_ros2/src && \
    git clone https://github.com/PX4/px4_ros_com.git ${HOME_DIR}/px4_ros_com_ros2/src/px4_ros_com && \
    git clone https://github.com/PX4/px4_msgs.git ${HOME_DIR}/px4_ros_com_ros2/src/px4_msgs && \
    source /opt/ros/melodic/setup.bash && \
    source /opt/ros/crystal/setup.bash && \
    cd ${HOME_DIR}/px4_ros_com_ros2 && colcon build --symlink-install --event-handlers console_direct+ \
"]
```

At the end I am launching a screenrc file that has three sessions: first launches a `px4_sitl_rtps`, second one launches `micrortps_agent -t UDP` and the third one is launching `sensor_combined_listener`. I am manually starting the `micrortps_client start -t UDP` in SITL session.

1. When I change line 36 in Dockerfile (`cd ${HOME_DIR}/px4_ros_com_ros2 && colcon build --symlink-install --event-handlers console_direct+ \`) to `${HOME_DIR}/px4_ros_com_ros2/src/px4_ros_com/scripts/ build_ros2_workspace.bash` I got this message: 
```
Cloning into '/root/px4_ros_com_ros2/src/px4_ros_com'...
Cloning into '/root/px4_ros_com_ros2/src/px4_msgs'...
ROS_DISTRO was set to 'melodic' before. Please make sure that the environment does not mix paths from different distributions.
Cloning into 'ros1_bridge'...
[3.793s] ERROR:colcon.colcon_core.package_identification:ROS package '/opt/ros/crystal/share/demo_nodes_py' with build type 'ament_python' has no 'setup.py' file
[3.844s] ERROR:colcon.colcon_core.package_identification:ROS package '/opt/ros/crystal/share/examples_rclpy_executors' with build type 'ament_python' has no 'setup.py' file
[3.846s] ERROR:colcon.colcon_core.package_identification:ROS package '/opt/ros/crystal/share/examples_rclpy_minimal_client' with build type 'ament_python' has no 'setup.py' file
[3.849s] ERROR:colcon.colcon_core.package_identification:ROS package '/opt/ros/crystal/share/examples_rclpy_minimal_publisher' with build type 'ament_python' has no 'setup.py' file
[3.851s] ERROR:colcon.colcon_core.package_identification:ROS package '/opt/ros/crystal/share/examples_rclpy_minimal_service' with build type 'ament_python' has no 'setup.py' file
[3.854s] ERROR:colcon.colcon_core.package_identification:ROS package '/opt/ros/crystal/share/examples_rclpy_minimal_subscriber' with build type 'ament_python' has no 'setup.py' file
[4.123s] ERROR:colcon.colcon_core.package_identification:ROS package '/opt/ros/crystal/share/teleop_twist_keyboard' with build type 'ament_python' has no 'setup.py' file
[4.179s] ERROR:colcon.colcon_core.package_identification:ROS package '/opt/ros/crystal/share/topic_monitor' with build type 'ament_python' has no 'setup.py' file
```
so I changed the script to manually building the `px4_ros_com_ros2` workspace. I will test if further.

2. `micrortps_client status` confirms that the client is working on PX4 SITL but there's no message in `micrortps_agent` and in `sensor_combined_listener`.

3. Typing `ros2 topic hz /SensorCombined_topic` in fourth sessions returns:
```
WARNING: topic [/SensorCombined_topic] does not appear to be published yet
```
OR
```
Cannot echo topic '/SensorCombined_topic', as it contains more than one type: [px4_msgs/SensorCombined, px4_ros_com/SensorCombined]
```

TSC21, 2019-04-07:
The problem you repor in 3. is known and is going tl be solved soon (it has to do with the major reestructure that happened on the repo).

ArkadiuszNiemiec, 2019-04-07:
And the problem no 1 is the problem with using a script in Dockerfile. The '$ROS_WS_DIR' is `/` and that's not a correct path. We can ignore it. So the only problem is that `sensor_combined_listener` is not reporting any data and I cannot print them using `ros2 topic echo`.

TSC21, 2019-04-07:
> And the problem no 1 is the problem with using a script in Dockerfile. The '$ROS_WS_DIR' is / and that's not a correct path. We can ignore it.

The only problem there is that you are not using the script correctly. It should be `./build_all.bash --ros1_ws_dir <your/path/to/ros1_ws>`.

TSC21, 2019-04-07:
> So the only problem is that sensor_combined_listener is not reporting any data and I cannot print them using ros2 topic echo.

Solution coming soon.

ArkadiuszNiemiec, 2019-04-07:
Ok, thanks. I will debug the software myself so I can contribute it the future.

TSC21, 2019-04-08:
@ArkadiuszNiemiec the related issue is https://github.com/PX4/px4_ros_com/issues/14 and it got fixed now. Closing this as solved.

closed 2019-04-08 by TSC21

ArkadiuszNiemiec, 2019-04-08:
@TSC21 yes, this fixes my problems. Thanks.