PX4/px4_ros_com #28: sender_thread.join() blocks the agent from quitting

ArkadiuszNiemiec, 2019-04-17:
I am not able to cleanly exit running `micrortps_agent` by pressing `ctrl + c`:

```
...
WindEstimate publisher started
CollisionConstraints publisher started
^CInterrupt signal (2) received.
Close sender socket
Close receiver socket
Read fail 9
^CInterrupt signal (2) received.
^CInterrupt signal (2) received.
^CInterrupt signal (2) received.
^CInterrupt signal (2) received.
^CInterrupt signal (2) received.
```

https://github.com/PX4/px4_ros_com/blob/59cfddd9dbe1d5e92b5e1b03dbee56e4243758d6/templates/microRTPS_agent.cpp.template#L303
...blocks the process until the
https://github.com/PX4/px4_ros_com/blob/59cfddd9dbe1d5e92b5e1b03dbee56e4243758d6/templates/microRTPS_agent.cpp.template#L185
finishes. The problem is that for me the 
https://github.com/PX4/px4_ros_com/blob/59cfddd9dbe1d5e92b5e1b03dbee56e4243758d6/templates/microRTPS_agent.cpp.template#L194
will only be false if `if (nullptr == topic_ID) return false;` and this will never happen.

Is there a way to fix it?

ArkadiuszNiemiec, 2019-04-17:
Maybe we can add a semaphore that breaks this loop?
https://github.com/PX4/px4_ros_com/blob/59cfddd9dbe1d5e92b5e1b03dbee56e4243758d6/templates/microRTPS_agent.cpp.template#L194

TSC21, 2019-04-18:
@ArkadiuszNiemiec a PR with a fix is welcomed.

closed 2019-04-21 by ArkadiuszNiemiec