PX4/px4_ros_com #30: Issues building px4_ros_com

paagutie, 2019-06-25:
I would like to communicate ROS2 with Pixhawk 2.4.8, but following the instruccions in the [documentation](https://dev.px4.io/en/middleware/micrortps.html#building-the-workspaces), the building of the px4_ros_com package ($ colcon build --symlink-install --package-skip ros1_bridge --event-handlers console_direct+) fails. 
What i get is as follows:

`Finished <<< px4_msgs [25.5s]`
`Starting >>> px4_ros_com`
`-- Using C++17 compiler`
`-- Found ament_cmake: 0.6.1 (/opt/ros/crystal/share/ament_cmake/cmake)`
`-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.8", minimum required is "3") `
`-- Using PYTHON_EXECUTABLE: /usr/bin/python3`
`-- Override CMake install command with custom implementation using symlinks instead of copying resources`
`-- Found builtin_interfaces: 0.6.3 (/opt/ros/crystal/share/builtin_interfaces/cmake)`
`-- Found rosidl_adapter: 0.6.3 (/opt/ros/crystal/share/rosidl_adapter/cmake)`
`-- Found rosidl_default_generators: 0.6.0 (/opt/ros/crystal/share/rosidl_default_generators/cmake)`
`-- Found PrismTech OpenSplice: /usr/share/opensplice/cmake`
`-- Found rclcpp: 0.6.4 (/opt/ros/crystal/share/rclcpp/cmake)`
`-- Found rmw_implementation_cmake: 0.6.1 (/opt/ros/crystal/share/rmw_implementation_cmake/cmake)`
`-- Found rmw_fastrtps_cpp: 0.6.2 (/opt/ros/crystal/share/rmw_fastrtps_cpp/cmake)`
`-- Found px4_msgs: 0.1.0 (/home/pablo/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/cmake)`
`-- Found PythonInterp: /usr/bin/python3 (found version "3.6.8") `
`-- Retrieving list of msgs to send...`
`-- List of msgs to send: CameraCapture, CameraTrigger, CollisionReport, DebugArray, DebugKeyValue, DebugValue, DebugVect, ObstacleDistance, OpticalFlow, PositionSetpoint, PositionSetpointTriplet, TrajectoryWaypoint, VehicleTrajectoryWaypoint`
`-- Retrieving list of msgs to receive...`
`-- List of msgs to receive: AdcReport, Airspeed, BatteryStatus, Cpuload, DistanceSensor, EstimatorStatus, HomePosition, IridiumsbdStatus, RadioStatus, SatelliteInfo, SensorBaro, SensorCombined, SensorSelection, VehicleAttitude, VehicleOdometry, VtolVehicleStatus, WindEstimate, CollisionConstraints`
`-- fastrtpsgen found in `
`-- px4_msgs message dir under /home/pablo/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/msg`
`-- Configuring done`                              
`-- Generating done`
`-- Build files have been written to: /home/pablo/px4_ros_com_ros2/build/px4_ros_com`
`[  1%] Generating micro-RTPS agent code...`
`All good. RTPS ID's are unique`
`Traceback (most recent call last):`
 `File "/home/pablo/px4_ros_com_ros2/src/px4_ros_com/scripts/generate_microRTPS_bridge.py", line 401, in <module>
    generate_agent(agent_out_dir)`
 ` File "/home/pablo/px4_ros_com_ros2/src/px4_ros_com/scripts/generate_microRTPS_bridge.py", line 291, in generate_agent
    package, px_generate_uorb_topic_files.INCL_DEFAULT, classifier.msg_id_map, uRTPS_PUBLISHER_SRC_TEMPL_FILE)`
  `File "/home/pablo/px4_ros_com_ros2/src/px4_ros_com/scripts/px_generate_uorb_topic_files.py", line 234, in generate_topic_file
    return generate_by_template(output_file, template_file, em_globals)`
  `File "/home/pablo/px4_ros_com_ros2/src/px4_ros_com/scripts/px_generate_uorb_topic_files.py", line 300, in generate_by_template
    interpreter.file(open(template_file))`
`FileNotFoundError: [Errno 2] No such file or directory: '/home/pablo/px4_ros_com_ros2/src/px4_ros_com/templates/Publisher.cpp.em'`
`Exception ignored in: <bound method Interpreter.__del__ of <empy pseudomodule/interpreter at 0x7f56fb3cb048>>`
`Traceback (most recent call last):`
 ` File "/usr/lib/python3/dist-packages/em.py", line 2094, in __del__`
`  File "/usr/lib/python3/dist-packages/em.py", line 2157, in shutdown`
`  File "/usr/lib/python3/dist-packages/em.py", line 2646, in finalize`
 ` File "/usr/lib/python3/dist-packages/em.py", line 2200, in push`
`AttributeError: '_io.TextIOWrapper' object has no attribute 'push'`
`CMakeFiles/micrortps_agent.dir/build.make:61: recipe for target '/home/pablo/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/microRTPS_agent.cpp' failed`
`CMakeFiles/Makefile2:387: recipe for target 'CMakeFiles/micrortps_agent.dir/all' failed`
`Makefile:140: recipe for target 'all' failed`
`make[2]: *** [/home/pablo/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/microRTPS_agent.cpp] Error 1`
`make[1]: *** [CMakeFiles/micrortps_agent.dir/all] Error 2`
`make: *** [all] Error 2`
`---`
`Failed   <<< px4_ros_com	[ Exited with code 2 ]`

Does anyone have any idea how to solve this?

TSC21, 2019-06-25:
Can you please try the provided scripts to build available under `scripts`?

TSC21, 2019-06-25:
Also, you seem to be trying to build the `px4_ros_com` package without the `px4_msgs` package. You need the `px4_msgs` package under the same workspace and build it as well.

paagutie, 2019-06-27:
Thank you for your answer. Both packages are under the same workspace. When I use the build_ros2_workspace.bash script the px4_msgs packete compilation works, but px4_ros_com always fails, despite meeting all dependencies. I've cleaned the workspace again, reinstalled ROS2 and dependencies, but the problem persists.

TSC21, 2019-06-27:
OK I just saw that there was an update to the scripts, which were expecting a `.em` extension on the template files instead of `.template`. This was updated now in https://github.com/PX4/px4_ros_com/commit/1b9df75a78abd4bdb2384b5232a8cae98e00e10f and it builds as expected. Please pull the new commit, retry and let me know.

paagutie, 2019-06-27:
With the new update of the scripts the error is gone, but now I have problems with "fastrtpsgen". I installed it from source and is hosted at /usr/local/bin.

`Finished <<< px4_msgs [1min 19s]`                                
`Starting >>> px4_ros_com`
`-- The C compiler identification is GNU 7.4.0 `    
`-- The CXX compiler identification is GNU 7.4.0 `      
`-- Check for working C compiler: /usr/bin/cc`          
`-- Check for working C compiler: /usr/bin/cc -- works` 
`-- Detecting C compiler ABI info`
`-- Detecting C compiler ABI info - done `              
`-- Detecting C compile features`
`-- Detecting C compile features - done `               
`-- Check for working CXX compiler: /usr/bin/c++`
`-- Check for working CXX compiler: /usr/bin/c++ -- works`
`-- Detecting CXX compiler ABI info`
`-- Detecting CXX compiler ABI info - done `            
`-- Detecting CXX compile features    `                 
`-- Detecting CXX compile features - done`               
`-- Performing Test COMPILER_SUPPORTS_CXX17 `            
`-- Performing Test COMPILER_SUPPORTS_CXX17 - Success  ` 
`-- Performing Test COMPILER_SUPPORTS_CXX14`
`-- Performing Test COMPILER_SUPPORTS_CXX14 - Success `  
`-- Performing Test COMPILER_SUPPORTS_CXX11`
`-- Performing Test COMPILER_SUPPORTS_CXX11 - Success`   
`-- Performing Test COMPILER_SUPPORTS_CXX0X`
`-- Performing Test COMPILER_SUPPORTS_CXX0X - Success `  
`-- Using C++17 compiler`
`-- Found ament_cmake: 0.6.1 (/opt/ros/crystal/share/ament_cmake/cmake)`
`-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.8", minimum required is "3") `
`-- Using PYTHON_EXECUTABLE: /usr/bin/python3`
`-- Override CMake install command with custom implementation using symlinks instead of copying resources`
`-- Found builtin_interfaces: 0.6.3 (/opt/ros/crystal/share/builtin_interfaces/cmake)`
`-- Found rosidl_adapter: 0.6.3 (/opt/ros/crystal/share/rosidl_adapter/cmake)`
`-- Found rosidl_default_generators: 0.6.0 (/opt/ros/crystal/share/rosidl_default_generators/cmake)`
`-- Found OpenSSL: /usr/lib/x86_64-linux-gnu/libcrypto.so (found version "1.1.1") `
`-- Found FastRTPS: /opt/ros/crystal/include     `       
`-- Found rosidl_generator_dds_idl: 0.6.0 (/opt/ros/crystal/share/rosidl_generator_dds_idl/cmake)`
`-- Found rclcpp: 0.6.4 (/opt/ros/crystal/share/rclcpp/cmake)`
`-- Found rmw_implementation_cmake: 0.6.1 (/opt/ros/crystal/share/rmw_implementation_cmake/cmake)`
`-- Found rmw_fastrtps_cpp: 0.6.2 (/opt/ros/crystal/share/rmw_fastrtps_cpp/cmake)`
`-- Found px4_msgs: 0.1.0 (/home/ubuntu/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/cmake)`
`-- Found PythonInterp: /usr/bin/python3 (found version "3.6.8") `
`-- Retrieving list of msgs to send...`
`-- List of msgs to send: CameraCapture, CameraTrigger, CollisionReport, DebugArray, DebugKeyValue, DebugValue, DebugVect, ObstacleDistance, OpticalFlow, PositionSetpoint, PositionSetpointTriplet, TrajectoryWaypoint, VehicleTrajectoryWaypoint`
`-- Retrieving list of msgs to receive...`
`-- List of msgs to receive: AdcReport, Airspeed, BatteryStatus, Cpuload, DistanceSensor, EstimatorStatus, HomePosition, IridiumsbdStatus, RadioStatus, SatelliteInfo, SensorBaro, SensorCombined, SensorSelection, VehicleAttitude, VehicleOdometry, VtolVehicleStatus, WindEstimate, CollisionConstraints`
`-- fastrtpsgen found in `
`-- px4_msgs message dir under /home/ubuntu/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/msg`
`-- Configuring done  `                                  
`-- Generating done`                                     
`-- Build files have been written to: /home/ubuntu/px4_ros_com_ros2/build/px4_ros_com`
`[  1%] Generating micro-RTPS agent code...  `           
`All good. RTPS ID's are unique     `                       
`Traceback (most recent call last):                        
  File "/home/ubuntu/px4_ros_com_ros2/src/px4_ros_com/scripts/generate_microRTPS_bridge.py", line 401, in <module>
    generate_agent(agent_out_dir)`
  `File "/home/ubuntu/px4_ros_com_ros2/src/px4_ros_com/scripts/generate_microRTPS_bridge.py", line 334, in generate_agent`
   ` "fastrtpsgen not found. Specify the location of fastrtpsgen with the -f flag")`
`Exception: fastrtpsgen not found. Specify the location of fastrtpsgen with the -f flag`
`CMakeFiles/micrortps_agent.dir/build.make:61: recipe for target '/home/ubuntu/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/microRTPS_agent.cpp' failed`
`make[2]: *** [/home/ubuntu/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/microRTPS_agent.cpp] Error 1`
`make[2]: *** Deleting file '/home/ubuntu/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/microRTPS_agent.cpp'`
`CMakeFiles/Makefile2:387: recipe for target 'CMakeFiles/micrortps_agent.dir/all' failed`
`make[1]: *** [CMakeFiles/micrortps_agent.dir/all] Error 2`
`Makefile:140: recipe for target 'all' failed`
`make: *** [all] Error 2`
`--- stderr: px4_ros_com        `                           
`Traceback (most recent call last):`
  `File "/home/ubuntu/px4_ros_com_ros2/src/px4_ros_com/scripts/generate_microRTPS_bridge.py", line 401, in <module>`
   ` generate_agent(agent_out_dir)`
  `File "/home/ubuntu/px4_ros_com_ros2/src/px4_ros_com/scripts/generate_microRTPS_bridge.py", line 334, in generate_agent`
  `  "fastrtpsgen not found. Specify the location of fastrtpsgen with the -f flag")`
`Exception: fastrtpsgen not found. Specify the location of fastrtpsgen with the -f flag`
`make[2]: *** [/home/ubuntu/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/microRTPS_agent.cpp] Error 1`
`make[2]: *** Deleting file '/home/ubuntu/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/microRTPS_agent.cpp'`
`make[1]: *** [CMakeFiles/micrortps_agent.dir/all] Error 2`
`make: *** [all] Error 2`
`---`
`Failed   <<< px4_ros_com	[ Exited with code 2 ]`

When I specify the location of fastrtpsgen in "generate_microRTPS_bridge" I get an exception with error code 126. What could this be due to?

TSC21, 2019-06-27:
Put `export FASTRTPSGEN_DIR="/usr/local/bin/"` on your `bashrc`.

paagutie, 2019-06-27:
It shouldn't be necessary, but that's how it worked! Thank you so much for your help :)

TSC21, 2019-06-27:
OK good. Closing then.

closed 2019-06-27 by TSC21