PX4/px4_ros_com #31: colcon cannot find the location of FastRTPSGen

MartinAtCoventry, 2019-11-12:
**Describe the bug**
I have some compilation problems when building a PX4-Fastrtps-bridge by the `px4_ros_com` on a Raspberry Pi4 board. The python script `generate_microRTPS_bridge.py` in px4_ros_com always cannot find the `fastrtpsgen` in my RPI4 system. Also, the `px4_msgs package` has been compiled successfully.

I have installed the libraries and dependency packages as follow:
fastcdr - 1.0.10
fastrtps - 1.9.2
foonathan_memory - 0.6.2.a
fastrtpsgen in /home/pi/Fast-RTPS-Gen/scripts/
Java1.8.0 - JDK 1.8.0_231 in /usr/lib/jvm/jdk1.8.0_231
Gradle - 4.4.1
ROS2 - Dashing (by Building from source)
px4_msgs - master

**To Reproduce**
1. export FASTRTPSGEN_DIR="/home/pi/Fast-RTPS-Gen/scripts/"
2. Into the workspace of the `px4_ros_com_ros2`.
3. colcon build --symlink-install --packages-skip px4_msgs --event-handlers console_direct+

**Expected behavior**
Help me to get a way which can make `generate_microRTPS_bridge.py` find the `fastrtpsgen` program.

**Log Files and Screenshots**
```
pi@OFFBOARD:~/px4_ros_com_ros2$ colcon build --symlink-install --packages-skip px4_msgs --event-handlers console_direct+
Starting >>> px4_ros_com
[  1%] Generating micro-RTPS agent code...      
Traceback (most recent call last):                 
  File "/home/pi/px4_ros_com_ros2/src/px4_ros_com/scripts/generate_microRTPS_bridge.py", line 261, in <module>
    "FastRTPSGen not found. Specify the location of fastrtpsgen with the -f flag")
Exception: FastRTPSGen not found. Specify the location of fastrtpsgen with the -f flag
CMakeFiles/micrortps_agent.dir/build.make:61: recipe for target '/home/pi/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/microRTPS_agent.cpp' failed
make[2]: *** [/home/pi/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/microRTPS_agent.cpp] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/micrortps_agent.dir/all' failed
Makefile:140: recipe for target 'all' failed
make[1]: *** [CMakeFiles/micrortps_agent.dir/all] Error 2
make: *** [all] Error 2
--- stderr: px4_ros_com
Traceback (most recent call last):
  File "/home/pi/px4_ros_com_ros2/src/px4_ros_com/scripts/generate_microRTPS_bridge.py", line 261, in <module>
    "FastRTPSGen not found. Specify the location of fastrtpsgen with the -f flag")
Exception: FastRTPSGen not found. Specify the location of fastrtpsgen with the -f flag
make[2]: *** [/home/pi/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/microRTPS_agent.cpp] Error 1
make[1]: *** [CMakeFiles/micrortps_agent.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< px4_ros_com	[ Exited with code 2 ]

Summary: 0 packages finished [4.54s]
  1 package failed: px4_ros_com
  1 package had stderr output: px4_ros_com
```

MartinAtCoventry, 2019-11-12:
**Describe the bug**
I have solved the problem that `generate_microRTPS_bridge.py` cannot find fastrtpsgen. Through checking the version information of Java on my board, I found a version conflict of Java. Uninstall a Java-1.11, which was installed during `sudo apt-get install gradle`, and remove the px4_ros_com fold in the build fold. And then to do : `colcon build --symlink-install --packages-skip px4_msgs --event-handlers console_direct+`. In the building process, the fastrtpsgen can be found again.

However, the building process throws other compilation errors as well.

**Expected behavior**
Help me to analyse these errors and solve them.

**Log Files and Screenshots**
1. colcon detected out the libraries on my board.
```
pi@OFFBOARD:~/px4_ros_com_ros2$ colcon build --symlink-install --packages-skip px4_msgs --event-handlers console_direct+
Starting >>> px4_ros_com
-- The C compiler identification is GNU 7.4.0   
-- The CXX compiler identification is GNU 7.4.0 
-- Check for working C compiler: /usr/bin/cc    
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done         
-- Detecting C compile features
-- Detecting C compile features - done          
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done       
-- Detecting CXX compile features               
-- Detecting CXX compile features - done        
-- Performing Test COMPILER_SUPPORTS_CXX17
-- Performing Test COMPILER_SUPPORTS_CXX17 - Success
-- Performing Test COMPILER_SUPPORTS_CXX14
-- Performing Test COMPILER_SUPPORTS_CXX14 - Success
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Using C++17 compiler
-- Found ament_cmake: 0.7.3 (/home/pi/ros2_dashing/install/ament_cmake/share/ament_cmake/cmake)
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.8", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Override CMake install command with custom implementation using symlinks instead of copying resources
-- Found builtin_interfaces: 0.7.4 (/home/pi/ros2_dashing/install/builtin_interfaces/share/builtin_interfaces/cmake)
-- Found rosidl_adapter: 0.7.7 (/home/pi/ros2_dashing/install/rosidl_adapter/share/rosidl_adapter/cmake)
-- Found rosidl_default_generators: 0.7.0 (/home/pi/ros2_dashing/install/rosidl_default_generators/share/rosidl_default_generators/cmake)
-- Found PrismTech OpenSplice: /usr/share/opensplice/cmake
-- Found OpenSSL: /usr/lib/aarch64-linux-gnu/libcrypto.so (found version "1.1.1") 
-- Found FastRTPS: /usr/local/include             
-- Found rclcpp: 0.7.11 (/home/pi/ros2_dashing/install/rclcpp/share/rclcpp/cmake)
-- Found rmw_implementation_cmake: 0.7.2 (/home/pi/ros2_dashing/install/rmw_implementation_cmake/share/rmw_implementation_cmake/cmake)
-- Looking for pthread.h
-- Looking for pthread.h - found                  
-- Looking for pthread_create
-- Looking for pthread_create - not found         
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found  
-- Found Threads: TRUE  
-- Found rmw_fastrtps_cpp: 0.7.6 (/home/pi/ros2_dashing/install/rmw_fastrtps_cpp/share/rmw_fastrtps_cpp/cmake)
-- Found px4_msgs: 2.0.1 (/home/pi/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/cmake)
-- Found PythonInterp: /usr/bin/python3 (found version "3.6.8") 
-- fastrtpsgen found in /home/pi/Fast-RTPS-Gen/scripts
java version "1.8.0_231"                          
Java(TM) SE Runtime Environment (build 1.8.0_231-b11)
Java HotSpot(TM) 64-Bit Server VM (build 25.231-b11, mixed mode)
-- fastrtpsgen version 1.0.2                      
-- Retrieving list of msgs to send...             
-- List of msgs to send: CameraCapture, CameraTrigger, CollisionReport, DebugArray, DebugKeyValue, DebugValue, DebugVect, ObstacleDistance, OpticalFlow, PositionSetpoint, PositionSetpointTriplet, TrajectoryWaypoint, VehicleTrajectoryWaypoint, VehicleMocapOdometry, VehicleVisualOdometry
-- Retrieving list of msgs to receive...
-- List of msgs to receive: AdcReport, Airspeed, BatteryStatus, Cpuload, DistanceSensor, EstimatorStatus, HomePosition, IridiumsbdStatus, RadioStatus, SatelliteInfo, SensorBaro, SensorCombined, SensorSelection, VehicleAttitude, VehicleOdometry, VtolVehicleStatus, WindEstimate, CollisionConstraints
-- fastrtpsgen found in /home/pi/Fast-RTPS-Gen/scripts/
[... 669 lines cut ...]
/home/pi/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/DistanceSensor_Publisher.cpp: In member function ‘virtual void DistanceSensor_Publisher::PubListener::onPublicationMatched(eprosima::fastrtps::Publisher*, eprosima::fastrtps::rtps::MatchingInfo&)’:
/home/pi/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/DistanceSensor_Publisher.cpp:83:77: warning: unused parameter ‘pub’ [-Wunused-parameter]
 void DistanceSensor_Publisher::PubListener::onPublicationMatched(Publisher* pub, MatchingInfo& info)
                                                                             ^~~
/home/pi/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/Cpuload_Publisher.cpp: In member function ‘bool Cpuload_Publisher::init()’:
[... cut ...]
make[2]: *** [CMakeFiles/micrortps_agent.dir/src/micrortps_agent/EstimatorStatus_Publisher.cpp.o] Error 1
make[2]: *** [CMakeFiles/micrortps_agent.dir/src/micrortps_agent/DistanceSensor_Publisher.cpp.o] Error 1
make[2]: *** [CMakeFiles/micrortps_agent.dir/src/micrortps_agent/Cpuload_Publisher.cpp.o] Error 1
make[1]: *** [CMakeFiles/micrortps_agent.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< px4_ros_com	[ Exited with code 2 ]

Summary: 0 packages finished [9min 54s]
  1 package failed: px4_ros_com
  1 package had stderr output: px4_ros_com
```

TSC21, 2019-11-12:
You need Fast-RTPS-Gen installed from source: https://github.com/eProsima/Fast-RTPS-Gen. And also Fast-RTPS 1.8.2.

TSC21, 2019-11-12:
Actually nevermind. Your problem is what is addressed in https://github.com/ros2/rosidl/pull/419. Though, I must warn you that ROS2 Dashing is still not supported. You need to use ROS2 Crystal with Fast-RTPS 1.7.2 (and respective *fastrtpsgen*).

MartinAtCoventry, 2019-11-13:
@TSC21, thank you for giving me the guidance. I will retry to install ROS2 Crystal and Fast-RTPS 1.7.2 on my board.

MartinAtCoventry, 2019-11-13:
**Describe the bug**
Hi @TSC21 , according to your guidance, I redeployed ROS2 crystal on my RPI4 board as well as reinstalled Fastrtps v1.8.2. Now, It seems to successfully build the `PX4_ROS_COM` package. There, however, are still some warning information about the `microRTPS_transport.cpp`, `microRTPS_agent.cpp` and some `*_Publisher.cpp` files.

So, I don't know whether these warings will impact the program `micrortps_agent` when it running.

Btw: The version of the libraries on my board
fastcdr - 1.0.8
fastrtps - 1.8.2
foonathan_memory - 0.6.2.a
fastrtpsgen 1.0.0 in /home/pi/Fast-RTPS-Gen/scripts/
Java1.8.0 - JDK 1.8.0_231 in /usr/lib/jvm/jdk1.8.0_231
Gradle - 6.0
ROS2 - Crystal (by Building from source)
px4_msgs - master

ROS_VERSION=2
ROS_PYTHON_VERSION=3
ROS_DISTRO=crystal

**To Reproduce**
1. source ~/ros2_crystal/install/local_setup.bash 
2. Into the workspace of the `px4_ros_com_ros2`.
3. colcon build --symlink-install --packages-skip px4_msgs --event-handlers console_direct+

**Expected behavior**
Help me to get a way which can make `generate_microRTPS_bridge.py` find the `fastrtpsgen` program.

**Log Files and Screenshots**
```
pi@RPI4:~/px4_ros_com_ros2$ colcon build --symlink-install --packages-skip px4_msgs --event-handlers console_direct+
Starting >>> px4_ros_com
-- The C compiler identification is GNU 7.4.0   
-- The CXX compiler identification is GNU 7.4.0 
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info                
-- Detecting C compiler ABI info - done         
-- Detecting C compile features                 
-- Detecting C compile features - done          
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done       
-- Detecting CXX compile features
-- Detecting CXX compile features - done        
-- Performing Test COMPILER_SUPPORTS_CXX17
-- Performing Test COMPILER_SUPPORTS_CXX17 - Success
-- Performing Test COMPILER_SUPPORTS_CXX14
-- Performing Test COMPILER_SUPPORTS_CXX14 - Success
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Using C++17 compiler
-- Found ament_cmake: 0.6.1 (/home/pi/ros2_crystal/install/ament_cmake/share/ament_cmake/cmake)
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.8", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Override CMake install command with custom implementation using symlinks instead of copying resources
-- Found builtin_interfaces: 0.6.3 (/home/pi/ros2_crystal/install/builtin_interfaces/share/builtin_interfaces/cmake)
-- Found rosidl_adapter: 0.6.3 (/home/pi/ros2_crystal/install/rosidl_adapter/share/rosidl_adapter/cmake)
-- Found rosidl_default_generators: 0.6.0 (/home/pi/ros2_crystal/install/rosidl_default_generators/share/rosidl_default_generators/cmake)
-- Found PrismTech OpenSplice: /usr/share/opensplice/cmake
-- Found OpenSSL: /usr/lib/aarch64-linux-gnu/libcrypto.so (found version "1.1.1") 
-- Found FastRTPS: /usr/local/include             
-- Found rclcpp: 0.6.4 (/home/pi/ros2_crystal/install/rclcpp/share/rclcpp/cmake)
-- Found rmw_implementation_cmake: 0.6.1 (/home/pi/ros2_crystal/install/rmw_implementation_cmake/share/rmw_implementation_cmake/cmake)
-- Found rmw_fastrtps_cpp: 0.6.2 (/home/pi/ros2_crystal/install/rmw_fastrtps_cpp/share/rmw_fastrtps_cpp/cmake)
-- Found px4_msgs: 2.0.1 (/home/pi/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/cmake)
-- Found PythonInterp: /usr/bin/python3 (found version "3.6.8") 
-- fastrtpsgen found in /home/pi/Fast-RTPS-Gen/scripts
[... 468 lines cut ...]
/home/pi/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/VehicleAttitude_Publisher.cpp:83:78: warning: unused parameter ‘pub’ [-Wunused-parameter]
 void VehicleAttitude_Publisher::PubListener::onPublicationMatched(Publisher* pub, MatchingInfo& info)
                                                                              ^~~
/home/pi/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/VehicleOdometry_Publisher.cpp: In member function ‘virtual void VehicleOdometry_Publisher::PubListener::onPublicationMatched(eprosima::fastrtps::Publisher*, eprosima::fastrtps::rtps::MatchingInfo&)’:
/home/pi/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/VehicleOdometry_Publisher.cpp:83:78: warning: unused parameter ‘pub’ [-Wunused-parameter]
 void VehicleOdometry_Publisher::PubListener::onPublicationMatched(Publisher* pub, MatchingInfo& info)
                                                                              ^~~
/home/pi/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/VtolVehicleStatus_Publisher.cpp: In member function ‘virtual void VtolVehicleStatus_Publisher::PubListener::onPublicationMatched(eprosima::fastrtps::Publisher*, eprosima::fastrtps::rtps::MatchingInfo&)’:
/home/pi/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/VtolVehicleStatus_Publisher.cpp:83:80: warning: unused parameter ‘pub’ [-Wunused-parameter]
 void VtolVehicleStatus_Publisher::PubListener::onPublicationMatched(Publisher* pub, MatchingInfo& info)
                                                                                ^~~
[... cut ...] tps::MatchingInfo&)’:
/home/pi/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/CollisionConstraints_Publisher.cpp:83:83: warning: unused parameter ‘pub’ [-Wunused-parameter]
 void CollisionConstraints_Publisher::PubListener::onPublicationMatched(Publisher* pub, MatchingInfo& info)
                                                                                   ^~~
---
Finished <<< px4_ros_com [3min 19s]

Summary: 1 package finished [3min 19s]
  1 package had stderr output: px4_ros_com
```

TSC21, 2019-11-13:
The warnings are not an issue. Closing this issue as the reported problem got solved.

closed 2019-11-13 by TSC21

MartinAtCoventry, 2019-11-13:
OK, thanks a lot for @TSC21 :)

MartinAtCoventry, 2019-11-14:
Hi @TSC21 , through test my ROS2 crystal, I found it cannot run well after successfully compiling out `px4_ros_com`. So, I cleaned my RPI4 board and re-deployed a new ROS2 crystal with its dependencies.
However, there are some new errors during building px4_ros_com. I don't know which component is wrong ?

Btw:
fastcdr v1.0.8
fastrtps v1.7.0
fastrtpsgen 1.0.0 in /home/pi/Fast-RTPS-Gen/scripts/
Java1.8.0 - JDK 1.8.0_231 in /usr/lib/jvm/jdk1.8.0_231
Gradle - 6.0

**Log Files and Screenshots**
```
pi@NanoPi-M4:~/px4_ros_com_ros2$ colcon build --symlink-install --packages-skip px4_msgs --event-handlers console_direct+
Starting >>> px4_ros_com
-- The C compiler identification is GNU 7.4.0   
-- The CXX compiler identification is GNU 7.4.0 
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done         
-- Detecting C compile features
-- Detecting C compile features - done          
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done       
-- Detecting CXX compile features
-- Detecting CXX compile features - done        
-- Performing Test COMPILER_SUPPORTS_CXX17
-- Performing Test COMPILER_SUPPORTS_CXX17 - Success
-- Performing Test COMPILER_SUPPORTS_CXX14
-- Performing Test COMPILER_SUPPORTS_CXX14 - Success
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Using C++17 compiler
-- Found ament_cmake: 0.6.1 (/home/pi/ros2_crystal/install/ament_cmake/share/ament_cmake/cmake)
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.8", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Override CMake install command with custom implementation using symlinks instead of copying resources
-- Found builtin_interfaces: 0.6.3 (/home/pi/ros2_crystal/install/builtin_interfaces/share/builtin_interfaces/cmake)
-- Found rosidl_adapter: 0.6.3 (/home/pi/ros2_crystal/install/rosidl_adapter/share/rosidl_adapter/cmake)
-- Found rosidl_default_generators: 0.6.0 (/home/pi/ros2_crystal/install/rosidl_default_generators/share/rosidl_default_generators/cmake)
-- Found PrismTech OpenSplice: /usr/share/opensplice/cmake
-- Found FastRTPS: /home/pi/ros2_crystal/install/fastrtps/include  
-- Found rclcpp: 0.6.4 (/home/pi/ros2_crystal/install/rclcpp/share/rclcpp/cmake)
-- Found rmw_implementation_cmake: 0.6.1 (/home/pi/ros2_crystal/install/rmw_implementation_cmake/share/rmw_implementation_cmake/cmake)
-- Found rmw_fastrtps_cpp: 0.6.2 (/home/pi/ros2_crystal/install/rmw_fastrtps_cpp/share/rmw_fastrtps_cpp/cmake)
-- Found px4_msgs: 2.0.1 (/home/pi/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/cmake)
-- Found PythonInterp: /usr/bin/python3 (found version "3.6.8") 
-- fastrtpsgen found in /home/pi/Fast-RTPS-Gen/scripts
java version "1.8.0_231"                          
Java(TM) SE Runtime Environment (build 1.8.0_231-b11)
Java HotSpot(TM) 64-Bit Server VM (build 25.231-b11, mixed mode)
-- fastrtpsgen version 1.0.0                      
-- Retrieving list of msgs to send...
-- List of msgs to send: CameraCapture, CameraTrigger, CollisionReport, DebugArray, DebugKeyValue, DebugValue, DebugVect, ObstacleDistance, OpticalFlow, PositionSetpoint, PositionSetpointTriplet, TrajectoryWaypoint, VehicleTrajectoryWaypoint, VehicleMocapOdometry, VehicleVisualOdometry
-- Retrieving list of msgs to receive...
-- List of msgs to receive: AdcReport, Airspeed, BatteryStatus, Cpuload, DistanceSensor, EstimatorStatus, HomePosition, IridiumsbdStatus, RadioStatus, SatelliteInfo, SensorBaro, SensorCombined, SensorSelection, VehicleAttitude, VehicleOdometry, VtolVehicleStatus, WindEstimate, CollisionConstraints
-- fastrtpsgen found in /home/pi/Fast-RTPS-Gen/scripts
-- px4_msgs message dir under /home/pi/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/msg
-- IDL definitions under /home/pi/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/msg
-- Configuring done                               
-- Generating done                                
-- Build files have been written to: /home/pi/px4_ros_com_ros2/build/px4_ros_com
[  1%] Generating micro-RTPS agent code...
Loading templates...                                           
Processing the file /home/pi/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/msg/IrlockReport.idl...
/home/pi/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/msg/IrlockReport.idl:30:9: error: Unexpected input '*'
[... 29 lines cut ...]
CMakeFiles/Makefile2:387: recipe for target 'CMakeFiles/micrortps_agent.dir/all' failed
Makefile:140: recipe for target 'all' failed                   
--- stderr: px4_ros_com
Exception in thread "main" java.lang.NullPointerException
	at com.eprosima.fastrtps.fastrtpsgen.execute(fastrtpsgen.java:336)
	at com.eprosima.fastrtps.fastrtpsgen.main(fastrtpsgen.java:1200)
Traceback (most recent call last):
  File "/home/pi/px4_ros_com_ros2/src/px4_ros_com/scripts/generate_microRTPS_bridge.py", line 491, in <module>
    generate_agent(agent_out_dir)
  File "/home/pi/px4_ros_com_ros2/src/px4_ros_com/scripts/generate_microRTPS_bridge.py", line 421, in generate_agent
    "/fastrtpsgen -example x64Linux2.6gcc " + fastrtpsgen_include + idl_file, shell=True)
[... cut ...] turned non-zero exit status 1.
make[2]: *** [/home/pi/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/microRTPS_agent.cpp] Error 1
make[2]: *** Deleting file '/home/pi/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/microRTPS_agent.cpp'
make[1]: *** [CMakeFiles/micrortps_agent.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< px4_ros_com	[ Exited with code 2 ]

Summary: 0 packages finished [2min 56s]
  1 package failed: px4_ros_com
  1 package had stderr output: px4_ros_com
```

TSC21, 2019-11-14:
For Fast-RTPS 1.7.2 you should build with `-DBUILD_JAVA=ON` and not use the Fast-RTPS-Gen repo. Just use the deployed *fastrtpsgen* with the 1.7.2 release branch.

MartinAtCoventry, 2019-11-14:
@TSC21 , my Fast-RTPS was compiled during building ROS2 crystal. I did not compile it independently. So, I did not set the option`-DBUILD_JAVA=ON` yet.

TSC21, 2019-11-14:
The instructions on the PX4 Devguide are quite clear regarding the required dependencies :)

MartinAtCoventry, 2019-11-14:
Ok, I think my ROS2 might be re-built, because the fast-rtps will change version on my board :(
Thank you for helping me :)