PX4/px4_ros_com #32: Cannot open a serial port under ROS2.

MartinAtCoventry, 2019-11-19:
**Describe the bug**
After successfully deployed `ROS2` and `px4_ros_com`, I cannot open a specified serial port. My system always return a error information.

Btw, I have set the access permission to all of serial port devices on my board:
sudo groups $USER
sudo usermod -a -G dialout $USER
Also, I have tested the serial port device on my board by `minicom`. It can communicate data with other serial port on another board each other.

**To Reproduce**
1. Method 1: Directly running the `micrortps_agent`.
./px4_ros_com_ros2/build/px4_ros_com/micrortps_agent -t UART -d /dev/ttyS4 -b 115200
The error information:
```
UART transport: device: /ttyS4; baudrate: 115200; sleep: 1us; poll: 0ms

Failed to open device: /ttyS4 (2)EXITING...
```

2. Method 2: Running the `micrortps_agent` by ROS2 after set ROS2 environment and PX4_ROS_COM environment.
source ~/ros2_crystal/install/local_setup.bash
source ~/px4_ros_com_ros2/install/local_setup.bash
printenv | grep ROS
```
ROS_VERSION=2
ROS_PYTHON_VERSION=3
ROS_DISTRO=crystal
```
ros2 run px4_ros_com micrortps_agent -t UART -d /dev/ttyS4 -b 115200
The error information:
```
No executable found
``` 

3. Methos 3: Directly running the `micrortps_agent` after set ROS2 environment and PX4_ROS_COM environment.
./px4_ros_com_ros2/build/px4_ros_com/micrortps_agent -t UART -d /dev/ttyS4 -b `115200`
The error information:
```
UART transport: device: /dev/ttyS4; baudrate: 115200; sleep: 1us; poll: 0ms

ERR SET BAUD /dev/ttyS4: Unsupported baudrate: 4098
	supported examples:
	9600, 19200, 38400, 57600, 115200, 230400, 460800, 500000, 921600, 1000000
Closed UART...EXITING...
```

./px4_ros_com_ros2/build/px4_ros_com/micrortps_agent -t UART -d /dev/ttyS4
The error information:
```
UART transport: device: /dev/ttyS4; baudrate: 460800; sleep: 1us; poll: 0ms

ERR SET BAUD /dev/ttyS4: Unsupported baudrate: 4100
	supported examples:
	9600, 19200, 38400, 57600, 115200, 230400, 460800, 500000, 921600, 1000000
Closed UART...EXITING...
```

**Expected behavior**
I have attempted to execute the `micrortps_agent` by the above methods. The serial port device cannot be opened. Especially, in the method 3 the program return information that told me the baudrate is set at 4098 and 4100, but the baudrate parameters I transmit to the `micrortps_agent` are 115200 and 460800 separately.
So, I don't know what's wrong I did.

Thanks in advance !

MartinAtCoventry, 2019-11-19:
Similarly, the similar error is happened on the flight control unit.
Running the `micrortps_client` process by command after I started the `PX4 flight stack` on my beaglebone board.
```
debian@BBBluePX4:~/PX4$ sudo ./px4 -s px4.config 
INFO  [px4] assuming working directory is rootfs, no symlinks needed.

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh px4.config 0
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
INFO  [bbblue_adc] Initializing librobotcontrol ...
705599 BMP280 sensor configuration succeeded
INFO  [df_bmp280_wrapper] BMP280 started
939237 Reset MPU9250
1021777 MPU9250 Reg: USER_CTRL:  0x60, FIFO EN:  0xF9, FIFO count: 88
1030402 IMU temperature initialized to: 52.781162
INFO  [bbblue_adc] BBBlueADC started
INFO  [bbblue_adc] ADC channel: 0; value: 3624
INFO  [bbblue_adc] ADC channel: 1; value: 3095
INFO  [bbblue_adc] ADC channel: 2; value: 3055
INFO  [bbblue_adc] ADC channel: 3; value: 1245
INFO  [bbblue_adc] ADC channel: 6; value: 1648, voltage:   7.97 V
INFO  [bbblue_adc] ADC channel: 5; value: 2545, voltage:  12.31 V
WARN  [commander] LED: getHandle fail

WARN  [commander] LED init failed
INFO  [mavlink] mode: Normal, data rate: 2800 B/s on /dev/ttyO5 @ 57600B
INFO  [linux_sbus] Device /dev/ttyS4 , channels: 12 

INFO  [linux_pwm_out] Loaded mixer from file ROMFS/px4fmu_common/mixers/quad_x.main.mix
INFO  [linux_pwm_out] Starting PWM output in bbblue_rc mode
INFO  [logger] logger started (mode=all)
INFO  [px4] Startup script returned successfully
pxh> INFO  [mavlink] partner IP: 0.0.0.0
INFO  [ecl/EKF] EKF aligned, (pressure height, IMU buf: 22, OBS buf: 14)

pxh>
```

And then started the `micrortps_client`, it will get a error information which is quite as same as the `micrortps_agent`'s on my offboard:
```
pxh> micrortps_client start -t UART -d /dev/ttyS1 -b 115200
INFO  [micrortps_client] UART transport: device: /dev/ttyS1; baudrate: 115200; sleep: 1ms; poll: 1ms
pxh> ERROR [micrortps_client] ERR SET BAUD /dev/ttyS1: Unsupported baudrate: 4098
	supported examples:
	9600, 19200, 38400, 57600, 115200, 230400, 460800, 500000, 921600, 1000000

WARN  [micrortps_client] Closed UART...
INFO  [micrortps_client] EXITING...

pxh>
```

MartinAtCoventry, 2019-11-19:
It seems that the function `baudtype getbaudrate()` cannot return a correct value in both of the subprojects(`PX4/Firmwire` & `PX4/px4_ros_com`).

@TSC21 , I think the function `baudtype getbaudrate(char *valstr)`, which is in /px4_ros_com/templates/microRTPS_agent.cpp.em, has a bug. The function `unsigned long int strtoul()` of the stand C library should be paid attention to it.

@Daniel Agar & @Beat Küng, please check the usage of the 
 `long int strtol()` in the function `int px4_get_parameter_value()`, which is located 
 in Firmware/src/platforms/common/px4_cli.cpp

TSC21, 2019-11-19:
@MartinAtCoventry the commit that updated the setup of the baudrate speed is https://github.com/PX4/Firmware/commit/6bb0046407917326ab067a33af42fdd816079510. Seems like something is failing in that process. I have been working mostly with SITL lately but should be back to real HW soon.

In the mean time, @gamoreno can you please have a look at this? I suppose when you made the change this was working for you.

MartinAtCoventry, 2019-11-19:
OK, thanks for your reply.
Could you tell me whether the fold `micrortps_agent` and its all of files in it are generated from the file `/px4_ros_com/templates/microRTPS_agent.cpp.em` during building the px4_ros_com package ?

TSC21, 2019-11-19:
> @TSC21 OK, thanks for your reply.
> Could you tell me whether the fold `micrortps_agent` and its all of files in it are generated from the file `/px4_ros_com/templates/microRTPS_agent.cpp.em` during building the px4_ros_com package ?

I can but I don't think that's relevant here (`.em` extensions are EmPy related and used to generate the code for the bridge, both for the agent and client).

What probably needs a review is https://github.com/PX4/px4_ros_com/blob/master/templates/microRTPS_agent.cpp.em#L142-L149. That would be as simple as adding some print statements and check what exactly it's being parsed from the command line args.

MartinAtCoventry, 2019-11-19:
OK, I will study and test it.

gamoreno, 2019-11-19:
> @MartinAtCoventry the commit that updated the setup of the baudrate speed is [PX4/Firmware@6bb0046](https://github.com/PX4/Firmware/commit/6bb0046407917326ab067a33af42fdd816079510). Seems like something is failing in that process. I have been working mostly with SITL lately but should be back to real HW soon.
> 
> In the mean time, @gamoreno can you please have a look at this? I suppose when you made the change this was working for you.

Yes, this used to work. I think the issue is that now the conversion is being done twice. First, here: https://github.com/PX4/px4_ros_com/blob/08fd40d6dd83f8048adcad26d053497808da7750/templates/microRTPS_agent.cpp.em#L164
and passing it to UART_Node here:
https://github.com/PX4/px4_ros_com/blob/08fd40d6dd83f8048adcad26d053497808da7750/templates/microRTPS_agent.cpp.em#L245
Note that this is passing the value of a constant like B115200, which in Linux it doesn't have the value 115200. 
This first conversion didn't exist when I made that previous fix.

Then, there is a second conversion invoked here:
https://github.com/PX4/px4_ros_com/blob/08fd40d6dd83f8048adcad26d053497808da7750/templates/microRTPS_transport.cpp#L339
The function `baudrate_to_speed()` expects a baud rate number like 115200, but is now receiving the value of the constant B115200.

TSC21, 2019-11-19:
> Yes, this used to work. I think the issue is that now the conversion is being done twice. First, here: [...]

@gamoreno https://github.com/PX4/px4_ros_com/blob/08fd40d6dd83f8048adcad26d053497808da7750/templates/microRTPS_agent.cpp.em#L164 was way there before: https://github.com/PX4/Firmware/commit/83133b1bcae2eae59383dad7d1b6070657e9d279. So when you added your change, this was already there. How exactly did you test this?

TSC21, 2019-11-19:
@gamoreno actually you are right: https://github.com/PX4/Firmware/commit/83133b1bcae2eae59383dad7d1b6070657e9d279 broke the usage of this. @drisebor can you please explain what was done to test this?

TSC21, 2019-11-19:
https://github.com/PX4/Firmware/issues/8808 is the related issue. This needs to be fixed and made uniform.

TSC21, 2019-11-19:
@gamoreno https://github.com/PX4/Firmware/issues/8808 reports a problem with your approach somehow.

gamoreno, 2019-11-19:
> @gamoreno [PX4/Firmware#8808](https://github.com/PX4/Firmware/issues/8808) reports a problem with your approach somehow.

I think the reason [PX4/Firmware#8808](https://github.com/PX4/Firmware/issues/8808) was an issue is that for some reason, at the time, my changes to https://github.com/PX4/Firmware/blob/master/msg/templates/urtps/microRTPS_transport.cpp were not there anymore. The history shows them in a commit from 2019, even though I fixed this in 2017. Not sure how to interpret that.

TSC21, 2019-11-19:
> I think the reason [PX4/Firmware#8808](https://github.com/PX4/Firmware/issues/8808) was an issue is that for some reason, at the time, my changes to [...]

Yeah I suppose something went wrong there. Anyway, I reverted that commit and left yours intact in https://github.com/PX4/Firmware/pull/13525

MartinAtCoventry, 2019-11-19:
@TSC21 , @gamoreno , Hi all ! I viewed your discussion. Thanks both of you for solving my issue.

I have found the problem point. In PX4/px4_ros_com side, the function `int main(int argc, char** argv)` transport a wrong type of parameter to the function `UART_node()` (px4_ros_com/templates/microRTPS_agent.cpp.em, Line245). 
The 2nd parameter of the function is assigned a `speed_t` type of variable, which is `_options.baudrate.code`. However, the definition of the function `UART_node();` is shown as follow:
```
UART_node::UART_node(const char *_uart_name, uint32_t _baudrate, uint32_t _poll_ms):
	uart_fd(-1),
	baudrate(_baudrate),
	poll_ms(_poll_ms)
```
You can find the function in the file `microRTPS_transport.cpp`. Obviously, the 2nd parameter of the function should be assigned an `uint32_t` type of variable, which should be `_options.baudrate.val`.
Other source code are perfect.

Btw, I have separately tested the `micrortps_client` and `micrortps_agent` after edited the function in the `int main()` and re-compiled them. Now, in the both sides (Firmware & px4_ros_com), micrortps_client / agent can set and open serial port device on my boards correctly.

So, I think you just fix the following files for these two projects:
px4_ros_com/templates/microRTPS_agent.cpp.em
Firmware/src/modules/micrortps_bridge/micrortps_client/microRTPS_client_main.cpp
Firmware/msg/templates/urtps/microRTPS_agent.cpp.em

I will continue to test the `micrortps_bridge` between my two boards.

TSC21, 2019-11-19:
https://github.com/PX4/Firmware/pull/13525 I already pushed a fix. Please test it

MartinAtCoventry, 2019-11-19:
@TSC21 , don't need fix other source code file !!!
You just need to fix the three file :
px4_ros_com/templates/microRTPS_agent.cpp.em (line245)
Firmware/src/modules/micrortps_bridge/micrortps_client/microRTPS_client_main.cpp (line165)
Firmware/msg/templates/urtps/microRTPS_agent.cpp.em (line245)

Change `_options.baudrate.code` to `_options.baudrate.val`.

MartinAtCoventry, 2019-11-19:
I will test the communication functionality of `micrortps_bridge` on my real HW tomorrow.

TSC21, 2019-11-19:
> @TSC21 , don't need to fix other source code files !!!
> You just need to fix the three files :
> px4_ros_com/templates/microRTPS_agent.cpp.em (line245)
> Firmware/src/modules/micrortps_bridge/micrortps_client/microRTPS_client_main.cpp (line165)
> Firmware/msg/templates/urtps/microRTPS_agent.cpp.em (line245)
> 
> Change `_options.baudrate.code` to `_options.baudrate.val`.

Yes its needed because the code logic is overall duplicated

MartinAtCoventry, 2019-11-20:
Hi @TSC21 , I am testing the `micrortps_bridge` between my beaglebone(as flight control platform) and RPI4(as offboard platform) by serial port device.

In `PX4_ROS_COM` side, I deleted my old px4_ros_com source code package, and gitted a new px4_ros_com package again. And then, I edited the file https://github.com/PX4/px4_ros_com/blob/master/templates/microRTPS_agent.cpp.em#L245, which just changes `_options.baudrate.code` to `_options.baudrate.val`. Re-building the package according your guidance documentary: https://dev.px4.io/master/en/middleware/micrortps.html

In `PX4/Firmware` side, I just edited the two code files:
https://github.com/PX4/Firmware/blob/master/src/modules/micrortps_bridge/micrortps_client/microRTPS_client_main.cpp#L165
https://github.com/PX4/Firmware/blob/master/msg/templates/urtps/microRTPS_agent.cpp.em#L245
And then, cross-compiled the `Firmware` project for my beaglebone board as well as the new `micrortps_agent` source code package was generated automatically during the compiling process.

Now, on my flight control platform(beaglebone), I add a line command into the configuration file `px4.config`: `micrortps_client start -t UART -d /ttyS1 -b 115200` in order to the `px4 autopilot stack` can start the `micrortps_client` daemon automatically.

So, the experimentation operation steps as below:
1. On flight control platform, start the `px4 autopilot stack`.
```
debian@BBBluePX4:~/PX4$ sudo ./px4 -s px4.config 
INFO  [px4] assuming working directory is rootfs, no symlinks needed.

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh px4.config 0
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
INFO  [bbblue_adc] Initializing librobotcontrol ...
643524 BMP280 sensor configuration succeeded
INFO  [df_bmp280_wrapper] BMP280 started
879552 Reset MPU9250
961905 MPU9250 Reg: USER_CTRL:  0x60, FIFO EN:  0xF9, FIFO count: 88
969569 IMU temperature initialized to: 53.656509
INFO  [bbblue_adc] BBBlueADC started
INFO  [bbblue_adc] ADC channel: 0; value: 3647
INFO  [bbblue_adc] ADC channel: 1; value: 2979
INFO  [bbblue_adc] ADC channel: 2; value: 3010
INFO  [bbblue_adc] ADC channel: 3; value: 1221
INFO  [bbblue_adc] ADC channel: 6; value: 1653, voltage:   7.99 V
INFO  [bbblue_adc] ADC channel: 5; value: 2561, voltage:  12.38 V
WARN  [commander] LED: getHandle fail

WARN  [commander] LED init failed
INFO  [mavlink] mode: Normal, data rate: 2800 B/s on /dev/ttyO5 @ 57600B
INFO  [linux_sbus] Device /dev/ttyS4 , channels: 12 

INFO  [linux_pwm_out] Loaded mixer from file ROMFS/px4fmu_common/mixers/quad_x.main.mix
INFO  [linux_pwm_out] Starting PWM output in bbblue_rc mode
INFO  [logger] logger started (mode=all)
INFO  [micrortps_client] UART transport: device: /dev/ttyS1; baudrate: 115200; sleep: 1ms; poll: 1ms
INFO  [micrortps_client] No flush
INFO  [px4] Startup script returned successfully
pxh> INFO  [mavlink] partner IP: 0.0.0.0
INFO  [ecl/EKF] EKF aligned, (pressure height, IMU buf: 22, OBS buf: 14)

pxh>
```
You can see the information : `[micrortps_client] UART transport: device: /dev/ttyS1; baudrate: 115200; sleep: 1ms; poll: 1ms` and `[micrortps_client] No flush`.

2. On offboard platform, starting the `micrortps_agent` daemon after setting ROS2 and PX4_ROS_COM environment.
source ~/ros2_crystal/install/local_setup.bash
source ~/px4_ros_com_ros2/install/local_setup.bash
```
pi@RPI4:~$ ./px4_ros_com_ros2/build/px4_ros_com/micrortps_agent -t UART -d /dev/ttyS4 -b 115200

UART transport: device: /dev/ttyS4; baudrate: 115200; sleep: 1us; poll: 0ms

No flushCameraCapture subscriber started
CameraTrigger subscriber started
[... 16 lines cut ...]
Cpuload publisher started
DistanceSensor publisher started
EstimatorStatus publisher started
HomePosition publisher started
IridiumsbdStatus publisher started
RadioStatus publisher started
SatelliteInfo publisher started
SensorBaro publisher started
SensorCombined publisher started
SensorSelection publisher started
VehicleAttitude publisher started
VehicleOdometry publisher started
VtolVehicleStatus publisher started
WindEstimate publisher started
CollisionConstraints publisher started
Bad CRC 49323 != 60946                                 (↓ 95)Bad CRC 64681 != 9327                                 (↓ 288)Bad CRC 65339 != 39346                                 (↓ 288)Bad CRC 26055 != 63977                                 (↓ 95)Bad CRC 19359 != 55564                                 (↓ 288)Bad CRC 16217 != 65095                                 (↓ 95)Bad CRC 16254 != 38974                                 (↓ 288)Bad CRC 11861 != 64783                                 (↓ 288)Bad CRC 23387 != 33052                                 (↓ 288)Bad CRC 48332 != 1051                                 (↓ 288)Bad CRC 11955 != 58279                                 (↓ 95)Bad CRC 11139 != 55201                                 (↓ 288)Bad CRC 32450 != 24080                                 (↓ 95)Bad CRC 17616 != 11502                                 (↓ 288)Bad CRC 30001 != 17140                                 (↓ 95)Bad CRC 32462 != 53236                                 (↓ 95)Bad CRC 40917 != 769                                 (↓ 95)Bad CRC 18901 != 8316                                 (↓ 288)Bad CRC 24135 != 7915                                 (↓ 95)Bad CRC 11184 != 56649               [... cut ...] = 42394                                 (↓ 62)
SENT:     0 messages - 0 bytes
RECEIVED: 11 messages - 451 bytes; 886549 LOOPS - 0.002 seconds - 258.61KB/s
```

**Describe the bug**
On the offboard side, there are many `Bad CRC` notes after starting micrortps_agent. Also, the micrortps_agent didn't receive new data from the flight control board after received 11 messages once.

I want to know what value other parameters should be assigned for `micrortps_client / agent`.
e.g. `-u`,`-l`,`-w`, `-p`.

TSC21, 2019-11-20:
@MartinAtCoventry first, and to close the main issue reported here: I tested the bridge with the changes in PX4/Firmware#13525 and it fixes it.

About the CRC issues: what you are reporting is completely normal considering that in the same UART link, you are sending both Mavlink and RTPS packets. As there's no protocol splitter on the agent side, the amount of CRC errors will be large because the agent is trying to deserialize Mavlink messages as they were RTPS messages, which will endup on the link to fail.

While a protocol splitter (for the agent, since in the PX4/client side it already exists) is still on the works, what you need to do is to stop the Mavlink stream instance you have on the port you are connecting and just launch the `micrortps_client` for that same port. If you need both Mavlink and RTPS on your Beaglebone, you need **2 separate UART connections**, one for the Mavlink stream and another for the RTPS stream.

Even if you get some CRC errors on the RTPS only stream (which are normal, depending on the stability of the link), you will see that the link works as expected and you can subscribe to the RTPS topics (ex: `ros2 topic echo /VehicleOdometry_PubSubTopic`).

I am closing this issue since the initial reported issue was fixed. If you have questions about the second part (CRC and dual link), please bring those to the PX4 Slack #messaging or #ros channel.

closed 2019-11-20 by TSC21

MartinAtCoventry, 2019-11-20:
Thanks @TSC21 :) I have cross-compiled and built `Firmware`(v1.10.0-rc1) and `px4_ros_com` again. My boards can communicate each other well.
Btw, on my beaglebone, the serial port `ttyO1` is used to micrortps bridge and `ttyO5` for mavlink.
I'm going to continue study the usage of micrortps_bridge with ROS2, and surf on PX4 Slack.