PX4/px4_ros_com #33: Compilation error: No IDL files found

jocacace, 2020-01-06:
Hi to all,

I am addressing a compilation error compiling the px4_ros_com package. Seems that the compiler is not able to find the idl message files, as you can see from the error:

`File "/home/jcacace/px4_ros_com_ros2/src/px4_ros_com/scripts/generate_microRTPS_bridge.py", line 425, in generate_agent
    raise Exception("No IDL files found in %s" % idl_dir)
Exception: No IDL files found in /home/jcacace/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/msg/dds_fastrtps`

I have compiled the px4_msgs and the client part of the fastrtps of the px4 firmware. Actually, I noticed that some idl messages are present in the upper level directory (the dds_fastrtps doesn't exist). Since I follow the documentation to use ROS2 to exchange messages with the px4: https://dev.px4.io/master/en/middleware/micrortps.html, I am quite surprised to have different message locations. Any other user experienced the same problem? Is there any possibility to specify a different location for the idl messages?

Thanks

TSC21, 2020-01-06:
@jocacace what exactly steps are you following to build the code. A reminder that you should have Fast-RTPS 1.7.2 built and installed from source, and you need to use ROS2 Crystal, as we are still working on adding support to ROS2 Dashing.

jocacace, 2020-01-06:
@TSC21  Thank you so much for your very fast reply. 
Ok...I have the correct version of Fast-RTPS... but actually I am using ROS2 Dashing... Hopefully this is the problem. I will try to follow again the guide in interfacing ROS/ROS2/PX4 in order to close this issue.

TSC21, 2020-01-06:
> Ok...I have the correct version of Fast-RTPS... but actually I am using ROS2 Dashing... Hopefully this is the problem. I will try to follow again the guide in interfacing ROS/ROS2/PX4 in order to close this issue.

Just uninstall Dashing and install Crystal instead and that should solve your issue.

deb0ch, 2020-01-06:
I am currently running into this issue as well trying to build with ROS2 Eloquent. I actually need ROS2 Eloquent, can you please point me somewhere to start fixing this ?

Would a contribution be welcomed if I come up with a good enough fix ?

TSC21, 2020-01-06:
@deb0ch as ROS2 Dashing is not yet supported, by consequence ROS2 Eloquent is not supported as well.

> Would a contribution be welcomed if I come up with a good enough fix ?

Sure, but we are already on track to solve this, so I would wait for a bit.

TSC21, 2020-01-06:
@deb0ch if you are interested, the solution has been discussed in https://github.com/eProsima/Fast-RTPS/issues/829.

deb0ch, 2020-01-06:
I am using FastRTPS 1.7.2 that I installed following PX4/Firmware docs, and manually executed all the steps listed in `px4_ros_com/scripts/build_ros2_workspace.bash` and `px4_ros_com/scripts/setup_system.bash`. The scripts need fixing btw, I could PR the fixes that I made for myself as well

deb0ch, 2020-01-06:
> @deb0ch if you are interested, the solution has been discussed in [eProsima/Fast-RTPS#829](https://github.com/eProsima/Fast-RTPS/issues/829).

Thank you so much for your swift reply, I will go dig into it right away ! 😺

TSC21, 2020-01-06:
> I am using FastRTPS 1.7.2

The you should not be using Dashing or Eloquent. 

> The scripts need fixing btw, I could PR the fixes that I made for myself as well

Why do the scripts need fixing?

TSC21, 2020-01-06:
@deb0ch what's actually missing is an update on the docs stating we currently do not support ROS2 Dashing or Eloquent and that users should for now stay with ROS2 Crystal and Fast-RTPS 1.7.2. The problem is related with the type naming that's generated by `fastrtpsgen` vs the type naming generated by `rmw_fastrtps` on ROS2.

TSC21, 2020-01-06:
> manually executed all the steps listed in `px4_ros_com/scripts/build_ros2_workspace.bash` and `px4_ros_com/scripts/setup_system.bash`

If you have your system correctly setup, you don't need to manually run the commands on the scripts.

deb0ch, 2020-01-06:
> > I am using FastRTPS 1.7.2
> 
> The you should not be using Dashing or Eloquent.
> 
> > The scripts need fixing btw, I could PR the fixes that I made for myself as well
> 
> Why do the scripts need fixing?

For the `build_ros2_workspace.bash`, the script doesn't work if you run it from outside of its directory and the `--no_ros1_bridge` option needs a random argument or else it is not taken into account.

For the `setup_system.bash`, the one that fixes the python3 `gencpp` and `genmsg` issue, there are some syntax errors (missing `\` for splitting commands on multiple lines), invalid arguments (`apt-get --qq` produces an error) and some `$ROS_DISTRO` variables that should seemingly be `$ROS1_DISTRO`.

> @deb0ch what's actually missing is an update on the docs stating we currently do not support ROS2 Dashing or Eloquent and that users should for now stay with ROS2 Crystal and Fast-RTPS 1.7.2. The [...]

I just noticed these two lines in `build_ros2_workspace.bash`:
```shell
# workaround for rosidl fix in Dashing, while backport isn't applied from Eloquent
# source ~/PX4/rosidl_ws/install/setup.bash
```

Is it related to eProsima/Fast-RTPS#829 ?

TSC21, 2020-01-06:
> For the `build_ros2_workspace.bash`, the script doesn't work if you run it from outside of its directory and the `--no_ros1_bridge` option needs a random argument or else it is not taken into [...]

Alright makes sense. I don't use the `--no_ros1_bridge` for a while now and I don't mind running the script from the scripts folder ;) Same thing with the `setup_system.bash`. PR is welcomed.

> I just noticed these two lines in `build_ros2_workspace.bash`:
>
> ```shell
> # workaround for rosidl fix in Dashing, while backport isn't applied from Eloquent
> # source [...]

No that's something else, something that remained from another thing. I would clean that myself.

jocacace, 2020-01-08:
Hi,

Now my setup should follow the compilation guidelines: FastRTPS version 1.7.2, ROS1: melodic and ROS2: crystal. 

The compilation of px4 firmware with rtps works well. Sadly when I try to compile px4_msgs I got several errors of the following form:

`In file included from /home/prisma/px4_ros_com_ros2/build/px4_msgs/rosidl_typesupport_opensplice_cpp/px4_msgs/msg/dds_opensplice/actuator_armed__type_support.cpp:4:0:
/home/prisma/px4_ros_com_ros2/build/px4_msgs/rosidl_typesupport_opensplice_cpp/px4_msgs/msg/actuator_armed__rosidl_typesupport_opensplice_cpp.hpp:40:18: error: ‘px4_msgs::msg::dds_’ has not been declared
   px4_msgs::msg::dds_::ActuatorArmed_ & dds_message);
`
Any suggestion?

TSC21, 2020-02-11:
@jocacace we are currently actively supporting ROS2 Dashing and Eloquent. I advise you to update your system, as we Crystal is EOL. Closing this issue.

closed 2020-02-11 by TSC21