PX4/px4_ros_com #37: Failure building ros2 workspace

Pman6543, 2020-02-14:
[image]
Whenever I try to build the ros1 and ros2 workspaces. The ros1 workspace succeeds, but the ros2 workspace fails.

I followed the instructions on the dev guide, but to no avail. 

At first I tried

source build_all.bash --ros1_ws_dir
The ros1 workspace would complete, but the ros2 build terminal would crash with no output, so I opted to try and build them manually.

In calling this command
colcon build --symlink-install --packages-skip ros1_bridge --event-handlers console_direct+
I get the above error.

TSC21, 2020-02-14:
@Pman6543 have you tried using the build script provided in https://github.com/PX4/px4_ros_com/blob/master/scripts/build_all.bash. It's there for easy the build process. And it seems above that you don't have `rosidl-generator-dds-idl` installed. `apt install ros-crystal-rosidl-generator-dds-idl`.

Pman6543, 2020-02-14:
@TSC21 

`apt install ros-crystal-rosidl-generator-dds-idl`

This got me a little bit further. I have been attempting to use the build_all script, but it always caused my ros2 workspace terminal to crash. I'm further than I was before now px4_msgs seems to build, but px4_ros_com is failing. 

Generating makefile solution
INFO: /home/nrivera/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/fastrtpsgen/makefile_x64Linux2.6gcc exists. Skipping.
Loading templates...
Processing the file /home/nrivera/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/msg/IrlockReport.idl...
^[[1m/home/nrivera/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/msg/IrlockReport.idl:30:9: ^[[0m^[[1;31merror: ^[[0mUnexpected input ^[[1m'*'^[[0m
^[[1m/home/nrivera/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/msg/IrlockReport.idl:30:12: ^[[0m^[[1;31merror: ^[[0m^[[1m'size' ^[[0mwas not defined previously
^[[1m/home/nrivera/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/msg/IrlockReport.idl:30:20: ^[[0m^[[1;31merror: ^[[0mmismatched input 'target' expecting {';', ','}
^[[1m/home/nrivera/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/msg/IrlockReport.idl:30:20: ^[[0m^[[1;31merror: ^[[0m^[[1m'target' ^[[0mwas not defined previously
^[[1m/home/nrivera/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/msg/IrlockReport.idl:30:33: ^[[0m^[[1;31merror: ^[[0mmismatched input 'camera' expecting {';', ','}
^[[1m/home/nrivera/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/msg/IrlockReport.idl:30:33: ^[[0m^[[1;31merror: ^[[0m^[[1m'camera' ^[[0mwas not defined previously
^[[1m/home/nrivera/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/msg/IrlockReport.idl:30:41: ^[[0m^[[1;31merror: ^[[0mmismatched input '-' expecting {';', ','}
^[[1m/home/nrivera/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/msg/IrlockReport.idl:30:47: ^[[0m^[[1;31merror: ^[[0mUnexpected input ^[[1m'in'^[[0m
^[[1m/home/nrivera/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/msg/IrlockReport.idl:30:42: ^[[0m^[[1;31merror: ^[[0m^[[1m'axis' ^[[0mwas not defined previously
^[[1m/home/nrivera/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/msg/IrlockReport.idl:30:56: ^[[0m^[[1;31merror: ^[[0mmismatched input 'of' expecting {';', ','}
^[[1m/home/nrivera/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/msg/IrlockReport.idl:30:56: ^[[0m^[[1;31merror: ^[[0m^[[1m'of' ^[[0mwas not defined previously
^[[1m/home/nrivera/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/msg/IrlockReport.idl:30:62: ^[[0m^[[1;31merror: ^[[0mmismatched input '(' expecting {';', ','}
^[[1m/home/nrivera/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/msg/IrlockReport.idl:30:68: ^[[0m^[[1;31merror: ^[[0mmismatched input ')' expecting {'::', ID}
^[[1;31mERROR<Exception>: ^[[0mnull
All good. RTPS ID's are unique
CMakeFiles/micrortps_agent.dir/build.make:61: recipe for target '/home/nrivera/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/microRTPS_agent.cpp' failed
CMakeFiles/Makefile2:419: recipe for target 'CMakeFiles/micrortps_agent.dir/all' failed
Makefile:140: recipe for target 'all' failed

Summary: 1 package finished [1min 15s]
  1 package failed: px4_ros_com
  1 package had stderr output: px4_ros_com

TSC21, 2020-02-14:
hmm never saw similar error before. What is the output of `fastrtpsgen -version`?

Pman6543, 2020-02-14:
openjdk version "1.8.0_242"
OpenJDK Runtime Environment (build 1.8.0_242-8u242-b08-0ubuntu3~18.04-b08)
OpenJDK 64-Bit Server VM (build 25.242-b08, mixed mode)
fastrtpsgen version 1.0.2

TSC21, 2020-02-14:
> openjdk version "1.8.0_242"
> OpenJDK Runtime Environment (build 1.8.0_242-8u242-b08-0ubuntu3~18.04-b08)
> OpenJDK 64-Bit Server VM (build 25.242-b08, mixed mode)
> fastrtpsgen version 1.0.2

You have to downgrade the `fastrtpsgen` version. ROS2 Crystal is compliant with Fast-RTPS version 1.7.2, which you have to compile from source so to have access to `fastrtpsgen` version 1.7.2 (note that 1.0.2 is Fast-RTPS-Gen latest version, detached from the main Fast-RTPS repo).

TSC21, 2020-02-14:
BTW, this is probably not a reasonable work to do since we do already support ROS2 Dashing and Eloquent. My advice is to install ROS2 Dashing and use it instead of Crystal (which is already EOL).

Pman6543, 2020-02-14:
I was having other issues with FastRTPS 1.7.2 that wouldn't let me build the micrortps_agent using it. I'll try it again. 

Will I be able to use a newer version of FastRTPS with ROS2 Dashing?

TSC21, 2020-02-14:
> Will I be able to use a newer version of FastRTPS with ROS2 Dashing?

Yes, answered above.

Pman6543, 2020-02-14:
I followed the instructions for downloading fastrtps 1.8.2 on the px4 dev

Scanning dependencies of target micrortps_agent
[  0%] Building CXX object CMakeFiles/micrortps_agent.dir/RtpsTopics.cpp.o
[  1%] Building CXX object CMakeFiles/micrortps_agent.dir/adc_report.cpp.o
[  2%] Building CXX object CMakeFiles/micrortps_agent.dir/adc_reportPubSubTypes.cpp.o
[  3%] Building CXX object CMakeFiles/micrortps_agent.dir/adc_report_Publisher.cpp.o
/home/nrivera/src/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/adc_report_Publisher.cpp: In member function ‘bool adc_report_Publisher::init()’:
/home/nrivera/src/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/adc_report_Publisher.cpp:62:25: error: ‘class eprosima::fastrtps::rtps::BuiltinAttributes’ has no member named ‘discovery_config’
     PParam.rtps.builtin.discovery_config.leaseDuration = c_TimeInfinite;
                         ^~~~~~~~~~~~~~~~
CMakeFiles/micrortps_agent.dir/build.make:134: recipe for target 'CMakeFiles/micrortps_agent.dir/adc_report_Publisher.cpp.o' failed
make[2]: *** [CMakeFiles/micrortps_agent.dir/adc_report_Publisher.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/micrortps_agent.dir/all' failed
make[1]: *** [CMakeFiles/micrortps_agent.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

This was what I am getting

Pman6543, 2020-02-14:
This was solved by me following the instructions from the FastRTPS github page

https://github.com/eProsima/Fast-RTPS

Pman6543, 2020-02-14:
After following the above instructions and installing Ros2 dashing this is my current output. 

Exception in thread "main" java.lang.NullPointerException
	at com.eprosima.fastrtps.fastrtpsgen.execute(fastrtpsgen.java:336)
	at com.eprosima.fastrtps.fastrtpsgen.main(fastrtpsgen.java:1200)
Traceback (most recent call last):
  File "/home/nrivera/px4_ros_com_ros2/src/px4_ros_com/scripts/generate_microRTPS_bridge.py", line 474, in <module>
    generate_agent(agent_out_dir)
  File "/home/nrivera/px4_ros_com_ros2/src/px4_ros_com/scripts/generate_microRTPS_bridge.py", line 404, in generate_agent
    "/fastrtpsgen -example x64Linux2.6gcc " + fastrtpsgen_include + idl_file, shell=True)
  File "/usr/lib/python3.6/subprocess.py", line 311, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '/usr/local/bin/fastrtpsgen -d /home/nrivera/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/fastrtpsgen -example x64Linux2.6gcc -I /home/nrivera/px4_ros_com_ros2/install/px4_msgs/share /home/nrivera/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/msg/IrlockReport.idl' returned non-zero exit status 1.
make[2]: *** [/home/nrivera/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/microRTPS_agent.cpp] Error 1
make[2]: *** Deleting file '/home/nrivera/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/microRTPS_agent.cpp'
make[1]: *** [CMakeFiles/micrortps_agent.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< px4_ros_com	[ Exited with code 2 ]

Summary: 1 package finished [1min 12s]
  1 package failed: px4_ros_com
  1 package had stderr output: px4_ros_com

TSC21, 2020-02-14:
How did you install `Fast-RTPS-Gen`?

Pman6543, 2020-02-14:
I followed these instructions. 

https://dev.px4.io/master/en/setup/fast-rtps-installation.html

Then these instructions 

https://eprosima-fast-rtps.readthedocs.io/en/latest/geninfo.html

Both yielded the same result

TSC21, 2020-02-14:
Make sure you clear the previously generated `micrortps_agent` source files before trying to build again. Use the `scrtips/clean_all.bash` script.

Pman6543, 2020-02-14:
I cleaned it, and this was my result. 
```
Exception in thread "main" java.lang.NullPointerException
	at com.eprosima.fastrtps.fastrtpsgen.execute(fastrtpsgen.java:336)
	at com.eprosima.fastrtps.fastrtpsgen.main(fastrtpsgen.java:1200)
Traceback (most recent call last):
  File "/home/nrivera/px4_ros_com_ros2/src/px4_ros_com/scripts/generate_microRTPS_bridge.py", line 474, in <module>
    generate_agent(agent_out_dir)
  File "/home/nrivera/px4_ros_com_ros2/src/px4_ros_com/scripts/generate_microRTPS_bridge.py", line 404, in generate_agent
    "/fastrtpsgen -example x64Linux2.6gcc " + fastrtpsgen_include + idl_file, shell=True)
  File "/usr/lib/python3.6/subprocess.py", line 311, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '/usr/local/bin/fastrtpsgen -d /home/nrivera/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/fastrtpsgen -example x64Linux2.6gcc -I /home/nrivera/px4_ros_com_ros2/install/px4_msgs/share /home/nrivera/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/msg/IrlockReport.idl' returned non-zero exit status 1.
make[2]: *** [/home/nrivera/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/microRTPS_agent.cpp] Error 1
make[2]: *** Deleting file '/home/nrivera/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/microRTPS_agent.cpp'
make[1]: *** [CMakeFiles/micrortps_agent.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< px4_ros_com	[ Exited with code 2 ]

Summary: 1 package finished [3min 4s]
  1 package failed: px4_ros_com
  1 package had stderr output: px4_ros_com

```

HairBallChunk, 2020-03-09:
I have a similar issue with the same build command and also with the build_all.bash script. It will build the px4_msgs, but at 8% of the px4_ros_com package it fails.

`[3.431s] openjdk version "1.8.0_242"
[3.431s] OpenJDK Runtime Environment (build 1.8.0_242-8u242-b08-0ubuntu3~18.04-b08)
[3.432s] OpenJDK 64-Bit Server VM (build 25.242-b08, mixed mode)
[4.440s] openjdk version "1.8.0_242"
[4.441s] OpenJDK Runtime Environment (build 1.8.0_242-8u242-b08-0ubuntu3~18.04-b08)
[4.442s] OpenJDK 64-Bit Server VM (build 25.242-b08, mixed mode)
[8.938s] Traceback (most recent call last):
[8.938s]   File "/home/michiel/px4_ros_com_ros2/src/px4_ros_com/scripts/generate_microRTPS_bridge.py", line 474, in <module>
[8.939s]     generate_agent(agent_out_dir)
[8.939s]   File "/home/michiel/px4_ros_com_ros2/src/px4_ros_com/scripts/generate_microRTPS_bridge.py", line 400, in generate_agent
[8.939s]     raise Exception("No IDL files found in %s" % idl_dir)
[8.940s] Exception: No IDL files found in /home/michiel/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/msg/dds_fastrtps
[8.976s] make[2]: *** [/home/michiel/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/microRTPS_agent.cpp] Error 1
[8.977s] make[2]: *** Deleting file '/home/michiel/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/microRTPS_agent.cpp'
[8.977s] make[1]: *** [CMakeFiles/micrortps_agent.dir/all] Error 2
[8.978s] make[1]: *** Waiting for unfinished jobs....
[17.688s] make: *** [all] Error 2`

HairBallChunk, 2020-03-09:
I have a similar issue with the same build command and also with the build_all.bash script. It will build the px4_msgs, but at 8% of the px4_ros_com package it fails.

> [3.431s] openjdk version "1.8.0_242"
[3.431s] OpenJDK Runtime Environment (build 1.8.0_242-8u242-b08-0ubuntu3~18.04-b08)
[3.432s] OpenJDK 64-Bit Server VM (build 25.242-b08, mixed mode)
[4.440s] openjdk version "1.8.0_242"
[4.441s] OpenJDK Runtime Environment (build 1.8.0_242-8u242-b08-0ubuntu3~18.04-b08)
[4.442s] OpenJDK 64-Bit Server VM (build 25.242-b08, mixed mode)
[8.938s] Traceback (most recent call last):
[8.938s]   File "/home/michiel/px4_ros_com_ros2/src/px4_ros_com/scripts/generate_microRTPS_bridge.py", line 474, in <module>
[8.939s]     generate_agent(agent_out_dir)
[8.939s]   File "/home/michiel/px4_ros_com_ros2/src/px4_ros_com/scripts/generate_microRTPS_bridge.py", line 400, in generate_agent
[8.939s]     raise Exception("No IDL files found in %s" % idl_dir)
[8.940s] Exception: No IDL files found in /home/michiel/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/msg/dds_fastrtps
[8.976s] make[2]: *** [/home/michiel/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/microRTPS_agent.cpp] Error 1
[8.977s] make[2]: *** Deleting file '/home/michiel/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/microRTPS_agent.cpp'
[8.977s] make[1]: *** [CMakeFiles/micrortps_agent.dir/all] Error 2
[8.978s] make[1]: *** Waiting for unfinished jobs....
[17.688s] make: *** [all] Error 2

TSC21, 2020-03-09:
@HairBallChunk it's not a similar issue. Please bring that to a different issue. Anyway, what is the ROS2 version and Fast-RTPS-Gen version you have installed?

TSC21, 2020-03-09:
> I cleaned it, and this was my result.
>
> ```
> Exception in thread "main" java.lang.NullPointerException
> 	at com.eprosima.fastrtps.fastrtpsgen.execute(fastrtpsgen.java:336)
> 	at [...]

@Pman6543 do you have `export FASTRTPSGEN_DIR="/usr/local/bin/"` on your bashrc. Does it match the place where you have Fast-RTPS-Gen installed?

TSC21, 2020-07-28:
Closing as stale.

closed 2020-07-28 by TSC21