PX4/px4_ros_com #38: Unable to establish FastRTPS connection with pixhawk.

Lilienthal-GF, 2020-02-17:
I'm trying to establish a FastRTPS conenction between a pixhawk and a companion computer. Following the the documentation for the RTPS/ROS2 Interface with some additions from the setup_system.bash script from the px4_ros_com script folder I was able to run the px4_sitl_rtps with javmsim, passing the data on to micrortps_agent and getting a read of the sensor data with the sensor_combined_listener on Ubuntu 18.04.03.
Now i'm trying to replace the simulated px4 with a real one. I use two USB serial interfaces (see attached image) to connect to the TELEM1 and TELEM2 ports of the pixhawk. I start the micrortps_client using the mavlink_shell from the first USB-Serial interface (terminal is documented in attached mavlink_shell-html). Then I start the micrortps_agent in a different terminal using the second USB-Serial interface. From the terminal it looks like the connection is established briefly but then collapses due to Bad CRC errors (documented in ros2_shell.html).
I reinstalled the Ubuntu machine a couple of times, to try different versions of FastRTPS and ROS2. The setup_ros2.sh script should contain everything to recreate the error when starting from a clean Ubuntu installation.

nicht762, 2020-03-09:
Have the same problem :(

TSC21, 2020-03-10:
This will be reviewed as soon as possible. But right now, we are prioritising the UDP links since soon enough the new FMU HW will support Ehternet, which will be our main networking technology target for this bridge. It doesn't mean we will not support serial as well.

TSC21, 2020-03-10:
@Lilienthal-GF as we spoke before, I am expecting that adding flow control to the `microRTPS_transport` serial link config would help fix this. Have you tried that?

Lilienthal-GF, 2020-03-10:
Unfortunately i did not have time last week to look into this. I expect to have some time this week, so i will give an update by friday.

PS: Is there a roadmap for the new FCU HW? Having an Ethernet connection for the bridge sounds very promising.

TSC21, 2020-03-10:
@Lilienthal-GF https://discuss.px4.io/uploads/short-url/hDRYO5hxdNfbmNhWvnSVrpAayuB.pdf

TSC21, 2020-03-13:
@Lilienthal-GF any updates?

AndrewWedemier, 2020-04-21:
HI @TSC21. I seem to be facing the same issue. Do you have any recommendations on how to add flow control to microRTPS_transport serial link config?

AndrewWedemier, 2020-05-12:
@TSC21 There seems to be a problematic case where the message being sent starts with the sequence ‘>>>>’ instead of ‘>>>’ which leads to a corruption of struct Header *header
in microRTPS_transport. When this happens, there is at least one possible outcome that causes the microRTPS_agent to be suck in an infinite loop because microRTPS_transport  will no longer ::read() any new data from the serial/ uart port.

Typical expected message to be received into microRTPS_transport: 

[>,>,>,topic_ID,seq,payload_length_H,payload_length_L,CRCHigh,CRCLow,payloadStart, ... ,payloadEnd]

 member variable rx_buff_pos = 0  @ start of execution.

1. On line 116 of microRTPS_transport,  a message beginning with the sequence >>>> is read into the  Transport_node member variable rx_buffer by the  member  function node_read(): 
node_read((void *)(rx_buffer + rx_buff_pos), sizeof(rx_buffer) – rx_buff_pos); This member function then returns the length of the characters ::read() 

2.  On line 116, The local variable len will be assigned the value 1024 as 1024 characters will have be read into the rx_buffer as requested.

3.  On line 132 of microRTPS_transport.cpp,  member variable rx_buff_pos will be incremented by the local variable len to be 1024.

4.  On lines 144 to 148 of microRTPS_tranport.cpp, a for loop searches for a possible start of a new message by looking for a sequence marker ‘>’ and then does a memcmp to see if the if it can find the sequence marker ‘>>>’. For this particular input, the loop will break with the for loop counter variable being msg_start_pos =0.

5.  On line 165 of microRTPS_transport,  rx_buffer[msg_start_pos] is packed into struct Header *header

struct __attribute__((packed)) Header {
char marker[3];
uint8_t topic_ID;
uint8_t seq;
uint8_t payload_len_h;
uint8_t payload_len_l;
uint8_t crc_h;
uint8_t crc_l;
};

so char marker[3] gets set to ‘>>>’ and topic_ID gets set to the integer equivalent of character ‘>’.

Since the data is read into the header sequentially, this results in data that  :
a). should have been assigned to topic_ID is assigned to seq instead
b)  should  have been assigned to seq is assigned to payload_len_h instead and so on. 

This results in a possible corruption of the header file.
6. On line 166 of microRTPS_transport, local variable payload_len is assigned the  payload length from the possible corrupted header variable.

7.  on line 169 of microRTPS_transport, an if condition checks to see if payload ( given by the now possibly corrupted payload_len) will fit into the buffer.  This results into at least 2 possible outcomes with one of them causing an infinite loop where no additional data is requested to be read from serial/uart port.

Possible Outcome 1:

8. Payload length is calculated to be too large  to  fit into the buffer.
9.  On line 170 of microRTPS_transport.cpp, member variable rx_buff_pos will still have the value 1024. and the return value of negative EMSGSIZE is passed to the main function in microRTPS_agent.
10. MicroRTPS_agent.cpp’s  outer while loop in the main function continues to be true and invokes  a call to transport_node->read(&topic_ID, data_buffer, BUFFER_SIZE).

 11. We end up back at line 116 in microRTPS_transport.cpp, passing the calculated value of sizeof(rx_buffer) – rx_buff_pos  as the length of the message to be read from serial/uart port, which evaluates to 1024 – 1024 = 0.
12. On line 116, The local variable len will be assigned the value 0 as 0 characters would have been read into the rx_buffer (as requested).
13.  Line 118 -130  of microRTPS_transport.cpp checks if local variable len is less than or equal to zero, and it may print an error. The value of len is returned to the main function in microRTPS_agent.cpp without rx_buffer, and rx_buff_pos being adjusted.
14.  Code execution goes back to point 10mentioned above and continues from there in an endless loop.

Possible Outcome 2:
8. Payload length is calculated to be small enough to  fit into the buffer.
9. On lines 189 and 200 of microRTPS_transport.cpp, CRC checks are done and the error “Bad CRC ...” is printed out to the screen and local variable len is assigned to -1.
10.  On lines  210 and 213 of microRTPS_transport.cpp,   the message will be discarded from rx_buffer, rx_buff_pos will be adjusted ,  and the value of len = -1 will be returned to the main function in microRTPS_agent.

One possible fix:

1.  Change line 118 in microRTPS_transport.cpp, from
     if (len <= 0) 	
     to 
     if (len < 0)

2. Change lines 169 – 171  in microRTPS_transport.cpp from
     if (buffer_len < header_size + payload_len) {
	return -EMSGSIZE;
       }

      to

    if (buffer_len < header_size + payload_len) {
        memmove(rx_buffer, rx_buffer + msg_start_pos + 1, rx_buff_pos - (msg_start_pos + 1) );
        rx_buff_pos = rx_buff_pos - ( msg_start_pos + 1 ); 
        return -EMSGSIZE;
    }

3.  Change lines 416, in microRTPS_transport.cpp from
       
      ssize_t ret = 0;

      to
       
      ssize_t ret = -1;

Note: I tested these changes on the ROS2 listener where it works. However, I haven’t had a chance to test the ROS2 advertiser. Apologies for the long post!!!!

TSC21, 2020-06-30:
@AndrewWedemier this last post of you totally past by me. I am sorry for the quite late answer. Would you might bring in a fix in upstream PX4 Firmware with the changes you did so to fix this issue? Thanks!

AndrewWedemier, 2020-07-15:
@TSC21 I would be happy to do so. Note that this will be my first time contributing to an open-source project (ever), so this is new for me. 

In addition to the above issue with the microRTPS_agent/transport side of things,  I also was running into some stability issues on the px4 while running the micrortps_client that I narrowed down to when the  PX4_ERR(),  PX4_INFO() statements ran and the stack usage. So, I increased the stack size to 4900 which fixed the instability issues. I am not sure if anyone else is/was experiencing that issue.  Should I push that change as well? 

In the meantime, I will take a look at the dev-guide for contributing to the PX4 community after which I will push my changes as soon as possible.

TSC21, 2020-07-15:
> @TSC21 I would be happy to do so. Note that this will be my first time contributing to an open-source project (ever), so this is new for me.
>
> In addition to the above issue with the [...]

@AndrewWedemier what version of the source code are you testing and what are your current setup? I was actually trying your changes but on my case the microRTPS client causes an hardfault and the board reboots. Is that the issue you were facing?

AndrewWedemier, 2020-07-15:
@TSC21  source code version : 746b312  (tag: v1.11.0-beta1)
setup: px4 connected to raspberry pi 4 via uart over telem 2

Yes, the board was rebooting ( I do believe its being caused by the stack usage).

TSC21, 2020-07-15:
> @TSC21 source code version : 746b312 (tag: v1.11.0-beta1)
> setup: px4 connected to raspberry pi 4 via uart over telem 2
> 
> Yes, the board was rebooting ( I do believe its being caused by the stack usage).

Interesting. Good catch. And is the timesync working properly to you? Can you maybe rebase your changes in current master and try and see if it still works?

TSC21, 2020-07-15:
@AndrewWedemier also last questions:
1. You have only the RTPS streaming on TELEM2, correct? (No MAVLink)
2. What baud rate are you using?

AndrewWedemier, 2020-07-15:
Question 1: I am not sure what timesync is suppose to be doing so I cannot say if it is working properly. However, I have included some images below:

Command: ros2 topic echo /Timesync_PubSubTopic
[image]

Command:  listener timesync -r 100
[image]

Question 2: Yes, I will rebase and check if the changes still work. Note that I have come to realize (after my second post on here) that only changes to lines 169 – 171 in microRTPS_transport.cpp are necessary on my setup for it to work.

Question 3: I checked mavlink status and it is only showing /dev/ttyACM0  and /dev/ttyS1. So as far as I am aware, RTPS is the only thing streaming on telem2 (/dev/ttyS2).  Picture below.

[image]

Question 4: Baudrate is 921600. Picture of my launch of agent and client on ros2 and px4  respectively are below:
[image]
[image]

TSC21, 2020-07-15:
@AndrewWedemier yep the above does look really good. Though, trying to reproduce the above and I still get problems when I start the micrortps_client, even after increasing the stack. After starting the client in the mavlink_shell, if you press enter are you still able to access the console?

TSC21, 2020-07-15:
@AndrewWedemier well I used exactly the same approach as you above and I still get CRC errors and the board still crashes. What Fast-RTPS version are you using at this moment? Also please try to test this against PX4 Firnware master and see if it works, because right now for me it isn't. I have even tried different FTDI cables.

TSC21, 2020-07-15:
Also note that I am using the micrortps_agent generated from the PX4 Firmware for now, not the one that gets generated with `px4_ros_com`.

TSC21, 2020-07-16:
Update: I just downgraded to `v1.11.0-beta1` and apply your changes, also using Fast-RTPS 1.8.3, and the problem persists. I am guessing this is probably an HW issue at this stage. I will try with different serial port cabling tomorrow.

@AndrewWedemier also please if you don't mind, can you share the branch where you have the changes you applied? I want to make sure that there's nothing missing besides what you wrote above.

AndrewWedemier, 2020-07-16:
@TSC21 
Question 1: Yes, I am still able to access console. 

However, there are some occasions where the px4 console will print "ERROR [micrortps_client] Read fail 2" ( or some other number), but even then messages are still being sent between the px4 and my raspberry-pi.  The read fail print doesn't always happen though and when it does I can ctrl-c and connect back via usb to nsh.  

CRC errors still happen, however, sensor and debug messages still get sent between PX4 and raspberry-pi.  I don't think my words or pictures are fully capturing the range of what I am seeing when I do a run. I also do not want to give any false impressions. So, I have uploaded a video to youtube showing an example of the above.

Youtube Video: https://youtu.be/BmS8oIVtNN4
or you can search for it on youtube typing in "Example run for TSC21"

Question 2: I am using fastrtpsgen version 1.0.4

[image]

Note: I used microRTPS_transport from px4_ros_com (which is running on my raspberry pi). I also applied the changes to both microRTPS_transport from px4_ros_com and microRTPS_transport from px4 Firmware.  Also, note that I did the adjustments to the stack size in two places.

Yes, there are a few other changes I made to the code ( which are rather ugly, but I did them mostly for convenience while I was trying to debug). What I will do is upload all my changes to my GitHub and send you a link so you can check out.  Please see blow:

Link to px4_ros_com:  https://github.com/AndrewWedemier/px4_ros_com_andrew_edit.git  (branch: master)
Link to Firmware: https://github.com/AndrewWedemier/Firmware_andrew_edit.git (branch: rtps_fix)

I will also rebase code and check if it is still running with my changes as you requested.

TSC21, 2020-07-16:
@AndrewWedemier thanks for sharing this.

So after some evaluation and testing, I realise that what you are proposing doesn't actually fix the problem, it just hides it. The CRC check is there and it's appearing for a reason, and the change you did is just going to result in another error eventually (which I am getting as bellow, regarding the CDR deserialization):
```
terminate called after throwing an instance of 'eprosima::fastcdr::exception::BadParamException'
  what():  Unexpected byte value in Cdr::deserialize(bool), expected 0 or 1
Aborted (core dumped)
```

Your suggestion just overlaps the problem, but it still remains there, so this still needs to be fixed.

Note also that I tried to activate flow control and it didn't help, so it needs to be investigated further.

AndrewWedemier, 2020-07-16:
@TSC21  I would like to ask a few questions:

1.  In microRTPS_transport,  if the payload len + header_size becomes greater than buffer_len, then what is the expected behavior of the program? Is it supposed to go into a loop where it continually requests data of len == 0  to be read?

2. In the case of microRTPS_client,  if there is indeed a case where these PX_ERR() or PX_INFO messages are executed, won't there still be a stack size issue?

  
3. Based on the error that you are seeing, it seems that a bad/incorrect message managed to pass the cyclic redundancy checks, and as a result Cdr::deserialize got a non-boolean value where it expected one and throws an exception.  I doubt CRC will catch all errors, how do we handle a case like that if it happens in the middle of a flight and an exception is thrown?

Additional comments:

I agree that the amount of CRC errors is definitely something to look at, but I doubt they will completely go away.

Sorry for taking up a lot of you time. I really want the UART connection to work as best as possible as I am not sure when I will be able to upgrade to a new FMU HW that allows UDP over ethernet.
I will keep working on it from my side.

TSC21, 2020-07-16:
> @TSC21 I would like to ask a few questions:
> 
> 1. In microRTPS_transport,  if the payload len + header_size becomes greater than buffer_len, then what is the expected behavior of the program? Is it supposed to go into a loop where it continually requests data of len == 0  to be read?

What you are seeing is not a loop but rather the messages being discarded constantly, which means that there is something absolutely wrong on the packet build or parsing. Considering that in the agen `WAIT_CNST` is set to 2, this means that after a couple of seconds, the agent stops receiving data.

> 2. In the case of microRTPS_client,  if there is indeed a case where these PX_ERR() or PX_INFO messages are executed, won't there still be a stack size issue?

Well I am not getting problem on the client at this stage, just in the agent. But if the amount of errors is the causing an high level of error output, then yes in someway it does help to increase. Though your solution to increase the stack size helps on this and it is acceptable.

> 3. Based on the error that you are seeing, it seems that a bad/incorrect message managed to pass the cyclic redundancy checks, and as a result Cdr::deserialize got a non-boolean value where it [...]

This is not supposed to happen in the first place. This just means that the message fields are completely wrong, and you are getting data types on fields where other data types where expected. Again, the CRC will not catch, but it is basically telling us that the packets are not well built/wrongly parsed, and so that will have consequences.

Take into account that I am currently testing with only two messages being streamed: timesync and sensor_combined, and I am still getting CRC issues constantly. So we have to aim to fix this in the first place, and not ignore it, because this is actually telling us there is a problem with the messages we are getting.

> Additional comments:
>
> I agree that the amount of CRC errors is definitely something to look at, but I doubt they will completely go away.
>
> Sorry for taking up a lot of you time. I really want the [...]

You are not taking my time. I do want to fix this as well at this point and your help is much appreciated. Please let's keep in touch so we can find a solution for this in the mean time. Thank you!

AndrewWedemier, 2020-07-16:
@TSC21  Your answers are appreciated. Thank you.

AndrewWedemier, 2020-07-20:
Hi @TSC21 , I have pushed a few updates to the two repos I posted above.  I would like for you to try them on your side.

Although CRC errors have not been completely eliminated, you should notice a significant reduction in the rate of  CRC errors. Also note, I am seeing the message `RTTI too high for timesync: 43ms Offset not update` a lot more (at least relatively). So maybe this suggestion is still just hiding the true issue. 

The exception error that you noted because of my previous suggestion does not seem to be deterministic (so re-producing it can be tricky at times ). So I am asking for your help on running these changes on your side to see if an exception error is still thrown. 

I should note that I ran this most recent update for 3+ hours straight and I had not seen an exception being thrown. I will keep doing tests on my side for the next week or two.

Please try the updates with the following params + your  appropriate serial ports

micrortps_agent -b 3000000 -w 0 -p 1 

micrortps_client start -b 3000000 -w 1 -p 1

If you were to ask me what is the most important change, I would say it is the increase in baud-rate.  

Although I haven’t had a chance to test with only the baud-rate increase, I suspect a baud-rate increase would help with most of issues seen in this thread.

If this is not a true fix, maybe it can help guide us to solving the true issue.

Please let me know if it works for you or if you see any problems on your side. 

Thanks

TSC21, 2020-07-20:
@AndrewWedemier thank you very much for all the help! Please have a look at https://github.com/PX4/Firmware/pull/15354#issuecomment-661052360.

AndrewWedemier, 2020-07-20:
Thanks, @TSC21.

I must thank you for all the work you are doing to link ROS2 to the PX4.  You are doing amazing work and probably saved me countless hours.

TSC21, 2020-07-20:
> Thanks, @TSC21.
> 
> I must thank you for all the work you are doing to link ROS2 to the PX4. You are doing amazing work and probably saved me countless hours.

No need to thank! You really helped unblocking this UART issue. Thank you!

TSC21, 2020-11-28:
This is already fixed upstream. Closing.

closed 2020-11-28 by TSC21

tugbakara, 2021-11-24:
> Also note that I am using the micrortps_agent generated from the PX4 Firmware for now, not the one that gets generated with `px4_ros_com
Hi, which location are you talking about for micrortps_agent from px4_ros_com folder, I have doubts about it. Where is it be located in px4_ros_com?