PX4/px4_ros_com #39: px4_ros_com Build Failure

HairBallChunk, 2020-03-09:
I have a similar issue with the same build command and also with the build_all.bash script. It will build the px4_msgs, but at 8% of the px4_ros_com package it fails. The ROS2 version is Dashing and I'm using fastrtpsgen version 1.7.1.

> [3.431s] openjdk version "1.8.0_242"
[3.431s] OpenJDK Runtime Environment (build 1.8.0_242-8u242-b08-0ubuntu3~18.04-b08)
[3.432s] OpenJDK 64-Bit Server VM (build 25.242-b08, mixed mode)
[4.440s] openjdk version "1.8.0_242"
[4.441s] OpenJDK Runtime Environment (build 1.8.0_242-8u242-b08-0ubuntu3~18.04-b08)
[4.442s] OpenJDK 64-Bit Server VM (build 25.242-b08, mixed mode)
[8.938s] Traceback (most recent call last):
[8.938s]   File "/home/michiel/px4_ros_com_ros2/src/px4_ros_com/scripts/generate_microRTPS_bridge.py", line 474, in <module>
[8.939s]     generate_agent(agent_out_dir)
[8.939s]   File "/home/michiel/px4_ros_com_ros2/src/px4_ros_com/scripts/generate_microRTPS_bridge.py", line 400, in generate_agent
[8.939s]     raise Exception("No IDL files found in %s" % idl_dir)
[8.940s] Exception: No IDL files found in /home/michiel/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/msg/dds_fastrtps
[8.976s] make[2]: *** [/home/michiel/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/microRTPS_agent.cpp] Error 1
[8.977s] make[2]: *** Deleting file '/home/michiel/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/microRTPS_agent.cpp'
[8.977s] make[1]: *** [CMakeFiles/micrortps_agent.dir/all] Error 2
[8.978s] make[1]: *** Waiting for unfinished jobs....
[17.688s] make: *** [all] Error 2

TSC21, 2020-03-09:
Did you follow the [documentation](https://dev.px4.io/master/en/middleware/micrortps.html)? If yes, you would have seen that with ROS2 Dashing ou have to use at least Fast-RTPS 1.8.2 and Fast-RTPS-Gen 1.0.2. The procedure to install both is all documented.

HairBallChunk, 2020-03-09:
I installed  Fast-RTPS 1.8.2 and Fast-RTPS-Gen 1.0.2 using the Installation by Sources from the linked documentation tab. Now the script (source build_all.bash --ros1_ws_dir ~/px4_ros_com_ros1) fails at the ros1_bridge build with:

> 
[13.452s] CMake Error at CMakeLists.txt:255 (find_package):
[13.452s]   By not providing "Findlaunch_testing_ament_cmake.cmake" in
[13.452s]   CMAKE_MODULE_PATH this project has asked CMake to find a package
[13.452s]   configuration file provided by "launch_testing_ament_cmake", but CMake did
[13.452s]   not find one.
[13.453s] 
[13.453s]   Could not find a package configuration file provided by
[13.453s]   "launch_testing_ament_cmake" with any of the following names:
[13.453s] 
[13.453s]     launch_testing_ament_cmakeConfig.cmake
[13.453s]     launch_testing_ament_cmake-config.cmake
[13.453s] 
[13.453s]   Add the installation prefix of "launch_testing_ament_cmake" to
[13.453s]   CMAKE_PREFIX_PATH or set "launch_testing_ament_cmake_DIR" to a directory
[13.453s]   containing one of the above files.  If "launch_testing_ament_cmake"
[13.454s]   provides a separate development package or SDK, be sure it has been
[13.454s]   installed.
[13.454s] 
[13.454s]

TSC21, 2020-03-10:
> I installed Fast-RTPS 1.8.2 and Fast-RTPS-Gen 1.0.2 using the Installation by Sources from the linked documentation tab. Now the script (source build_all.bash --ros1_ws_dir ~/px4_ros_com_ros1) fails [...]

It is quite clear what it is asking for you to install ;)

HairBallChunk, 2020-03-10:
I believe I skipped some essential steps in installing ROS melodic and ROS2 dashing. After going back and installing them via the guide. I get another error involving the generation of the micro-rtps agent code:

> 

[2.367s] openjdk version "1.8.0_242"
[2.367s] OpenJDK Runtime Environment (build 1.8.0_242-8u242-b08-0ubuntu3~18.04-b08)
[2.368s] OpenJDK 64-Bit Server VM (build 25.242-b08, mixed mode)
[2.799s] CMakeFiles/micrortps_agent.dir/build.make:347: warning: overriding recipe for target '/home/michiel/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/Timesync.cpp'
[2.799s] CMakeFiles/micrortps_agent.dir/build.make:176: warning: ignoring old recipe for target '/home/michiel/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/Timesync.cpp'
[2.799s] CMakeFiles/micrortps_agent.dir/build.make:350: warning: overriding recipe for target '/home/michiel/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/TimesyncPubSubTypes.cpp'
[2.799s] CMakeFiles/micrortps_agent.dir/build.make:179: warning: ignoring old recipe for target '/home/michiel/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/TimesyncPubSubTypes.cpp'
[2.799s] CMakeFiles/micrortps_agent.dir/build.make:353: warning: overriding recipe for target '/home/michiel/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/TimesyncPubSubTypes.h'
[2.799s] CMakeFiles/micrortps_agent.dir/build.make:182: warning: ignoring old recipe for target '/home/michiel/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/TimesyncPubSubTypes.h'
[2.982s] openjdk version "1.8.0_242"
[2.982s] OpenJDK Runtime Environment (build 1.8.0_242-8u242-b08-0ubuntu3~18.04-b08)
[2.983s] OpenJDK 64-Bit Server VM (build 25.242-b08, mixed mode)
[4.316s] Traceback (most recent call last):
[4.317s]   File "/home/michiel/px4_ros_com_ros2/src/px4_ros_com/scripts/generate_microRTPS_bridge.py", line 480, in <module>
[4.317s]     generate_agent(agent_out_dir)
[4.317s]   File "/home/michiel/px4_ros_com_ros2/src/px4_ros_com/scripts/generate_microRTPS_bridge.py", line 392, in generate_agent
[4.317s]     urtps_templates_dir, package, px_generate_uorb_topic_files.INCL_DEFAULT, classifier.msg_id_map, fastrtpsgen_version, ros2_distro, uRTPS_TIMESYNC_H_TEMPL_FILE)
[4.317s]   File "/home/michiel/px4_ros_com_ros2/src/px4_ros_com/scripts/px_generate_uorb_topic_files.py", line 243, in generate_uRTPS_general
[4.318s]     return generate_by_template(output_file, template_file, merged_em_globals)
[4.318s]   File "/home/michiel/px4_ros_com_ros2/src/px4_ros_com/scripts/px_generate_uorb_topic_files.py", line 338, in generate_by_template
[4.318s]     interpreter.file(open(template_file))
[4.318s]   File "/usr/lib/python3/dist-packages/em.py", line 2359, in file
[4.318s]     self.safe(scanner, done, locals)
[4.319s]   File "/usr/lib/python3/dist-packages/em.py", line 2401, in safe
[4.319s]     self.parse(scanner, locals)
[4.319s]   File "/usr/lib/python3/dist-packages/em.py", line 2421, in parse
[4.319s]     token.run(self, locals)
[4.319s]   File "/usr/lib/python3/dist-packages/em.py", line 1384, in run
[4.319s]     interpreter.serialize(self.code, locals)
[4.320s]   File "/usr/lib/python3/dist-packages/em.py", line 2535, in serialize
[4.320s]     result = self.evaluate(expression, locals)
[4.320s]   File "/usr/lib/python3/dist-packages/em.py", line 2574, in evaluate
[4.320s]     result = eval(expression, self.globals)
[4.320s]   File "<string>", line 1, in <module>
[4.320s] NameError: name 'param' is not defined
[4.334s] Exception ignored in: <bound method Interpreter.__del__ of <empy pseudomodule/interpreter at 0x7f59d3085080>>
[4.334s] Traceback (most recent call last):
[4.334s]   File "/usr/lib/python3/dist-packages/em.py", line 2094, in __del__
[4.334s]   File "/usr/lib/python3/dist-packages/em.py", line 2157, in shutdown
[4.334s]   File "/usr/lib/python3/dist-packages/em.py", line 2646, in finalize
[4.335s]   File "/usr/lib/python3/dist-packages/em.py", line 2200, in push
[4.335s] AttributeError: '_io.TextIOWrapper' object has no attribute 'push'
[4.346s] make[2]: *** [/home/michiel/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/microRTPS_agent.cpp] Error 1
[4.346s] make[2]: *** Deleting file '/home/michiel/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/microRTPS_agent.cpp'
[4.346s] make[1]: *** [CMakeFiles/micrortps_agent.dir/all] Error 2
[4.347s] make[1]: *** Waiting for unfinished jobs....
[12.047s] make: *** [all] Error 2

HairBallChunk, 2020-03-10:
Thanks for the help so far, it has given me some trouble so far to get the necessary dependencies and packages installed for ROS. I guess that can be expected since I'm quite new to this.

TSC21, 2020-07-28:
@HairBallChunk are you still with issues?

CarmChao, 2020-10-02:
When I compile on the Jetson nano, ubuntu18.04, Fast-RTPS 1.8.4 and Fast-RTPS-Gen 1.0.4. I have the same problem
[4.346s] make[2]: *** [l/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/microRTPS_agent.cpp] Error 
but no other information.
Have you solved it?

closed 2020-11-28 by TSC21