PX4/px4_ros_com #40: px4_msgs class member error.

McKayCyborg, 2020-03-12:
Compiling px4_ros_comm on Ubuntu 18.04 with ARM64 architecture and ROS crystal using the following command:

sudo bash src/px4_ros_com/scripts/build_ros2_workspace.bash --no_ros1_bridge --ros_distro crystal --ros_path ~/ros2_crystal/ros2-linux/setup.bash 

It successfully compiles px4_msgs but fails during compilation of px4_ros_comm with about 55 errors similar to: 

`/home/ubuntu/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/RtpsTopics.cpp:297:42: error: ‘class px4_msgs::msg::dds_::BatteryStatus_’ has no member named ‘timestamp’; did you mean ‘timestamp_’?
             _timesync->subtractOffset(st.timestamp());`

I have cloned the latest px4_msgs and px4_ros_comm repos. If this is not a bug, could someone please point me towards the solution?

TSC21, 2020-03-12:
ROS2 Crystal is EOL. Can you update to Dashing?

TSC21, 2020-03-12:
Also, together with that, FastRTPSGen 1.0.2 or higher should be installed.

TSC21, 2020-03-12:
Another detail: you don't need to run the script with `sudo`.

McKayCyborg, 2020-03-12:
Thanks for the fast response!

Okay, it might be worth updating the PX4 documentation if Dashing is required. Currently, it recommends ROS Melodic, ROS2 Crystal and Ubuntu 18.04:

https://dev.px4.io/v1.9.0/en/middleware/micrortps.html

I am using FastRTPSGen version 1.7.2.

McKayCyborg, 2020-03-12:
I will update to dashing and see if that fixes the problem. Will keep you posted.

McKayCyborg, 2020-03-12:
OH MY GOD. I was looking at the wrong version of the docs. Please forgive me.

McKayCyborg, 2020-03-13:
Okay, so I've removed everything crystal-related using: 
`sudo apt remove ros-crystal-* && sudo apt autoremove`

And installed Dashing using:
`sudo apt install ros-dashing-ros-base`

Then I removed the px4_msgs package, re-cloned it and re-ran the following command:
`bash src/px4_ros_com/scripts/build_ros2_workspace.bash --no_ros1_bridge --ros_distro dashing --ros_path /opt/ros/dashing/setup.bash`

It has failed again while compiling px4_ros_com, although with a different error this time:

`Starting >>> px4_ros_com
--- stderr: px4_ros_com                                
java version "1.8.0_241"
Java(TM) SE Runtime Environment (build 1.8.0_241-b07)
Java HotSpot(TM) 64-Bit Server VM (build 25.241-b07, mixed mode)
Traceback (most recent call last):
  File "/home/ubuntu/px4_ros_com_ros2/src/px4_ros_com/scripts/uorb_rtps_classifier.py", line 232, in <module>
    classifier = (Classifier(os.path.abspath(args.yaml_file), msg_dir) if os.path.isabs(args.yaml_file)
  File "/home/ubuntu/px4_ros_com_ros2/src/px4_ros_com/scripts/uorb_rtps_classifier.py", line 55, in __init__
    self.check_if_listed(yaml_file)
  File "/home/ubuntu/px4_ros_com_ros2/src/px4_ros_com/scripts/uorb_rtps_classifier.py", line 142, in check_if_listed
    "to be sent or received by the micro-RTPS bridge.\n"
AssertionError: 
The following messages are not listen under  /home/ubuntu/px4_ros_com_ros2/src/px4_ros_com/templates/uorb_rtps_message_ids.yaml: OrbTestMediumQueue, OrbMultitest, OrbTestLarge, OrbTestMediumMulti, OrbTestMedium, OrbTest, OrbTestMediumQueuePoll

Please add them to the yaml file with the respective ID and, if applicable, mark them to be sent or received by the micro-RTPS bridge.
NOTE: If the message has multi-topics (#TOPICS), these should be added as well.

CMake Error at cmake/GenerateMicroRTPSAgent.cmake:106 (string):
  string sub-command FIND requires 3 or 4 parameters.
Call Stack (most recent call first):
  CMakeLists.txt:60 (include)`

It looks as though some of the uORB messages are not being included for some reason.

My fasrtpsgen version is 1.7.2, java version is 1.8.0_241, ARM64 architecture running on Ubuntu 18.04

McKayCyborg, 2020-03-13:
Alrighty, so I re-cloned px4-ros-com and it seems to compile alright now, although there are a bunch of (about 22) warnings along the lines of:

`/home/ubuntu/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/RtpsTopics.h: In member function ‘uint64_t RtpsTopics::setMsgTimestamp(T*, const uint64_t&)’:
/home/ubuntu/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/RtpsTopics.h:175:105: warning: no return statement in function returning non-void [-Wreturn-type]
     inline uint64_t setMsgTimestamp(T* msg, const uint64_t& timestamp) { msg->timestamp_() = timestamp; }`

It seems these functions aren't returning anything. Is this to be expected?

TSC21, 2020-03-13:
> Alrighty, so I re-cloned px4-ros-com and it seems to compile alright now, although there are a bunch of (about 22) warnings along the lines of: [...]

It's not an issue, rather a warning which I will fix later. Closing the issue as it is solved.

closed 2020-03-13 by TSC21