PX4/px4_ros_com #41: Building ros1_bridge failure

McKayCyborg, 2020-03-15:
My system:
ROS1 Melodic, ROS2 Dashing on an ARM64 RPi4 running Ubuntu 18.04 Bionic

What I've done so far:
1. Successfully built px4 ros2 workspace and can read uORB messages from pixhawk 4.
2. Installed ROS1 melodic using: sudo apt install ros_melodic_ros_base.
3. Sourced /opt/ros/melodic/setup.bash
4. Cloned px4 ros1 packages. 
5. Built using build_ros1_workspace.bash - Fails with copious errors regarding packages not have a setup.py file and errors such as: "distutils.core.setup() was never called -- perhaps 'setup.py' us bit a Disutils setup script?".
6. Built using: colcon build --symlink-install --event-handlers console_direct+. Succeeds with no issues.
7. Sourced px4_ros_com_ros1/install/setup.bash
8. Sourced px4_ros_com_ros2/install/setup.bash
9. Ran colcon build --symlink-install --packages-select ros1_bridge --cmake-force-configure --event-    handlers console_direct+ 

What I expect:
Successful build of ros1_bridge

What happened:
Failure during building of simple_bridge_2_to_1.cpp.o with the following error:

`Traceback (most recent call last):                   
  File "bin/ros1_bridge_generate_factories", line 11, in <module>
    from ros1_bridge import generate_cpp
  File "/home/ubuntu/px4_ros_com_ros2/src/ros1_bridge/ros1_bridge/__init__.py", line 709, in <module>
    rosidl_parser.Field.__hash__ = FieldHash
AttributeError: module 'rosidl_parser' has no attribute 'Field'
make[2]: *** [generated/get_factory.cpp] Error 1     
CMakeFiles/ros1_bridge.dir/build.make:61: recipe for target 'generated/get_factory.cpp' failed
make[1]: *** [CMakeFiles/ros1_bridge.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/ros1_bridge.dir/all' failed`

My comments:
Unsure if this invalid member reference is a developer issue or if it is an issue with my environment. I'm happy to supply the error output generated while running the build_ros1_workspace.bash if you feel its necessary. Any help would be greatly appreciated.

McKayCyborg, 2020-03-16:
Note: I installed the latest FasrtRTPS 1.9.3 from the binaries on eProsima's company website, so fastrtpsgen version is 1.0.2

TSC21, 2020-03-16:
> My system:
> ROS1 Melodic, ROS2 Dashing on an ARM64 RPi4 running Ubuntu 18.04 Bionic
>
> What I've done so far:
>
> 1. Successfully built px4 ros2 workspace and can read uORB messages from pixhawk 4.
> 2. [...]

This doesn't seem a problem with this package but rather with the `ros1_bridge` package. Have you checked if this problem has been reported and/or solved on the package repo? Even so, this won't be the place to report it.

TSC21, 2020-03-16:
> I'm happy to supply the error output generated while running the build_ros1_workspace.bash if you feel its necessary.

I appreciate so even though I believe it has nothing to do with your error.

TSC21, 2020-07-28:
I assume this got fixed with the suggestion in ros2/ros1_bridge#248.

closed 2020-07-28 by TSC21