PX4/px4_ros_com #42: Error when I try to build px4_ros_com_ros2 package : CMake Error at CMakeLists.txt:13 (find_package)

bastientaillandier, 2020-04-10:
Hi !

I'm on ROS2-Eloquent and I follow this doc : https://dev.px4.io/master/en/middleware/micrortps.html
And I follow this page to install Fast-RTPS and Fast-RTPS-Gen : https://dev.px4.io/master/en/setup/fast-rtps-installation.html

The installation of FastRTPS-1.8.2 succeed but for FastRTPS-Gen, after cloning the repository, I tried to assemble with gradle 6.3, it worked but I can not gradle install because of Permission Denied then I tried with sudo gradle install but zsh (and bash) said : sudo: gradle : command not found
Then I tried to install gradle with apt install because I can sudo with it, but when I assemble : 

[image]

This error occured.
Then a just gradle assemble and copied fastrtpsgen script to /usr/local/bin/ and fastrtpsgen.jar to /usr/local/share/fastrtpsgen/java

And when I tried to build px4_ros_com_ros2 package I get this issue :

[image]

Thank's a lot for helping me !

TSC21, 2020-04-12:
@bastientaillandier use `./build_ros2_workspace.bash` instead of `source ...`.

What version of the Firmware are you using? Please clone and use PX4 master.

Also make sure that `fastrtpsgen -version` returns `1.0.4`.

bastientaillandier, 2020-04-13:
I clone the the PX4 master and fastrtpsgen -version return 1.0.4.
When I tried to use ./build_ros2_workspace.bash I get this issue : 
[image]

any idea ?

bastientaillandier, 2020-04-16:
An other version of FastRtps was installed during the installation of ROS2-Eloquent.
I just rm -rf /opt/ros/eloquent/share/fastrtps and fastcdr and it solved my problem.
Thank you

TSC21, 2020-04-16:
> An other version of FastRtps was installed during the installation of ROS2-Eloquent.
> I just rm -rf /opt/ros/eloquent/share/fastrtps and fastcdr and it solved my problem.
> Thank you

You are not supposed to delete those versions.

bastientaillandier, 2020-04-16:
If I don't delete Eloquent's version in `/opt/ros/eloquent/`, the wrong version of FastRTPS (1.9.3) is used to build the workspace and the compilation error in my first screenshot occurs.

What is your own compilation process for building the workspace ?

Here is what I did :
- [install Fast-RTPS 1.8.2 from source](https://dev.px4.io/master/en/setup/fast-rtps-installation.html)
- install Fast-RTPS-GEN (`fastrtpsgen -version` return 1.0.4)
- create a ws with px4_ros_com, px4_msgs and Firmware, all on `master`
- `make px4_sitl_rtps`
- `sudo ./src/px4_ros_com/scripts/build_ros2_workspace.bash --no_ros1_bridge --ros_distro eloquent --ros_path /opt/ros/eloquent/setup.bash`

Then when I tried `micrortps_agent -t UDP` I had the following error:
`micrortps_agent: symbol lookup error: micrortps_agent: undefined symbol: _ZN8eprosima8fastrtps11Parameter_tC2Ev`

TSC21, 2020-04-16:
> `sudo ./src/px4_ros_com/scripts/build_ros2_workspace.bash --no_ros1_bridge --ros_distro eloquent --ros_path /opt/ros/eloquent/setup.bash`

This is wrong. You are not supposed to run that script with sudo.

bastientaillandier, 2020-04-16:
> > `sudo ./src/px4_ros_com/scripts/build_ros2_workspace.bash --no_ros1_bridge --ros_distro eloquent --ros_path /opt/ros/eloquent/setup.bash`
> 
> This is wrong. You are not supposed to run that script with sudo.

I tried without sudo too and it did the same error. I just use sudo because somewhere it said : Permission denied

TSC21, 2020-04-16:
Also, if you are using ROS2 Eloquent, then you do have to install Fast-RTPS 1.9.3, not 1.8.2. That's why we always match the supported versions of Fast-RTPS with the ones being used on the ROS2 package dependencies.
Meaning:
- ROS2 Eloquent - install Fast-RTPS 1.9.3
- ROS2 Dashing - install Fast-RTPS 1.8.2.

You don't find any information on the docs regarding this because we officially offer support to ROS2 Dashing, since it's LTS.

`./src/px4_ros_com/scripts/build_ros2_workspace.bash --no_ros1_bridge` should be enough.

TSC21, 2020-04-16:
> I tried without sudo too and it did the same error. I just use sudo because somewhere it said : Permission denied

That's because you already used it with sudo before and now you have all your build, log and install files requiring sudo permissions.

bastientaillandier, 2020-04-16:
> That's because you already used it with sudo before and now you have all your build, log and install files requiring sudo permissions.

Ok thank you !

bastientaillandier, 2020-04-16:
> Also, if you are using ROS2 Eloquent, then you do have to install Fast-RTPS 1.9.3, not 1.8.2. That's why we always match the supported versions of Fast-RTPS with the ones being used on the ROS2 [...]

Ok I use the Fast-RTPS-1.9.3 with Eloquent and it works well.
But I use fastrtpsgen 1.0.2 (by default with the binaries of the 1.9.3 version). I think it work anyway but I'm not sure because micrortps_client display : command not found. The agent work.
Does  PX4/Firmware starts the Client as a permanent daemon process now ?

TSC21, 2020-04-16:
> But I use fastrtpsgen 1.0.2 (by default with the binaries of the 1.9.3 version). I think it work anyway but I'm not sure because micrortps_client display : command not found. The agent work.

You need 1.0.4 because of the way that the Publisher/Subscribers are generated. I am sure that it builds but it won't work if you try to get any data on a ROS2 node.

> Does PX4/Firmware starts the Client as a permanent daemon process now ?
No. The command is `micrortps_client start` and you need to build a PX4 target with `_rtps` suffix (ex. `make px4_sitl_rtps`).

closed 2020-07-08 by TSC21