PX4/px4_ros_com #44: Build failure with colcon

Jaeyoung-Lim, 2020-06-06:
I have installed fastrtpsgen following this [doc](https://dev.px4.io/master/en/setup/fast-rtps-installation.html), and followed the installation [tutorials](https://dev.px4.io/master/en/middleware/micrortps.html#building-the-workspaces).

However, the build fails when running 
```
colcon build --symlink-install --packages-skip ros1_bridge --event-handlers console_direct+
```

The error log is as the following. 
```
/sbin/ldconfig.real: /lib/i386-linux-gnu/ld-2.27.so is the dynamic linker, ignoring

/sbin/ldconfig.real: /lib/x86_64-linux-gnu/ld-2.27.so is the dynamic linker, ignoring

/sbin/ldconfig.real: /lib32/ld-2.27.so is the dynamic linker, ignoring

/sbin/ldconfig.real: Can't create temporary cache file /etc/ld.so.cache~: Permission denied
All good. RTPS ID's are unique
Traceback (most recent call last):                 
  File "/home/jaeyoung/px4_ros_com_ros2/src/px4_ros_com/scripts/generate_microRTPS_bridge.py", line 497, in <module>
    generate_agent(agent_out_dir)
  File "/home/jaeyoung/px4_ros_com_ros2/src/px4_ros_com/scripts/generate_microRTPS_bridge.py", line 415, in generate_agent
    raise Exception("No IDL files found in %s" % idl_dir)
Exception: No IDL files found in /home/jaeyoung/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/msg/dds_fastrtps
CMakeFiles/micrortps_agent.dir/build.make:61: recipe for target '/home/jaeyoung/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/microRTPS_agent.cpp' failed
make[2]: *** [/home/jaeyoung/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/microRTPS_agent.cpp] Error 1
make[2]: *** Deleting file '/home/jaeyoung/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/microRTPS_agent.cpp'
CMakeFiles/Makefile2:136: recipe for target 'CMakeFiles/micrortps_agent.dir/all' failed
make[1]: *** [CMakeFiles/micrortps_agent.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
--- stderr: px4_ros_com
CMakeFiles/micrortps_agent.dir/build.make:353: warning: overriding recipe for target '/home/jaeyoung/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/Timesync_.cpp'
CMakeFiles/micrortps_agent.dir/build.make:182: warning: ignoring old recipe for target '/home/jaeyoung/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/Timesync_.cpp'
CMakeFiles/micrortps_agent.dir/build.make:356: warning: overriding recipe for target '/home/jaeyoung/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/Timesync_PubSubTypes.cpp'
CMakeFiles/micrortps_agent.dir/build.make:185: warning: ignoring old recipe for target '/home/jaeyoung/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/Timesync_PubSubTypes.cpp'
CMakeFiles/micrortps_agent.dir/build.make:359: warning: overriding recipe for target '/home/jaeyoung/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/Timesync_PubSubTypes.h'
CMakeFiles/micrortps_agent.dir/build.make:188: warning: ignoring old recipe for target '/home/jaeyoung/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/Timesync_PubSubTypes.h'
openjdk version "1.8.0_252"
OpenJDK Runtime Environment (build 1.8.0_252-8u252-b09-1~18.04-b09)
OpenJDK 64-Bit Server VM (build 25.252-b09, mixed mode)
/sbin/ldconfig.real: Can't stat /usr/local/lib/i386-linux-gnu: No such file or directory
/sbin/ldconfig.real: Can't stat /usr/local/lib/i686-linux-gnu: No such file or directory
/sbin/ldconfig.real: Can't stat /lib/i686-linux-gnu: No such file or directory
/sbin/ldconfig.real: Can't stat /usr/lib/i686-linux-gnu: No such file or directory
/sbin/ldconfig.real: Can't stat /usr/local/lib/x86_64-linux-gnu: No such file or directory
/sbin/ldconfig.real: Path `/lib/x86_64-linux-gnu' given more than once
/sbin/ldconfig.real: Path `/usr/lib/x86_64-linux-gnu' given more than once
/sbin/ldconfig.real: /lib/i386-linux-gnu/ld-2.27.so is the dynamic linker, ignoring

/sbin/ldconfig.real: /lib/x86_64-linux-gnu/ld-2.27.so is the dynamic linker, ignoring

/sbin/ldconfig.real: /lib32/ld-2.27.so is the dynamic linker, ignoring

/sbin/ldconfig.real: Can't create temporary cache file /etc/ld.so.cache~: Permission denied
Traceback (most recent call last):
  File "/home/jaeyoung/px4_ros_com_ros2/src/px4_ros_com/scripts/generate_microRTPS_bridge.py", line 497, in <module>
    generate_agent(agent_out_dir)
  File "/home/jaeyoung/px4_ros_com_ros2/src/px4_ros_com/scripts/generate_microRTPS_bridge.py", line 415, in generate_agent
    raise Exception("No IDL files found in %s" % idl_dir)
Exception: No IDL files found in /home/jaeyoung/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/msg/dds_fastrtps
make[2]: *** [/home/jaeyoung/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/microRTPS_agent.cpp] Error 1
make[2]: *** Deleting file '/home/jaeyoung/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/microRTPS_agent.cpp'
make[1]: *** [CMakeFiles/micrortps_agent.dir/all] Error 2
make: *** [all] Error 2
---

```

TSC21, 2020-06-06:
Can you please try the build scripts we have available first?

Jaeyoung-Lim, 2020-06-06:
@TSC21 I initially tried with the `build_all.sh` but had the same error. I was under the impression that something was wrong related to https://github.com/PX4/px4_ros_com/issues/43

TSC21, 2020-06-11:
@Jaeyoung-Lim does the problem persist?

closed 2020-07-28 by TSC21