PX4/px4_ros_com #45: sensor_combined_listener does not print out message contents.

AndyBlightLeeds, 2020-06-11:
I have built and run the sensor_combined_listener example.  Everything builds and runs but this is the only output I get from the sensor_combined_listener:
```bash
ros2 launch px4_ros_com sensor_combined_listener.launch.py
[INFO] [launch]: All log files can be found below /home/andy/.ros/log/2020-06-11-09-21-17-147249-andy-Precision-7510-26498
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [sensor_combined_listener-1]: process started with pid [26510]
[sensor_combined_listener-1] Starting sensor_combined listener node...
```

The setup I am using is:
* Ubuntu 18.04LTS (fresh install on 5 June 2020 and fully updated).
* ROS Eloquent (Note: ROS Melodic is not installed as I want to create a ROS2 only solution).
* Fast-RTPS version: 1.9.2
* Fast-RTPS-Gen version: 1.0.2

I started the apps as follows:

Terminal 1. Start gazebo SITL using:

```bash
cd ~/git/PX4/Firmware
make px4_sitl_rtps gazebo
```

After the usual first time run delay, an iris copter is displayed on an ashpalt ground plane being shown in Gazebo.

Terminal 2. Start the client using:

```bash
cd ~/git/PX4/Firmware
./build/px4_sitl_rtps/bin/px4-micrortps_client start -t UDP
```

`px4-micrortps_client` is a symbolic link to `px4`.  The `px4` daemon is started with the command `make px4_sitl_rtps gazebo`.

Terminal 3. Start the agent using:

```bash
 . /opt/ros/eloquent/setup.bash
~/px4/ros_com/install/px4_ros_com/bin/micrortps_agent -t UDP
```

Terminal 4.  Start the combined listener example.

```bash
cd ~/px4/ros_com/
. install/local_setup.bash
ros2 launch px4_ros_com sensor_combined_listener.launch.py
```

You should see regular updates but nothing is every printed. 

I then tried using `ros2 topic echo /SensorCombined_PubSubTopic` but this produced an error message:
```text
Cannot echo topic '/SensorCombined_PubSubTopic', as it contains more than one type: [px4_msgs::msg::SensorCombined, px4_msgs/msg/SensorCombined]
```
This looks like it is the same problem reported in #35.

TSC21, 2020-06-11:
@AndyBlightLeeds on the documentation it is explicit that the required [Fast-RTPS-Gen version required is 1.0.4](https://dev.px4.io/master/en/middleware/micrortps.html#code-generation). What you are seeing above is a result of something that was solved already in this version.

AndyBlightLeeds, 2020-06-11:
Thanks for pointing out my error.  After rebuilding and installing fastrtpsgen v1.0.4, this problem has gone.

closed 2020-06-11 by AndyBlightLeeds