PX4/px4_ros_com #46: MicroRTPS agent causes seg fault when stopped.

AndyBlightLeeds, 2020-06-11:
I have built and run the sensor_combined_listener example on this page <https://dev.px4.io/master/en/middleware/micrortps.html>.  Every time I stop the agent using `Ctrl+c` the agent, the agent causes a seg fault.  The GDB back trace from the core dump is shown below:
```text
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Core was generated by `/home/andy/px4/ros_com/install/px4_ros_com/bin/micrortps_agent -t UDP'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  _int_malloc (av=av@entry=0x7fbed6135c40 <main_arena>, bytes=bytes@entry=8) at malloc.c:3779
3779    malloc.c: No such file or directory.
[Current thread is 1 (Thread 0x7fbdc57fa700 (LWP 21342))]
(gdb) bt
#0  _int_malloc (av=av@entry=0x7fbed6135c40 <main_arena>, bytes=bytes@entry=8) at malloc.c:3779
#1  0x00007fbed5de12ed in __GI___libc_malloc (bytes=8) at malloc.c:3065
#2  0x00007fbed6784298 in operator new(unsigned long) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007fbed71ecc6f in std::vector<eprosima::fastrtps::rtps::WriterProxyData*, std::allocator<eprosima::fastrtps::rtps::WriterProxyData*> >::reserve(unsigned long) () from /opt/ros/eloquent/lib/libfastrtps.so.1
#4  0x00007fbed71ea0d9 in eprosima::fastrtps::rtps::PDP::removeWriterProxyData(eprosima::fastrtps::rtps::GUID_t const&) ()
   from /opt/ros/eloquent/lib/libfastrtps.so.1
#5  0x00007fbed7220b5f in eprosima::fastrtps::rtps::EDPSimplePUBListener::onNewCacheChangeAdded(eprosima::fastrtps::rtps::RTPSReader*, eprosima::fastrtps::rtps::CacheChange_t const*) () from /opt/ros/eloquent/lib/libfastrtps.so.1
#6  0x00007fbed706cb7f in eprosima::fastrtps::rtps::StatefulReader::NotifyChanges(eprosima::fastrtps::rtps::WriterProxy*) ()
   from /opt/ros/eloquent/lib/libfastrtps.so.1
#7  0x00007fbed706cc77 in eprosima::fastrtps::rtps::StatefulReader::change_received(eprosima::fastrtps::rtps::CacheChange_t*, eprosima::fastrtps::rtps::WriterProxy*) () from /opt/ros/eloquent/lib/libfastrtps.so.1
#8  0x00007fbed706df5e in eprosima::fastrtps::rtps::StatefulReader::processDataMsg(eprosima::fastrtps::rtps::CacheChange_t*) ()
   from /opt/ros/eloquent/lib/libfastrtps.so.1
#9  0x00007fbed709593d in eprosima::fastrtps::rtps::MessageReceiver::findAllReaders(eprosima::fastrtps::rtps::EntityId_t const&, std::function<void (eprosima::fastrtps::rtps::RTPSReader*)>) () from /opt/ros/eloquent/lib/libfastrtps.so.1
#10 0x00007fbed709632d in eprosima::fastrtps::rtps::MessageReceiver::proc_Submsg_Data(eprosima::fastrtps::rtps::CDRMessage_t*, eprosima::fastrtps::rtps::SubmessageHeader_t*) () from /opt/ros/eloquent/lib/libfastrtps.so.1
#11 0x00007fbed709ad96 in eprosima::fastrtps::rtps::MessageReceiver::processCDRMsg(eprosima::fastrtps::rtps::Locator_t const&, eprosima::fastrtps::rtps::CDRMessage_t*) () from /opt/ros/eloquent/lib/libfastrtps.so.1
#12 0x00007fbed709d5ba in eprosima::fastrtps::rtps::ReceiverResource::OnDataReceived(unsigned char const*, unsigned int, eprosima::fastrtps::rtps::Locator_t const&, eprosima::fastrtps::rtps::Locator_t const&) () from /opt/ros/eloquent/lib/libfastrtps.so.1
#13 0x00007fbed70e0fc4 in eprosima::fastrtps::rtps::UDPChannelResource::perform_listen_operation(eprosima::fastrtps::rtps::Locator_t) () from /opt/ros/eloquent/lib/libfastrtps.so.1
#14 0x00007fbed70e223c in std::thread::_State_impl<std::thread::_Invoker<std::tuple<void (eprosima::fastrtps::rtps::UDPChannelResource::*)(eprosima::fastrtps::rtps::Locator_t), eprosima::fastrtps::rtps::UDPChannelResource*, eprosima::fastrtps::rtps::Locator_t> > >::_M_run() () from /opt/ros/eloquent/lib/libfastrtps.so.1
#15 0x00007fbed67ae6df in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#16 0x00007fbed6a816db in start_thread (arg=0x7fbdc57fa700) at pthread_create.c:463
#17 0x00007fbed5e6b88f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
(gdb)
```

The setup I am using is:
* Ubuntu 18.04LTS (fresh install on 5 June 2020 and fully updated).
* ROS Eloquent (Note: ROS Melodic is not installed as I want to create a ROS2 only solution).
* Fast-RTPS version: /opt/ros/eloquent/lib/libfastrtps.so.1.9.3
* Fast-RTPS-Gen version: 1.0.2

TSC21, 2020-06-11:
It's not clear what is causing the segfault. What is clear is that you are not using the correct Fast-RTPS-Gen version, which should be 1.0.4.

AndyBlightLeeds, 2020-06-11:
I have rebuilt fastrtpsgen and now have version 1.0.4 installed.

```bash
$ /usr/local/bin/fastrtpsgen -version
openjdk version "11.0.7" 2020-04-14
OpenJDK Runtime Environment (build 11.0.7+10-post-Ubuntu-2ubuntu218.04)
OpenJDK 64-Bit Server VM (build 11.0.7+10-post-Ubuntu-2ubuntu218.04, mixed mode, sharing)
fastrtpsgen version 1.0.4
```
I have deleted my px4_ros_com directory containing source and all build files.  I have then cloned the px4_msgs and px4_ros_com repos and rebuilt.

The seg fault is still happening.  Thinking it might possibly be a version problem with the libraries that fastrtpsgen is using, I have included the output from ldd.
```bash
$ ldd /home/andy/px4/ros_com/install/px4_ros_com/bin/micrortps_agent     
        linux-vdso.so.1 (0x00007fff57564000)
        libfastrtps.so.1 => /opt/ros/eloquent/lib/libfastrtps.so.1 (0x00007fba334b9000)
        libfastcdr.so.1 => /opt/ros/eloquent/lib/libfastcdr.so.1 (0x00007fba332a7000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fba33088000)
        libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fba32cff000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fba32961000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fba32749000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fba32358000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fba32154000)
        libtinyxml2.so.6 => /usr/lib/x86_64-linux-gnu/libtinyxml2.so.6 (0x00007fba31f40000)
        libssl.so.1.1 => /usr/lib/x86_64-linux-gnu/libssl.so.1.1 (0x00007fba31cb3000)
        libcrypto.so.1.1 => /usr/lib/x86_64-linux-gnu/libcrypto.so.1.1 (0x00007fba317e8000)
        /lib64/ld-linux-x86-64.so.2 (0x00007fba33f4e000)
```

These are the library versions being used:
```bash
$ ll /opt/ros/eloquent/lib/libfast*
lrwxrwxrwx 1 root root      15 Dec 13 04:13 /opt/ros/eloquent/lib/libfastcdr.so -> libfastcdr.so.1
lrwxrwxrwx 1 root root      20 Dec 13 04:13 /opt/ros/eloquent/lib/libfastcdr.so.1 -> libfastcdr.so.1.0.10
-rw-r--r-- 1 root root   72480 Dec 13 04:13 /opt/ros/eloquent/lib/libfastcdr.so.1.0.10
lrwxrwxrwx 1 root root      16 Feb 18 17:47 /opt/ros/eloquent/lib/libfastrtps.so -> libfastrtps.so.1
lrwxrwxrwx 1 root root      20 Feb 18 17:47 /opt/ros/eloquent/lib/libfastrtps.so.1 -> libfastrtps.so.1.9.3
-rw-r--r-- 1 root root 5954248 Feb 18 17:47 /opt/ros/eloquent/lib/libfastrtps.so.1.9.3
```
Could it be that I need to build a different version of the other fast* libraries to pick up the fix you mentioned in another thread?

TSC21, 2020-07-28:
> I have rebuilt fastrtpsgen and now have version 1.0.4 installed.
> 
> ```shell
> $ /usr/local/bin/fastrtpsgen -version
> openjdk version "11.0.7" 2020-04-14
> OpenJDK Runtime Environment (build 11.0.7+10-post-Ubuntu-2ubuntu218.04)
> OpenJDK 64-Bit Server VM (build 11.0.7+10-post-Ubuntu-2ubuntu218.04, mixed mode, sharing)
> fastrtpsgen version 1.0.4
> ```
> 
> I have deleted my px4_ros_com directory containing source and all build files. I have then cloned the px4_msgs and px4_ros_com repos and rebuilt.
> 
> The seg fault is still happening. Thinking it might possibly be a version problem with the libraries that fastrtpsgen is using, I have included the output from ldd.
> 
> ```shell
> $ ldd /home/andy/px4/ros_com/install/px4_ros_com/bin/micrortps_agent     
>         linux-vdso.so.1 (0x00007fff57564000)
>         libfastrtps.so.1 => /opt/ros/eloquent/lib/libfastrtps.so.1 (0x00007fba334b9000)
>         libfastcdr.so.1 => /opt/ros/eloquent/lib/libfastcdr.so.1 (0x00007fba332a7000)
>         libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fba33088000)
>         libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fba32cff000)
>         libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fba32961000)
>         libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fba32749000)
>         libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fba32358000)
>         libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fba32154000)
>         libtinyxml2.so.6 => /usr/lib/x86_64-linux-gnu/libtinyxml2.so.6 (0x00007fba31f40000)
>         libssl.so.1.1 => /usr/lib/x86_64-linux-gnu/libssl.so.1.1 (0x00007fba31cb3000)
>         libcrypto.so.1.1 => /usr/lib/x86_64-linux-gnu/libcrypto.so.1.1 (0x00007fba317e8000)
>         /lib64/ld-linux-x86-64.so.2 (0x00007fba33f4e000)
> ```
> 
> These are the library versions being used:
> 
> ```shell
> $ ll /opt/ros/eloquent/lib/libfast*
> lrwxrwxrwx 1 root root      15 Dec 13 04:13 /opt/ros/eloquent/lib/libfastcdr.so -> libfastcdr.so.1
> lrwxrwxrwx 1 root root      20 Dec 13 04:13 /opt/ros/eloquent/lib/libfastcdr.so.1 -> libfastcdr.so.1.0.10
> -rw-r--r-- 1 root root   72480 Dec 13 04:13 /opt/ros/eloquent/lib/libfastcdr.so.1.0.10
> lrwxrwxrwx 1 root root      16 Feb 18 17:47 /opt/ros/eloquent/lib/libfastrtps.so -> libfastrtps.so.1
> lrwxrwxrwx 1 root root      20 Feb 18 17:47 /opt/ros/eloquent/lib/libfastrtps.so.1 -> libfastrtps.so.1.9.3
> -rw-r--r-- 1 root root 5954248 Feb 18 17:47 /opt/ros/eloquent/lib/libfastrtps.so.1.9.3
> ```
> 
> Could it be that I need to build a different version of the other fast* libraries to pick up the fix you mentioned in another thread?

This might be actually a problem with the agent itself and how it closes the application. I will dig in further when I have some time. Thanks for reporting.

TSC21, 2021-07-01:
This was solved meanwhile. Closing.

closed 2021-07-01 by TSC21