PX4/px4_ros_com #47: ROS2 installation error - fastrtps

de-ernsten, 2020-06-13:
Hey guys,

I installed ROS 2 dashing and be using ROS melodic. 
I Installed Fast RTPS 1.8.2 and FastRTPSGen 1.0.4, cloned the px4 repositories like described [here](https://dev.px4.io/master/en/middleware/micrortps.html).

However, when I run colcon build --symlink-install --packages-skip ros1_bridge --event-handlers console_direct+ i get the following output:

```
Finished <<< px4_msgs [15.0s]
Starting >>> px4_ros_com
-- Using C++17 compiler                          
-- Found ament_cmake: 0.7.4 (/opt/ros/dashing/share/ament_cmake/cmake)
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Override CMake install command with custom implementation using symlinks instead of copying resources
-- Found builtin_interfaces: 0.7.4 (/opt/ros/dashing/share/builtin_interfaces/cmake)
-- Found rosidl_adapter: 0.7.9 (/opt/ros/dashing/share/rosidl_adapter/cmake)
-- Found rclcpp: 0.7.13 (/opt/ros/dashing/share/rclcpp/cmake)
-- Found rmw_implementation_cmake: 0.7.2 (/opt/ros/dashing/share/rmw_implementation_cmake/cmake)
-- Using RMW implementation 'rmw_fastrtps_cpp' as default
-- Found rmw_fastrtps_cpp: 0.7.7 (/opt/ros/dashing/share/rmw_fastrtps_cpp/cmake)
-- Found px4_msgs: 2.0.1 (/home/falke/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/cmake)
-- Found sensor_msgs: 0.7.0 (/opt/ros/dashing/share/sensor_msgs/cmake)
-- Found eigen3_cmake_module: 0.1.1 (/opt/ros/dashing/share/eigen3_cmake_module/cmake)
-- fastrtpsgen found in /usr/local/bin
openjdk version "1.8.0_252"
OpenJDK Runtime Environment (build 1.8.0_252-8u252-b09-1~18.04-b09)
OpenJDK 64-Bit Server VM (build 25.252-b09, mixed mode)
-- fastrtpsgen version 1.0.4
-- Retrieving list of msgs to send...
-- List of msgs to send: CameraCapture, CameraTrigger, CollisionReport, DebugArray, DebugKeyValue, DebugValue, DebugVect, ObstacleDistance, OpticalFlow, PositionSetpoint, PositionSetpointTriplet, Timesync, TrajectoryWaypoint, VehicleTrajectoryWaypoint, OnboardComputerStatus, VehicleMocapOdometry, VehicleVisualOdometry
-- Retrieving list of msgs to receive...
-- List of msgs to receive: AdcReport, Airspeed, BatteryStatus, Cpuload, DistanceSensor, EstimatorStatus, HomePosition, IridiumsbdStatus, RadioStatus, SatelliteInfo, SensorBaro, SensorCombined, SensorSelection, Timesync, VehicleAttitude, VehicleOdometry, VtolVehicleStatus, WindEstimate, CollisionConstraints
-- fastrtpsgen found in 
-- px4_msgs message dir under /home/falke/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/msg
-- IDL definitions under /home/falke/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/msg
-- Configuring done
-- Generating done                               
-- Build files have been written to: /home/falke/px4_ros_com_ros2/build/px4_ros_com
[  2%] Built target sensor_combined_listener     
[  3%] Built target debug_vect_advertiser
[  5%] Built target vehicle_gps_position_listener
[  7%] Built target frame_transforms
CMakeFiles/micrortps_agent.dir/build.make:373: warning: overriding recipe for target '/home/falke/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/Timesync.cpp'
CMakeFiles/micrortps_agent.dir/build.make:202: warning: ignoring old recipe for target '/home/falke/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/Timesync.cpp'
CMakeFiles/micrortps_agent.dir/build.make:376: warning: overriding recipe for target '/home/falke/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/TimesyncPubSubTypes.cpp'
CMakeFiles/micrortps_agent.dir/build.make:205: warning: ignoring old recipe for target '/home/falke/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/TimesyncPubSubTypes.cpp'
CMakeFiles/micrortps_agent.dir/build.make:379: warning: overriding recipe for target '/home/falke/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/TimesyncPubSubTypes.h'
CMakeFiles/micrortps_agent.dir/build.make:208: warning: ignoring old recipe for target '/home/falke/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/TimesyncPubSubTypes.h'
CMakeFiles/micrortps_agent.dir/build.make:373: warning: overriding recipe for target '/home/falke/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/Timesync.cpp'
CMakeFiles/micrortps_agent.dir/build.make:202: warning: ignoring old recipe for target '/home/falke/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/Timesync.cpp'
CMakeFiles/micrortps_agent.dir/build.make:376: warning: overriding recipe for target '/home/falke/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/TimesyncPubSubTypes.cpp'
CMakeFiles/micrortps_agent.dir/build.make:205: warning: ignoring old recipe for target '/home/falke/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/TimesyncPubSubTypes.cpp'
CMakeFiles/micrortps_agent.dir/build.make:379: warning: overriding recipe for target '/home/falke/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/TimesyncPubSubTypes.h'
CMakeFiles/micrortps_agent.dir/build.make:208: warning: ignoring old recipe for target '/home/falke/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/TimesyncPubSubTypes.h'
[  9%] Building CXX object CMakeFiles/micrortps_agent.dir/src/micrortps_agent/Airspeed_Publisher.cpp.o
[  9%] Building CXX object CMakeFiles/micrortps_agent.dir/src/micrortps_agent/AdcReport_Publisher.cpp.o
[ 10%] Building CXX object CMakeFiles/micrortps_agent.dir/src/micrortps_agent/BatteryStatus_Publisher.cpp.o
[ 11%] Building CXX object CMakeFiles/micrortps_agent.dir/src/micrortps_agent/Cpuload_Publisher.cpp.o
[ 12%] Building CXX object CMakeFiles/micrortps_agent.dir/src/micrortps_agent/DistanceSensor_Publisher.cpp.o
[ 13%] Building CXX object CMakeFiles/micrortps_agent.dir/src/micrortps_agent/HomePosition_Publisher.cpp.o
[ 14%] Building CXX object CMakeFiles/micrortps_agent.dir/src/micrortps_agent/IridiumsbdStatus_Publisher.cpp.o
[ 14%] Building CXX object CMakeFiles/micrortps_agent.dir/src/micrortps_agent/EstimatorStatus_Publisher.cpp.o
/home/falke/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/DistanceSensor_Publisher.cpp: In member function ‘bool DistanceSensor_Publisher::init()’:
/home/falke/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/DistanceSensor_Publisher.cpp:67:25: error: ‘class eprosima::fastrtps::rtps::BuiltinAttributes’ has no member named ‘discovery_config’
     PParam.rtps.builtin.discovery_config.leaseDuration = c_TimeInfinite;
                         ^~~~~~~~~~~~~~~~
/home/falke/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/BatteryStatus_Publisher.cpp: In member function ‘bool BatteryStatus_Publisher::init()’:
/home/falke/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/BatteryStatus_Publisher.cpp:67:25: error: ‘class eprosima::fastrtps::rtps::BuiltinAttributes’ has no member named ‘discovery_config’
     PParam.rtps.builtin.discovery_config.leaseDuration = c_TimeInfinite;
                         ^~~~~~~~~~~~~~~~
[... 88 lines cut ...]
                         ^~~~~~~~~~~~~~~~
/home/falke/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/HomePosition_Publisher.cpp: In member function ‘bool HomePosition_Publisher::init()’:
/home/falke/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/HomePosition_Publisher.cpp:67:25: error: ‘class eprosima::fastrtps::rtps::BuiltinAttributes’ has no member named ‘discovery_config’
     PParam.rtps.builtin.discovery_config.leaseDuration = c_TimeInfinite;
                         ^~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/micrortps_agent.dir/src/micrortps_agent/DistanceSensor_Publisher.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/micrortps_agent.dir/src/micrortps_agent/Cpuload_Publisher.cpp.o] Error 1
make[2]: *** [CMakeFiles/micrortps_agent.dir/src/micrortps_agent/BatteryStatus_Publisher.cpp.o] Error 1
make[2]: *** [CMakeFiles/micrortps_agent.dir/src/micrortps_agent/IridiumsbdStatus_Publisher.cpp.o] Error 1
make[2]: *** [CMakeFiles/micrortps_agent.dir/src/micrortps_agent/Airspeed_Publisher.cpp.o] Error 1
make[2]: *** [CMakeFiles/micrortps_agent.dir/src/micrortps_agent/EstimatorStatus_Publisher.cpp.o] Error 1
make[2]: *** [CMakeFiles/micrortps_agent.dir/src/micrortps_agent/AdcReport_Publisher.cpp.o] Error 1
make[2]: *** [CMakeFiles/micrortps_agent.dir/src/micrortps_agent/HomePosition_Publisher.cpp.o] Error 1
make[1]: *** [CMakeFiles/micrortps_agent.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< px4_ros_com	[ Exited with code 2 ]

Summary: 1 package finished [17.6s]
  1 package failed: px4_ros_com
  2 packages had stderr output: px4_msgs px4_ros_com
```

What might cause this? Thanks for your help!

sb-k, 2020-07-26:
Hello,do you find how to solve the problem? I have the same question,and I donot know what to do.

TSC21, 2020-07-28:
Fix in PX4/Firmware@a7112c4.
You can manually set it on your side or wait that the fix gets in. Thanks for reporting.

closed 2020-07-28 by TSC21