PX4/px4_ros_com #51: micrortps_agent compile error

McKayCyborg, 2020-07-27:
My system:

Hardware: Raspberry Pi 4
CPU Architecture: aarch64
Operating System: Ubuntu 18.04 Server (headless)
ROS Version: Dashing
Fast-CDR Version: 1.0.14
Fast-RTPS Version: 1.8.3
Fast-RTPS-Gen Version: 1.0.4

My Goal:
To compile px4_ros_com WITHOUT rods1_bridge because I am only running ROS2 on my system, not ROS1.

My Actions: 
I'm following the master version of this guide: https://dev.px4.io/master/en/middleware/micrortps.html
I'm running the px4_ros_com/scripts/build_ros2_workspace.bash file

My Problem:
px4_msgs succeeds in building and installing, but px4_ros_com fails when compiling the micrortps_agent object files with the following error output:

`openjdk version "1.8.0_252"
OpenJDK Runtime Environment (build 1.8.0_252-8u252-b09-1~18.04-b09)
OpenJDK 64-Bit Server VM (build 25.252-b09, mixed mode)
Cloning into 'ros1_bridge'...
remote: Enumerating objects: 13, done.
remote: Counting objects: 100% (13/13), done.
remote: Compressing objects: 100% (11/11), done.
remote: Total 1261 (delta 4), reused 4 (delta 1), pack-reused 1248
Receiving objects: 100% (1261/1261), 1.48 MiB | 1.04 MiB/s, done.
Resolving deltas: 100% (770/770), done.
Starting >>> px4_msgs
[Processing: px4_msgs] [px4_msgs:build 7% - 30.0s]
[Processing: px4_msgs]                                 
[Processing: px4_msgs]                                       
[Processing: px4_msgs]                                       
[Processing: px4_msgs]                                       
[Processing: px4_msgs]                                       
[Processing: px4_msgs]                                       
[Processing: px4_msgs]                                       
[Processing: px4_msgs]                                       
[Processing: px4_msgs]                                       
[Processing: px4_msgs]                                       
[Processing: px4_msgs]                                       
[Processing: px4_msgs]                                        
[Processing: px4_msgs]                                        
Finished <<< px4_msgs [7min 6s]                               
Starting >>> px4_ros_com  
[Processing: px4_ros_com]                                 
[Processing: px4_ros_com]                                 
[Processing: px4_ros_com]                                      
--- stderr: px4_ros_com                                        
openjdk version "1.8.0_252"
OpenJDK Runtime Environment (build 1.8.0_252-8u252-b09-1~18.04-b09)
OpenJDK 64-Bit Server VM (build 25.252-b09, mixed mode)
CMakeFiles/micrortps_agent.dir/build.make:353: warning: overriding recipe for target '/home/ubuntu/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/Timesync.cpp'
CMakeFiles/micrortps_agent.dir/build.make:182: warning: ignoring old recipe for target '/home/ubuntu/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/Timesync.cpp'
CMakeFiles/micrortps_agent.dir/build.make:356: warning: overriding recipe for target '/home/ubuntu/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/TimesyncPubSubTypes.cpp'
CMakeFiles/micrortps_agent.dir/build.make:185: warning: ignoring old recipe for target '/home/ubuntu/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/TimesyncPubSubTypes.cpp'
CMakeFiles/micrortps_agent.dir/build.make:359: warning: overriding recipe for target '/home/ubuntu/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/TimesyncPubSubTypes.h'
CMakeFiles/micrortps_agent.dir/build.make:188: warning: ignoring old recipe for target '/home/ubuntu/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/TimesyncPubSubTypes.h'
CMakeFiles/micrortps_agent.dir/build.make:353: warning: overriding recipe for target '/home/ubuntu/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/Timesync.cpp'
CMakeFiles/micrortps_agent.dir/build.make:182: warning: ignoring old recipe for target '/home/ubuntu/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/Timesync.cpp'
CMakeFiles/micrortps_agent.dir/build.make:356: warning: overriding recipe for target '/home/ubuntu/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/TimesyncPubSubTypes.cpp'
CMakeFiles/micrortps_agent.dir/build.make:185: warning: ignoring old recipe for target '/home/ubuntu/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/TimesyncPubSubTypes.cpp'
CMakeFiles/micrortps_agent.dir/build.make:359: warning: overriding recipe for target '/home/ubuntu/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/TimesyncPubSubTypes.h'
CMakeFiles/micrortps_agent.dir/build.make:188: warning: ignoring old recipe for target '/home/ubuntu/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/TimesyncPubSubTypes.h'
In file included from /home/ubuntu/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/microRTPS_agent.cpp:52:0:
/home/ubuntu/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/microRTPS_timesync.h: In member function ‘uint64_t TimeSync::setMsgTimestamp(timesync_msg_t*, const uint64_t&)’:
/home/ubuntu/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/microRTPS_timesync.h:170:114: warning: no return statement in function returning non-void [-Wreturn-type]
  inline uint64_t setMsgTimestamp(timesync_msg_t* msg, const uint64_t& timestamp) { msg->timestamp() = timestamp; }
[... 91 lines cut ...]
/home/ubuntu/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/Airspeed_Publisher.cpp:67:25: error: ‘class eprosima::fastrtps::rtps::BuiltinAttributes’ has no member named ‘discovery_config’
     PParam.rtps.builtin.discovery_config.leaseDuration = c_TimeInfinite;
                         ^~~~~~~~~~~~~~~~
/home/ubuntu/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/BatteryStatus_Publisher.cpp: In member function ‘bool BatteryStatus_Publisher::init()’:
/home/ubuntu/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/BatteryStatus_Publisher.cpp:67:25: error: ‘class eprosima::fastrtps::rtps::BuiltinAttributes’ has no member named ‘discovery_config’
     PParam.rtps.builtin.discovery_config.leaseDuration = c_TimeInfinite;
                         ^~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/micrortps_agent.dir/src/micrortps_agent/AdcReport_Publisher.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/home/ubuntu/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/Cpuload_Publisher.cpp: In member function ‘bool Cpuload_Publisher::init()’:
/home/ubuntu/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/Cpuload_Publisher.cpp:67:25: error: ‘class eprosima::fastrtps::rtps::BuiltinAttributes’ has no member named ‘discovery_config’
     PParam.rtps.builtin.discovery_config.leaseDuration = c_TimeInfinite;
                         ^~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/micrortps_agent.dir/src/micrortps_agent/Airspeed_Publisher.cpp.o] Error 1
make[2]: *** [CMakeFiles/micrortps_agent.dir/src/micrortps_agent/BatteryStatus_Publisher.cpp.o] Error 1
make[2]: *** [CMakeFiles/micrortps_agent.dir/src/micrortps_agent/Cpuload_Publisher.cpp.o] Error 1
make[1]: *** [CMakeFiles/micrortps_agent.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< px4_ros_com [1min 55s, exited with code 2]

Summary: 1 package finished [9min 1s]
  1 package failed: px4_ros_com
  1 package had stderr output: px4_ros_com
`

McKayCyborg, 2020-07-28:
UPDATE:

I've just attempted this build again with the following changes to my system:

Hardware: Raspberry Pi 4
CPU Architecture: aarch64
Operating System: Ubuntu 20.04 Server (headless)
ROS Version: Foxy
Fast-CDR Version: 1.0.14
Fast-RTPS Version: 1.10.0
Fast-RTPS-Gen Version: 1.0.4

And I am still experiencing a very similar error during the building stage of px4_ros_com:

'make[2]: *** [CMakeFiles/micrortps_agent.dir/build.make:1600: CMakeFiles/micrortps_agent.dir/src/micrortps_agent/Airspeed_Publisher.cpp.o] Error 1
/home/ubuntu/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/BatteryStatus_Publisher.cpp: In member function ‘bool BatteryStatus_Publisher::init()’:
/home/ubuntu/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/BatteryStatus_Publisher.cpp:66:25: error: ‘class eprosima::fastrtps::rtps::BuiltinAttributes’ has no member named ‘domainId’
   66 |     PParam.rtps.builtin.domainId = 0;'
`

TSC21, 2020-07-28:
Fix in https://github.com/PX4/Firmware/pull/15354/commits/a7112c4f7514908d0643b7d3d5373c7517ebd7c7.
You can manually set it on your side or wait that the fix gets in. Thanks for reporting.

closed 2020-07-28 by TSC21

McKayCyborg, 2020-07-28:
Thankyou so much for your swift update. I will let you know how the changes go.