PX4/px4_ros_com #52: Topic type format

stmoon, 2020-07-29:
Hi,

When I use px4_ros_com, I found the topic type is different from ROS.

For example, I thought that valid topic type is "px4_msgs/msg/AdcReport" not px4_msgs::msg::AdcReport/

So when I test it, I found the error message lilke below

```
root@stmoon-XPS-13-9370:~/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent# ros2 topic echo /Airspeed_PubSubTopic
The passed message type is invalid
```

How can I change the message type?

```
root@stmoon-XPS-13-9370:/# ros2 topic list -t
/AdcReport_PubSubTopic [px4_msgs::msg::AdcReport]
/Airspeed_PubSubTopic [px4_msgs::msg::Airspeed]
/BatteryStatus_PubSubTopic [px4_msgs::msg::BatteryStatus]
/CameraCapture_PubSubTopic [px4_msgs::msg::CameraCapture]
/CameraTrigger_PubSubTopic [px4_msgs::msg::CameraTrigger]
/CollisionConstraints_PubSubTopic [px4_msgs::msg::CollisionConstraints]
/CollisionReport_PubSubTopic [px4_msgs::msg::CollisionReport]
/Cpuload_PubSubTopic [root@stmoon-XPS-13-9370:/# ros2 topic list -t
/AdcReport_PubSubTopic [px4_msgs::msg::AdcReport]
/Airspeed_PubSubTopic [px4_msgs::msg::Airspeed]
/BatteryStatus_PubSubTopic [px4_msgs::msg::BatteryStatus]
/CameraCapture_PubSubTopic [px4_msgs::msg::CameraCapture]
/CameraTrigger_PubSubTopic [px4_msgs::msg::CameraTrigger]
/CollisionConstraints_PubSubTopic [px4_msgs::msg::CollisionConstraints]
/CollisionReport_PubSubTopic [px4_msgs::msg::CollisionReport]
/Cpuload_PubSubTopic [px4_msgs::msg::Cpuload]
/DebugArray_PubSubTopic [px4_msgs::msg::DebugArray]
/DebugKeyValue_PubSubTopic [px4_msgs::msg::DebugKeyValue]
/DebugValue_PubSubTopic [px4_msgs::msg::DebugValue]
/DebugVect_PubSubTopic [px4_msgs::msg::DebugVect]
/DistanceSensor_PubSubTopic [px4_msgs::msg::DistanceSensor]
/EstimatorStatus_PubSubTopic [px4_msgs::msg::EstimatorStatus]
/HomePosition_PubSubTopic [px4_msgs::msg::HomePosition]
/IridiumsbdStatus_PubSubTopic [px4_msgs::msg::IridiumsbdStatus]
/ObstacleDistance_PubSubTopic [px4_msgs::msg::ObstacleDistance]
/OnboardComputerStatus_PubSubTopic [px4_msgs::msg::OnboardComputerStatus]
/OpticalFlow_PubSubTopic [px4_msgs::msg::OpticalFlow]
/PositionSetpointTriplet_PubSubTopic [px4_msgs::msg::PositionSetpointTriplet]
/PositionSetpoint_PubSubTopic [px4_msgs::msg::PositionSetpoint]
/RadioStatus_PubSubTopic [px4_msgs::msg::RadioStatus]
/SatelliteInfo_PubSubTopic [px4_msgs::msg::SatelliteInfo]
/SensorBaro_PubSubTopic [px4_msgs::msg::SensorBaro]
/SensorCombined_PubSubTopic [px4_msgs::msg::SensorCombined]
/SensorSelection_PubSubTopic [px4_msgs::msg::SensorSelection]
/Timesync_PubSubTopic [px4_msgs::msg::Timesync]
/TrajectoryWaypoint_PubSubTopic [px4_msgs::msg::TrajectoryWaypoint]
/VehicleAttitude_PubSubTopic [px4_msgs::msg::VehicleAttitude]
/VehicleMocapOdometry_PubSubTopic [px4_msgs::msg::VehicleMocapOdometry]
/VehicleOdometry_PubSubTopic [px4_msgs::msg::VehicleOdometry]
/VehicleTrajectoryWaypoint_PubSubTopic [px4_msgs::msg::VehicleTrajectoryWaypoint]
/VehicleVisualOdometry_PubSubTopic [px4_msgs::msg::VehicleVisualOdometry]
/VtolVehicleStatus_PubSubTopic [px4_msgs::msg::VtolVehicleStatus]
/WindEstimate_PubSubTopic [px4_msgs::msg::WindEstimate]
/parameter_events [rcl_interfaces/msg/ParameterEvent]
/rosout [rcl_interfaces/msg/Log]]
/DebugArray_PubSubTopic [px4_msgs::msg::DebugArray]
/DebugKeyValue_PubSubTopic [px4_msgs::msg::DebugKeyValue]
/DebugValue_PubSubTopic [px4_msgs::msg::DebugValue]
/DebugVect_PubSubTopic [px4_msgs::msg::DebugVect]
/DistanceSensor_PubSubTopic [px4_msgs::msg::DistanceSensor]
/EstimatorStatus_PubSubTopic [px4_msgs::msg::EstimatorStatus]
/HomePosition_PubSubTopic [px4_msgs::msg::HomePosition]
/IridiumsbdStatus_PubSubTopic [px4_msgs::msg::IridiumsbdStatus]
/ObstacleDistance_PubSubTopic [px4_msgs::msg::ObstacleDistance]
/OnboardComputerStatus_PubSubTopic [px4_msgs::msg::OnboardComputerStatus]
/OpticalFlow_PubSubTopic [px4_msgs::msg::OpticalFlow]
/PositionSetpointTriplet_PubSubTopic [px4_msgs::msg::PositionSetpointTriplet]
/PositionSetpoint_PubSubTopic [px4_msgs::msg::PositionSetpoint]
/RadioStatus_PubSubTopic [px4_msgs::msg::RadioStatus]
/SatelliteInfo_PubSubTopic [px4_msgs::msg::SatelliteInfo]
/SensorBaro_PubSubTopic [px4_msgs::msg::SensorBaro]
/SensorCombined_PubSubTopic [px4_msgs::msg::SensorCombined]
/SensorSelection_PubSubTopic [px4_msgs::msg::SensorSelection]
/Timesync_PubSubTopic [px4_msgs::msg::Timesync]
/TrajectoryWaypoint_PubSubTopic [px4_msgs::msg::TrajectoryWaypoint]
/VehicleAttitude_PubSubTopic [px4_msgs::msg::VehicleAttitude]
/VehicleMocapOdometry_PubSubTopic [px4_msgs::msg::VehicleMocapOdometry]
/VehicleOdometry_PubSubTopic [px4_msgs::msg::VehicleOdometry]
/VehicleTrajectoryWaypoint_PubSubTopic [px4_msgs::msg::VehicleTrajectoryWaypoint]
/VehicleVisualOdometry_PubSubTopic [px4_msgs::msg::VehicleVisualOdometry]
/VtolVehicleStatus_PubSubTopic [px4_msgs::msg::VtolVehicleStatus]
/WindEstimate_PubSubTopic [px4_msgs::msg::WindEstimate]
/parameter_events [rcl_interfaces/msg/ParameterEvent]
/rosout [rcl_interfaces/msg/Log]
```

stmoon, 2020-07-29:
I found this problem can be solved if the version of fastrtpsgen is more than v1.0.4.

Thanks.

closed 2020-07-29 by stmoon

TSC21, 2020-07-29:
> I found this problem can be solved if the version of fastrtpsgen is more than v1.0.4.
> 
> Thanks.

It needs to be at least 1.0.4

tank104, 2022-08-18:
I have got a similiar issue to this when doing fresh build of Ubuntu 22.04 + ROS2 Humble + PX4.

I have tried fastrtpsgen v1.0.4.  Also tried the latest, and always my messages are showing wrong.

Any idea how I can fix this?

$ ros2 topic list -t
/fmu/collision_constraints/out [px4_msgs::msg::CollisionConstraints]
/fmu/debug_array/in [px4_msgs::msg::DebugArray]
/fmu/debug_key_value/in [px4_msgs::msg::DebugKeyValue]
/fmu/debug_value/in [px4_msgs::msg::DebugValue]
/fmu/debug_vect/in [px4_msgs::msg::DebugVect]
/fmu/offboard_control_mode/in [px4_msgs::msg::OffboardControlMode]
/fmu/onboard_computer_status/in [px4_msgs::msg::OnboardComputerStatus]
/fmu/position_setpoint/in [px4_msgs::msg::PositionSetpoint]
/fmu/position_setpoint_triplet/in [px4_msgs::msg::PositionSetpointTriplet]
/fmu/sensor_combined/out [px4_msgs::msg::SensorCombined]
/fmu/sensor_optical_flow/in [px4_msgs::msg::SensorOpticalFlow]
/fmu/telemetry_status/in [px4_msgs::msg::TelemetryStatus]
/fmu/timesync/in [px4_msgs::msg::Timesync]
/fmu/timesync/out [px4_msgs::msg::Timesync]
/fmu/trajectory_bezier/in [px4_msgs::msg::TrajectoryBezier]
/fmu/trajectory_setpoint/in [px4_msgs::msg::TrajectorySetpoint]
/fmu/trajectory_waypoint/out [px4_msgs::msg::TrajectoryWaypoint]
/fmu/vehicle_command/in [px4_msgs::msg::VehicleCommand]
/fmu/vehicle_control_mode/out [px4_msgs::msg::VehicleControlMode]
/fmu/vehicle_local_position_setpoint/in [px4_msgs::msg::VehicleLocalPositionSetpoint]
/fmu/vehicle_mocap_odometry/in [px4_msgs::msg::VehicleMocapOdometry]
/fmu/vehicle_odometry/out [px4_msgs::msg::VehicleOdometry]
/fmu/vehicle_status/out [px4_msgs::msg::VehicleStatus]
/fmu/vehicle_trajectory_bezier/in [px4_msgs::msg::VehicleTrajectoryBezier]
/fmu/vehicle_trajectory_waypoint/in [px4_msgs::msg::VehicleTrajectoryWaypoint]
/fmu/vehicle_trajectory_waypoint_desired/out [px4_msgs::msg::VehicleTrajectoryWaypointDesired]
/fmu/vehicle_visual_odometry/in [px4_msgs::msg::VehicleVisualOdometry]
/parameter_events [rcl_interfaces/msg/ParameterEvent]
/rosout [rcl_interfaces/msg/Log]
/timesync_status [px4_msgs::msg::TimesyncStatus]

Posted similiar question here: https://github.com/PX4/micrortps_agent/issues/2

nigeltiany, 2022-08-22:
This issue is related to using ros humble from source. The workaround is to force the generator to generate ros2 types 
Change this line https://github.com/PX4/PX4-Autopilot/blob/d5d88cba5ba37f0d6583544d5fce32977c4b635c/msg/tools/generate_microRTPS_bridge.py#L357

from
```
    gen_ros2_typename = ""
    if ros2_distro and ros2_distro in ['dashing', 'eloquent', 'foxy', 'galactic', 'humble', 'rolling'] and fastrtpsgen_version >= version.Version("1.0.4"):
        gen_ros2_typename = "-typeros2 "
```
to
```
    # gen_ros2_typename = ""
    # if ros2_distro and ros2_distro in ['dashing', 'eloquent', 'foxy', 'galactic', 'humble', 'rolling'] and fastrtpsgen_version >= version.Version("1.0.4"):
    gen_ros2_typename = "-typeros2 "
```
this forces the generator to append the ```-typeros2``` flag. The flag to remove the conflicts causing the data not to pass through

nigeltiany, 2022-08-22:
Actually never mind. Before the modification, ensure the ROS_DISTRO environment variable is set to rolling or humble
and ROS_VERSION is set to 2

tank104, 2022-08-28:
Thanks @nigeltiany - Literally that same day before you posted I reinstalled Ubuntu 20.05 and Galactic, as I am still new to this, and realised using an unsupported version just now was probably not best.
Thanks for replying though, and will try it soon.
Will look to go back when supported though.