PX4/px4_ros_com #54: px4_ros_com ROS2 foxy build failure

InigoMonreal, 2020-08-25:
Hello everyone,
I'm facing the following error in _**Building the Workspaces step 3.**_ in the [PX4 Dev Guide](https://dev.px4.io/master/en/middleware/micrortps.html) when trying to build the px4_ros_com package for ROS2 foxy.

`Starting >>> px4_msgs
Finished <<< px4_msgs [3.26s]                     
Starting >>> px4_ros_com
--- stderr: px4_ros_com                            
CMakeFiles/micrortps_agent.dir/build.make:353: warning: overriding recipe for target '/home/user/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/Timesync.cpp'
CMakeFiles/micrortps_agent.dir/build.make:182: warning: ignoring old recipe for target '/home/user/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/Timesync.cpp'
CMakeFiles/micrortps_agent.dir/build.make:356: warning: overriding recipe for target '/home/user/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/TimesyncPubSubTypes.cpp'
CMakeFiles/micrortps_agent.dir/build.make:185: warning: ignoring old recipe for target '/home/user/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/TimesyncPubSubTypes.cpp'
CMakeFiles/micrortps_agent.dir/build.make:359: warning: overriding recipe for target '/home/user/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/TimesyncPubSubTypes.h'
CMakeFiles/micrortps_agent.dir/build.make:188: warning: ignoring old recipe for target '/home/user/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/TimesyncPubSubTypes.h'
openjdk version "14.0.1" 2020-04-14
OpenJDK Runtime Environment (build 14.0.1+7-Ubuntu-1ubuntu1)
OpenJDK 64-Bit Server VM (build 14.0.1+7-Ubuntu-1ubuntu1, mixed mode, sharing)
/sbin/ldconfig.real: Can't stat /usr/local/lib/i386-linux-gnu: No such file or directory
/sbin/ldconfig.real: Path `/usr/lib/i386-linux-gnu' given more than once
/sbin/ldconfig.real: Can't stat /usr/local/lib/i686-linux-gnu: No such file or directory
/sbin/ldconfig.real: Can't stat /lib/i686-linux-gnu: No such file or directory
/sbin/ldconfig.real: Can't stat /usr/lib/i686-linux-gnu: No such file or directory
/sbin/ldconfig.real: Can't stat /usr/local/lib/x86_64-linux-gnu: No such file or directory
/sbin/ldconfig.real: Path `/usr/lib/x86_64-linux-gnu' given more than once
/sbin/ldconfig.real: Path `/lib/x86_64-linux-gnu' given more than once
/sbin/ldconfig.real: Path `/usr/lib/x86_64-linux-gnu' given more than once
/sbin/ldconfig.real: Path `/usr/lib' given more than once
/sbin/ldconfig.real: /lib/i386-linux-gnu/ld-2.31.so is the dynamic linker, ignoring

/sbin/ldconfig.real: /lib/x86_64-linux-gnu/ld-2.31.so is the dynamic linker, ignoring

/sbin/ldconfig.real: Can't create temporary cache file /etc/ld.so.cache~: Permission denied
Traceback (most recent call last):
  File "/home/user/px4_ros_com_ros2/src/px4_ros_com/scripts/generate_microRTPS_bridge.py", line 313, in <module>
    classifier = (Classifier(os.path.abspath(args.yaml_file), msg_dir) if os.path.isabs(args.yaml_file)
  File "/home/user/px4_ros_com_ros2/src/px4_ros_com/scripts/uorb_rtps_classifier.py", line 55, in __init__
    self.check_if_listed(yaml_file)
  File "/home/user/px4_ros_com_ros2/src/px4_ros_com/scripts/uorb_rtps_classifier.py", line 139, in check_if_listed
    raise AssertionError(
AssertionError: 
The following messages are not listen under  /home/user/px4_ros_com_ros2/src/px4_ros_com/templates/uorb_rtps_message_ids.yaml: SensorPreflightImu, TelemetryHeartbeat, SensorPreflightMag, Px4IoStatus

Please add them to the yaml file with the respective ID and, if applicable, mark them to be sent or received by the micro-RTPS bridge.
NOTE: If the message has multi-topics (#TOPICS), these should be added as well.

make[2]: *** [CMakeFiles/micrortps_agent.dir/build.make:62: /home/user/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/microRTPS_agent.cpp] Error 1
make[1]: *** [CMakeFiles/Makefile2:112: CMakeFiles/micrortps_agent.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
In file included from /opt/ros/foxy/include/rcl/types.h:18:0,
                 from /opt/ros/foxy/include/rcl/arguments.h:20,
                 from /opt/ros/foxy/include/rcl/context.h:26,
                 from /opt/ros/foxy/include/rcl/guard_condition.h:24,
                 from /opt/ros/foxy/include/rclcpp/executor.hpp:29,
                 from /opt/ros/foxy/include/rclcpp/executors/multi_threaded_executor.hpp:25,
                 from /opt/ros/foxy/include/rclcpp/executors.hpp:21,
                 from /opt/ros/foxy/include/rclcpp/rclcpp.hpp:146,
                 from /home/user/px4_ros_com_ros2/src/px4_ros_com/src/listeners/vehicle_gps_position_listener.cpp:40:
/opt/ros/foxy/include/rmw/types.h:412:3: error: expected '}' before '__attribute__'
   RMW_DECLARE_DEPRECATED(
   ^
/opt/ros/foxy/include/rmw/types.h:414:62: error: expected unqualified-id before '=' token
     RMW_QOS_POLICY_LIVELINESS_MANUAL_BY_NODE_DEPRECATED_MSG) = 2,
                                                              ^
In file included from /opt/ros/foxy/include/rclcpp/executor.hpp:29:0,
                 from /opt/ros/foxy/include/rclcpp/executors/multi_threaded_executor.hpp:25,
                 from /opt/ros/foxy/include/rclcpp/executors.hpp:21,
                 from /opt/ros/foxy/include/rclcpp/rclcpp.hpp:146,
                 from /home/user/px4_ros_com_ros2/src/px4_ros_com/src/listeners/vehicle_gps_position_listener.cpp:40:
/opt/ros/foxy/include/rcl/guard_condition.h:262:1: error: expected declaration before '}' token
 }
 ^
[... 38 lines cut ...]
In file included from /opt/ros/foxy/include/rclcpp/executor.hpp:29:0,
                 from /opt/ros/foxy/include/rclcpp/executors/multi_threaded_executor.hpp:25,
                 from /opt/ros/foxy/include/rclcpp/executors.hpp:21,
                 from /opt/ros/foxy/include/rclcpp/rclcpp.hpp:146,
                 from /home/user/px4_ros_com_ros2/src/px4_ros_com/src/listeners/sensor_combined_listener.cpp:42:
/opt/ros/foxy/include/rcl/guard_condition.h:262:1: error: expected declaration before '}' token
 }
 ^
make[2]: *** [CMakeFiles/vehicle_gps_position_listener.dir/build.make:63: CMakeFiles/vehicle_gps_position_listener.dir/src/listeners/vehicle_gps_position_listener.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:140: CMakeFiles/vehicle_gps_position_listener.dir/all] Error 2
make[2]: *** [CMakeFiles/debug_vect_advertiser.dir/build.make:63: CMakeFiles/debug_vect_advertiser.dir/src/advertisers/debug_vect_advertiser.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:167: CMakeFiles/debug_vect_advertiser.dir/all] Error 2
make[2]: *** [CMakeFiles/sensor_combined_listener.dir/build.make:63: CMakeFiles/sensor_combined_listener.dir/src/listeners/sensor_combined_listener.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:248: CMakeFiles/sensor_combined_listener.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< px4_ros_com [0.52s, exited with code 2]

Summary: 1 package finished [3.86s]
  1 package failed: px4_ros_com
  1 package had stderr output: px4_ros_com
`
Is this a problem on my end that I could resolve? (And if so, how?)
And if it is not on my end, as far as I know a fix for the px4_ros_com package is being developed, but does anyone have any suggestions other than to wait?
Thank you in advance.

TSC21, 2020-08-25:
@InigoMonreal you seem to have a lot of different problems above.

```
The following messages are not listen under  /home/user/px4_ros_com_ros2/src/px4_ros_com/templates/uorb_rtps_message_ids.yaml: SensorPreflightImu, TelemetryHeartbeat, SensorPreflightMag, Px4IoStatus
```

Your `uorb_rtps_message_ids.yaml` in px4_ros_com seems to be out of sync with the current existing message definitions. Are you using `px4_ros_com` and `px4_msgs` latest master?

```
In file included from /opt/ros/foxy/include/rclcpp/executor.hpp:29:0,
                 from /opt/ros/foxy/include/rclcpp/executors/multi_threaded_executor.hpp:25,
                 from /opt/ros/foxy/include/rclcpp/executors.hpp:21,
                 from /opt/ros/foxy/include/rclcpp/rclcpp.hpp:146,
                 from /home/user/px4_ros_com_ros2/src/px4_ros_com/src/listeners/sensor_combined_listener.cpp:42:
/opt/ros/foxy/include/rcl/guard_condition.h:262:1: error: expected declaration before '}' token
 }
 ^
```

Even though it's failing the test (needs investigation), you can see that the overall build passes correctlu: https://travis-ci.com/github/PX4/px4_ros_com/jobs/376167062. I don't have an idea of what's causing that problem.

InigoMonreal, 2020-08-25:
Thanks for the prompt reply, @TSC21.

Good to know that the problem seems to be at my end. I manually cloned the packages from the master again and I can't see the yaml error anymore, but the px4_ros_com package is still failing to build (same errors as above, I believe), yielding the following:

```
[ 97%] Linking CXX executable micrortps_agent                   
/home/linuxbrew/.linuxbrew/bin/ld: /opt/ros/foxy/lib/libfastrtps.so.2.0.1: undefined reference to `std::__cxx11::basic_stringstream<char, std::char_traits<char>, std::allocator<char> >::basic_stringstream()@GLIBCXX_3.4.26'
/home/linuxbrew/.linuxbrew/bin/ld: /opt/ros/foxy/lib/libfastrtps.so.2.0.1: undefined reference to `__cxa_init_primary_exception@CXXABI_1.3.11'
/home/linuxbrew/.linuxbrew/bin/ld: /opt/ros/foxy/lib/libfastrtps.so.2.0.1: undefined reference to `typeinfo for std::thread::_State@GLIBCXX_3.4.22'
/home/linuxbrew/.linuxbrew/bin/ld: /opt/ros/foxy/lib/libfastrtps.so.2.0.1: undefined reference to `std::thread::_M_start_thread(std::unique_ptr<std::thread::_State, std::default_delete<std::thread::_State> >, void (*)())@GLIBCXX_3.4.22'
/home/linuxbrew/.linuxbrew/bin/ld: /opt/ros/foxy/lib/libfastrtps.so.2.0.1: undefined reference to `std::thread::_State::~_State()@GLIBCXX_3.4.22'
/home/linuxbrew/.linuxbrew/bin/ld: /opt/ros/foxy/lib/libfastrtps.so.2.0.1: undefined reference to `std::__exception_ptr::exception_ptr::exception_ptr(void*)@CXXABI_1.3.11'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/micrortps_agent.dir/build.make:2288: micrortps_agent] Error 1
make[1]: *** [CMakeFiles/Makefile2:113: CMakeFiles/micrortps_agent.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
--- stderr: px4_ros_com
openjdk version "14.0.1" 2020-04-14
OpenJDK Runtime Environment (build 14.0.1+7-Ubuntu-1ubuntu1)
OpenJDK 64-Bit Server VM (build 14.0.1+7-Ubuntu-1ubuntu1, mixed mode, sharing)
CMake Deprecation Warning at /opt/ros/foxy/share/ament_cmake_export_interfaces/cmake/ament_export_interfaces.cmake:37 (message):
  ament_export_interfaces() is deprecated, use ament_export_targets() instead
Call Stack (most recent call first):
  CMakeLists.txt:109 (ament_export_interfaces)

CMakeFiles/micrortps_agent.dir/build.make:353: warning: overriding recipe for target '/home/user/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/Timesync.cpp'
CMakeFiles/micrortps_agent.dir/build.make:182: warning: ignoring old recipe for target '/home/user/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/Timesync.cpp'
CMakeFiles/micrortps_agent.dir/build.make:356: warning: overriding recipe for target '/home/user/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/TimesyncPubSubTypes.cpp'
CMakeFiles/micrortps_agent.dir/build.make:185: warning: ignoring old recipe for target '/home/user/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/TimesyncPubSubTypes.cpp'
CMakeFiles/micrortps_agent.dir/build.make:359: warning: overriding recipe for target '/home/user/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/TimesyncPubSubTypes.h'
CMakeFiles/micrortps_agent.dir/build.make:188: warning: ignoring old recipe for target '/home/user/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/TimesyncPubSubTypes.h'
openjdk version "14.0.1" 2020-04-14
OpenJDK Runtime Environment (build 14.0.1+7-Ubuntu-1ubuntu1)
OpenJDK 64-Bit Server VM (build 14.0.1+7-Ubuntu-1ubuntu1, mixed mode, sharing)
In file included from /opt/ros/foxy/include/rcl/types.h:18:0,
                 from /opt/ros/foxy/include/rcl/arguments.h:20,
                 from /opt/ros/foxy/include/rcl/context.h:26,
                 from /opt/ros/foxy/include/rcl/guard_condition.h:24,
                 from /opt/ros/foxy/include/rclcpp/executor.hpp:29,
                 from /opt/ros/foxy/include/rclcpp/executors/multi_threaded_executor.hpp:25,
                 from /opt/ros/foxy/include/rclcpp/executors.hpp:21,
                 from /opt/ros/foxy/include/rclcpp/rclcpp.hpp:146,
                 from /home/user/px4_ros_com_ros2/src/px4_ros_com/src/listeners/vehicle_gps_position_listener.cpp:40:
/opt/ros/foxy/include/rmw/types.h:412:3: error: expected '}' before '__attribute__'
   RMW_DECLARE_DEPRECATED(
   ^
/opt/ros/foxy/include/rmw/types.h:414:62: error: expected unqualified-id before '=' token
     RMW_QOS_POLICY_LIVELINESS_MANUAL_BY_NODE_DEPRECATED_MSG) = 2,
                                                              ^
In file included from /opt/ros/foxy/include/rclcpp/executor.hpp:29:0,
                 from /opt/ros/foxy/include/rclcpp/executors/multi_threaded_executor.hpp:25,
                 from /opt/ros/foxy/include/rclcpp/executors.hpp:21,
                 from /opt/ros/foxy/include/rclcpp/rclcpp.hpp:146,
                 from /home/user/px4_ros_com_ros2/src/px4_ros_com/src/listeners/vehicle_gps_position_listener.cpp:40:
/opt/ros/foxy/include/rcl/guard_condition.h:262:1: error: expected declaration before '}' token
 }
 ^
In file included from /opt/ros/foxy/include/rcl/types.h:18:0,
                 from /opt/ros/foxy/include/rcl/arguments.h:20,
                 from /opt/ros/foxy/include/rcl/context.h:26,
                 from /opt/ros/foxy/include/rcl/guard_condition.h:24,
                 from /opt/ros/foxy/include/rclcpp/executor.hpp:29,
                 from /opt/ros/foxy/include/rclcpp/executors/multi_threaded_executor.hpp:25,
                 from /opt/ros/foxy/include/rclcpp/executors.hpp:21,
                 from /opt/ros/foxy/include/rclcpp/rclcpp.hpp:146,
[... cut ...]
make[2]: *** [CMakeFiles/micrortps_agent.dir/build.make:2288: micrortps_agent] Error 1
make[1]: *** [CMakeFiles/Makefile2:113: CMakeFiles/micrortps_agent.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< px4_ros_com [1min 26s, exited with code 2]
                                      
Summary: 1 package finished [3min 45s]
  1 package failed: px4_ros_com
  2 packages had stderr output: px4_msgs px4_ros_com
 
```
I'll keep trying, but if you have any idea or suggestion over the coming days, feel free to share it.
Thanks and have a good week.

InigoMonreal, 2020-08-25:
Okay, so just for reference for other people (not sure of what solved it but this is what I did to make it work):
Since my previous error message showed a lot of undefined references pointing to linuxbrew, I uninstalled Homebrew for Linux (which I had used to install Gradle for the Fast-RTPS-Gen installation) and reinstalled Gradle using SDKMAN.
I then deleted the px4_ros_com_ros2 directory and started over again.
And just like that, this time it decided to work. Thanks for the help and keep up the good work.

closed 2020-08-25 by TSC21