PX4/px4_ros_com #55: Build failing for ROS2 workspace

Sanjayso27, 2020-09-13:
Running the script `build_ros2_workspace.bash ` in the workspace fails with the error:
```bash
AssertionError: 
The following message is not listen under  /root/dev_ws/src/px4_ros_com/templates/uorb_rtps_message_ids.yaml: SensorPreflightImu

Please add them to the yaml file with the respective ID and, if applicable, mark them to be sent or received by the micro-RTPS bridge.
NOTE: If the message has multi-topics (#TOPICS), these should be added as well.

CMake Error at cmake/GenerateMicroRTPSAgent.cmake:132 (string):
string sub-command FIND requires 3 or 4 parameters.

```
**System:-**
Ubuntu 18.04
ROS2 Dashing
FastRTPS 1.8.2
```bash
fastrtpsgen -version
openjdk version "1.8.0_265"
OpenJDK Runtime Environment (build 1.8.0_265-8u265-b01-0ubuntu2~18.04-b01)
OpenJDK 64-Bit Server VM (build 25.265-b01, mixed mode)
fastrtpsgen version 1.0.4
```
Installed FastRTPSGen & FastRTPS (from source) following this guide  https://dev.px4.io/master/en/setup/fast-rtps-installation.html

Followed https://dev.px4.io/master/en/middleware/micrortps.html to build the workspace.

**Full Error Output**
```bash
root@fbdacbc26a95:~/dev_ws/src/px4_ros_com/scripts# ./build_ros2_workspace.bash 
openjdk version "1.8.0_265"
OpenJDK Runtime Environment (build 1.8.0_265-8u265-b01-0ubuntu2~18.04-b01)
OpenJDK 64-Bit Server VM (build 25.265-b01, mixed mode)
Starting >>> px4_msgs
Finished <<< px4_msgs [14.0s]                         
Starting >>> px4_ros_com
--- stderr: px4_ros_com                          
openjdk version "1.8.0_265"
OpenJDK Runtime Environment (build 1.8.0_265-8u265-b01-0ubuntu2~18.04-b01)
OpenJDK 64-Bit Server VM (build 25.265-b01, mixed mode)
Traceback (most recent call last):
  File "/root/dev_ws/src/px4_ros_com/scripts/uorb_rtps_classifier.py", line 232, in <module>
    classifier = (Classifier(os.path.abspath(args.yaml_file), msg_dir) if os.path.isabs(args.yaml_file)
  File "/root/dev_ws/src/px4_ros_com/scripts/uorb_rtps_classifier.py", line 55, in __init__
    self.check_if_listed(yaml_file)
  File "/root/dev_ws/src/px4_ros_com/scripts/uorb_rtps_classifier.py", line 142, in check_if_listed
    "to be sent or received by the micro-RTPS bridge.\n"
AssertionError: 
The following message is not listen under  /root/dev_ws/src/px4_ros_com/templates/uorb_rtps_message_ids.yaml: SensorPreflightImu

Please add them to the yaml file with the respective ID and, if applicable, mark them to be sent or received by the micro-RTPS bridge.
NOTE: If the message has multi-topics (#TOPICS), these should be added as well.

CMake Error at cmake/GenerateMicroRTPSAgent.cmake:106 (string):
  string sub-command FIND requires 3 or 4 parameters.
Call Stack (most recent call first):
  CMakeLists.txt:60 (include)

Traceback (most recent call last):
  File "/root/dev_ws/src/px4_ros_com/scripts/uorb_rtps_classifier.py", line 232, in <module>
    classifier = (Classifier(os.path.abspath(args.yaml_file), msg_dir) if os.path.isabs(args.yaml_file)
  File "/root/dev_ws/src/px4_ros_com/scripts/uorb_rtps_classifier.py", line 55, in __init__
    self.check_if_listed(yaml_file)
  File "/root/dev_ws/src/px4_ros_com/scripts/uorb_rtps_classifier.py", line 142, in check_if_listed
    "to be sent or received by the micro-RTPS bridge.\n"
AssertionError: 
The following message is not listen under  /root/dev_ws/src/px4_ros_com/templates/uorb_rtps_message_ids.yaml: SensorPreflightImu

Please add them to the yaml file with the respective ID and, if applicable, mark them to be sent or received by the micro-RTPS bridge.
NOTE: If the message has multi-topics (#TOPICS), these should be added as well.

CMake Error at cmake/GenerateMicroRTPSAgent.cmake:132 (string):
  string sub-command FIND requires 3 or 4 parameters.
Call Stack (most recent call first):
  CMakeLists.txt:60 (include)

---
Failed   <<< px4_ros_com [2.99s, exited with code 1]

Summary: 1 package finished [18.3s]
  1 package failed: px4_ros_com
  1 package had stderr output: px4_ros_com

```

CarmChao, 2020-10-01:
I have the same problem，have you solved it？

Lien182, 2020-10-19:
Some news here?

Lien182, 2020-10-19:
I had a similar issue today. For me, the problem was that I worked with the master branch of px4_ros_com and px4_msgs. During my experiments, the px4_msgs repo was updated while px4_ros_com not. This led to missing messages in the px4_ros_com package.

TSC21, 2020-11-28:
As said by @Lien182, you need to have the `px4_msgs` and `px4_ros_com` in sync.

closed 2020-11-28 by TSC21