PX4/px4_ros_com #59: MicroRTPS Agent

wrgtri, 2020-12-28:
I am using Ubuntu 20.04 LTS and ROS2 foxy. I have followed [this guide](https://dev.px4.io/master/en/middleware/micrortps.html) and have installed Fast-RTPS 1.8.2 and Fast-RTPS-Gen 1.0.2. Yet when I run ./build_ros2_workspace.bash, I receive the error below. I tried git cloning the [micrortps folder](https://github.com/PX4/micrortps_agent) to the specified directory and placing microRTPS_agent.cpp where specified, but when run again, it automatically deletes microRTPS_agent.cpp. How do I solve this issue?

```
pi4@pi4-desktop:~/px4_ros_com_ros2/src/px4_ros_com/scripts$ ./build_ros2_workspace.bash 

openjdk version "1.8.0_275"
OpenJDK Runtime Environment (build 1.8.0_275-8u275-b01-0ubuntu1~20.04-b01)
OpenJDK 64-Bit Server VM (build 25.275-b01, mixed mode)
Starting >>> px4_msgs
[Processing: px4_msgs]                              
Finished <<< px4_msgs [46.4s]                    
Starting >>> px4_ros_com
--- stderr: px4_ros_com                              
make[2]: *** [CMakeFiles/micrortps_agent.dir/build.make:62: /home/pi4/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/microRTPS_agent.cpp] Error 1
make[1]: *** [CMakeFiles/Makefile2:113: CMakeFiles/micrortps_agent.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< px4_ros_com [3.20s, exited with code 2]

Summary: 1 package finished [50.2s]
  1 package failed: px4_ros_com
  1 package had stderr output: px4_ros_com
```

TSC21, 2020-12-28:
Can you please use the `--verbose` option on the bash script? Otherwise is kinda difficult to understand where is it failing. Also note that for ROS2 Foxy you should use FastDDS 2.0.0, not 1.8.2.

wrgtri, 2020-12-29:
Thank you, I am new to all of this, and did not know about the --verbose option. It allowed me to fix this error I had been stuck on for about a week by seeing it was a simple missing package error. I was able to install it and the build finished with "ROS2 workspace ready..."

I am now following the first example [here](https://dev.px4.io/master/en/middleware/micrortps.html) about the Fast RTPS Listener application. When running `fastrtpsgen -example x64Linux2.6gcc ../micrortps_client/micrortps_agent/idl/sensor_combined.idl` I get the error that this file does not exist (detailed below).

```
openjdk version "1.8.0_275"
OpenJDK Runtime Environment (build 1.8.0_275-8u275-b01-0ubuntu1~20.04-b01)
OpenJDK 64-Bit Server VM (build 25.275-b01, mixed mode)
Loading templates...
Processing the file ../micrortps_client/micrortps_agent/idl/sensor_combined.idl...
cpp: error: ../micrortps_client/micrortps_agent/idl/sensor_combined.idl: No such file or directory
cpp: warning: ‘-x c’ after last input file has no effect
cpp: fatal error: no input files
compilation terminated.
ERROR<callPreprocessor>: Preprocessor return an error 1
Exception in thread "main" java.lang.NullPointerException
        at com.eprosima.fastrtps.fastrtpsgen.execute(fastrtpsgen.java:343)
        at com.eprosima.fastrtps.fastrtpsgen.main(fastrtpsgen.java:1207)
```

How do I get this file to continue with the example? Before running this command, I cloned PX4_AutoPilot, ran the ubuntu.sh, and built it with `make px4_sitl_rtps`. Then I attempted the example mentioned.

Thank you again for your help. I am new to this and very confused.

TSC21, 2020-12-29:
The dir is actually wrong. The docs need to be updated. Did you check what files exist in `build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_agent/idl/`?

TSC21, 2020-12-29:
Also note that that example will not work with the agent generated by px4_ros_com because it is only compatible with the ROS2 types/topics. It will only work with the agent generated from the PX4 Firmware `rtps` target.

wrgtri, 2020-12-29:
> The dir is actually wrong. The docs need to be updated. Did you check what files exist in `build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_agent/idl/`?

Ah I see it now!

> Also note that that example will not work with the agent generated by px4_ros_com because it is only compatible with the ROS2 types/topics. It will only work with the agent generated from the PX4 Firmware `rtps` target.

How do I go about generating an agent from the PX4 Firmware? Is there some documentation I could follow on this?

TSC21, 2020-12-29:
> Ah I see it now!
>
>
> How do I go about generating an agent from the PX4 Firmware? Is there some documentation I could follow on this?

It's exactly the same documentation. You can read there there are chapters regarding the ROS2 side apllications and othera that are specific for Fast-RTPS applications.

TSC21, 2020-12-29:
I am going to close this since the initial problem was solved. For questions, please bring them to the #ros channel in the PX4 slack or to discussion.px4.io

closed 2020-12-29 by TSC21

wrgtri, 2020-12-29:
Great, thank you for your help!