PX4/px4_ros_com #63: Failing build

IanBurwell, 2021-01-18:
px4_ros_com is failing colcon build with a clean build:

```bash
Traceback (most recent call last):
  File "/home/en/px4_ros_com_ros2/src/px4_ros_com/scripts/uorb_rtps_classifier.py", line 232, in <module>
    classifier = (Classifier(os.path.abspath(args.yaml_file), msg_dir) if os.path.isabs(args.yaml_file)
  File "/home/en/px4_ros_com_ros2/src/px4_ros_com/scripts/uorb_rtps_classifier.py", line 55, in __init__
    self.check_if_listed(yaml_file)
  File "/home/en/px4_ros_com_ros2/src/px4_ros_com/scripts/uorb_rtps_classifier.py", line 139, in check_if_listed
    raise AssertionError(
AssertionError: 
The following messages are not listen under  /home/en/px4_ros_com_ros2/src/px4_ros_com/templates/uorb_rtps_message_ids.yaml: ActuatorControls4, ActuatorControls5

Please add them to the yaml file with the respective ID and, if applicable, mark them to be sent or received by the micro-RTPS bridge.
NOTE: If the message has multi-topics (#TOPICS), these should be added as well.

CMake Error at cmake/GenerateMicroRTPSAgent.cmake:106 (string):
  string sub-command FIND requires 3 or 4 parameters.
Call Stack (most recent call first):
  CMakeLists.txt:62 (include)

Traceback (most recent call last):
  File "/home/en/px4_ros_com_ros2/src/px4_ros_com/scripts/uorb_rtps_classifier.py", line 232, in <module>
    classifier = (Classifier(os.path.abspath(args.yaml_file), msg_dir) if os.path.isabs(args.yaml_file)
  File "/home/en/px4_ros_com_ros2/src/px4_ros_com/scripts/uorb_rtps_classifier.py", line 55, in __init__
    self.check_if_listed(yaml_file)
  File "/home/en/px4_ros_com_ros2/src/px4_ros_com/scripts/uorb_rtps_classifier.py", line 139, in check_if_listed
    raise AssertionError(
AssertionError: 
The following messages are not listen under  /home/en/px4_ros_com_ros2/src/px4_ros_com/templates/uorb_rtps_message_ids.yaml: ActuatorControls4, ActuatorControls5

Please add them to the yaml file with the respective ID and, if applicable, mark them to be sent or received by the micro-RTPS bridge.
NOTE: If the message has multi-topics (#TOPICS), these should be added as well.

CMake Error at cmake/GenerateMicroRTPSAgent.cmake:132 (string):
  string sub-command FIND requires 3 or 4 parameters.
Call Stack (most recent call first):
  CMakeLists.txt:62 (include)
```

krihankru, 2021-01-19:
Same issue building on AGX Xavier.

Newly added messages to px4_msgs, ActuatorControls4 & ActuatorControls5 (https://github.com/PX4/px4_msgs/commit/61ee97622d7a1eb1ac8473d71cf3fe7395cace3f) not present in "uorb_rtps_message_ids.yaml" (https://github.com/PX4/px4_ros_com/blob/master/templates/uorb_rtps_message_ids.yaml)

Discovered a workaround (to make it build):

Managed to get it to build by deleting ActuatorControls4 & ActuatorControls5 from px4_msgs/msg
Also had to delete px4_msgs from /build and /install in workspace (I deleted both directories entirely, as i was only building px4_msgs and px4_ros_com)

Build completed for both packages, px4_ros_com had stderr output

I cannot recommend using this workaround however, as i've not tested whether or not it breaks anything else - i've only verified that it succeeds in building.

If nothing else I hope it can make it easier for maintainers to identify and fix the error faster

bvarillo, 2021-01-20:
I have exactly the same error when building px4_ros_com.
I am using ubuntu 20.04 with Foxy, Fast-RTPS 1.8.2 and Fast-RTPS-Gen 1.0.4

96moustafa, 2021-01-26:
Same issue!

96moustafa, 2021-01-26:
Same issue!

TSC21, 2021-01-26:
What is missing is adding those messages in the PX4-Firmware side uorb_rtps_message_ids.yaml so it can be deployed. As explained before, any MULTI-TOPICS/alias messages are not being made mandatory on the Firmware side to have RTPS ids, and that has to be fixed. I will do that when I have some time.

TSC21, 2021-01-27:
Hot-fix in https://github.com/PX4/px4_ros_com/commit/dcfbdf4c8d4ea9d511e43f6f79da8543f4375399. Fix upstream will come next.

closed 2021-01-27 by TSC21