PX4/px4_ros_com #64: Unable to start offboard mode. RTL after arming

96moustafa, 2021-01-20:
I am trying to start the offboard mode in this o[ffboard_control](https://github.com/PX4/px4_ros_com/blob/master/src/examples/offboard/offboard_control.cpp) example: 
as specified by @TSC21  in this [video](https://drive.google.com/file/d/1HqxoIAHeYwOskJBH2Dg20vTBdvBOqgEi/view).

However, every time I run the example, I get:

INFO [commander] Armed by external command
INFO [navigator] RTL: landing at home position.
INFO [logger] Start file log (type: full)
INFO [logger] [logger] ./log/2021-01-20/13_17_46.ulg
INFO [logger] Opened full log file: ./log/2021-01-20/13_17_46.ulg
INFO [tone_alarm] notify negative
INFO [commander] Disarmed by auto preflight disarming
INFO [logger] closed logfile, bytes written: 753394

So, it lands to home position right after arming and then gets disarmed right away.
Both COM_OBL_ACT and COM_OBL_RC_ACT are disabled.

When I have my joystick connected, I don't get the **RTL: landing at home position** but still the drone doesn't fly and it gets disarmed after a few seconds.

I checked that the required messages are sent as expected from the micrortps agent to the client.

TSC21, 2021-01-20:
Please check https://px4.slack.com/archives/C7MTU2K17/p1611132178047600?thread_ts=1610595571.002000&cid=C7MTU2K17. The example needs to be udapted.

TSC21, 2021-01-27:
The source code in https://github.com/PX4/px4_ros_com/blob/master/src/examples/offboard/offboard_control.cpp was updated. Can you please test again?

96moustafa, 2021-01-27:
I tested it and now it works. Thanks.

closed 2021-01-27 by TSC21

96moustafa, 2021-01-27:
Could you please confirm that the offboard mode only works with local coordinates only. I checked the **FlightTaskOffboard.cpp** and as far as I can see, it only considers the local x/y/z points and does not deal with lat/long/alt. Is there another way to control the offboard mode using a global frame?

yj-Tang, 2025-07-24:
> The source code in https://github.com/PX4/px4_ros_com/blob/master/src/examples/offboard/offboard_control.cpp was updated. Can you please test again?

Hi Nuno, I experienced the same problem with the Python code. Could you please also update the Python version?