PX4/px4_ros_com #65: Add InputRc to published messages

dk7xe, 2021-02-03:
For some usecases like starting video recording on a companion computer or triggering a ROS based (offboard control) mission control execution the status of RC channel is needed in ROS(2). This allows the user to start/stop such functionalities from the RC transmitter.

Therefore `/px4_ros_com/templates` entry for InputRc to be changed accordingly.
Set `send: true`

TSC21, 2021-02-03:
Please fell free to make a PR against the PX4-Autopilot repo to update https://github.com/PX4/PX4-Autopilot/blob/master/msg/tools/uorb_rtps_message_ids.yaml.

dk7xe, 2021-02-03:
@TSC21 https://github.com/PX4/PX4-Autopilot/pull/16743
should i do PR here as well?

TSC21, 2021-02-03:
> @TSC21 [PX4/PX4-Autopilot#16743](https://github.com/PX4/PX4-Autopilot/pull/16743)
> should i do PR here as well?

Thanks. No. The files here are deployed automatically by the PX4-Autopilot CI whenever a change occurs to them there.

TSC21, 2021-02-03:
I think we can close this now.

closed 2021-02-03 by TSC21