PX4/px4_ros_com #69: Micrortps_client - unexpected topic ID

jmalbert97, 2021-02-28:
OS: Ubuntu 18.04 
FCU: PixHawk 4 
Firmware: px4_fmu-v5_rtps 

I am trying to get a simple offboard control to work, as described here: http://docs.px4.io/master/en/ros/ros2...

When I start the agent on a companion computer (Odroid XU4) and client on the PixHawk4, the connection is established as expected. When i run sensor_combined_listener, i have no issues, so I know the connection is established. However, when I attempt to run the offboard_control, i receive a message from the client flight controller :

`WARN [micrortps_client] Unexpected topic ID
`
I have checked both the yaml file on the companion computer in px4_ros_com/src/px4_ros_com/templates/uorb_rtps_message_ids.yaml, as well as X4-Autopilot/msg/tools/uorb_rtps_message_ids.yaml, and both show the required fields as 'receive: true' for offboard_control_mode and position_setpoint_triplet.

Any ideas why I may be getting this unexpected topic ID error? Here is the output from the agent, showing that the topics have been matched:
```
----   Publishers  ----
- InputRc publisher started
- SatelliteInfo publisher started
- SensorCombined publisher started
- Timesync publisher started
- VehicleControlMode publisher started
- VehicleOdometry publisher started

- CollisionConstraints publisher started
-----------------------
[   micrortps_agent   ] OffboardControlMode subscriber matched
[   micrortps_agent   ] PositionSetpointTriplet subscriber matched
[   micrortps_agent   ] VehicleCommand subscriber matched
[   micrortps_agent   ] Timesync publisher matched
```
here is the output from the other terminal, showing the offboard_control file stuck at arming (this is when i get the error on the client, saying unexpected topic ID)

```
odroid@odroid:~/px4_ros_com_ros2/build/px4_ros_com$ ros2 run px4_ros_com offboard_control
Starting offboard control node...
[INFO] [offboard_control]: Arm command send
```

LeQuangHien, 2021-03-08:
Hi, I have also faced this problem. Any update on this?

jmalbert97, 2021-03-08:
> Hi, I have also faced this problem. Any update on this?

Yes! The documentation is just missing a key step, which after a few hours of frustration i realized. Hopefully I can save you that frustration. First of all, make sure you have edited the .yaml files on both the ros2 workspace, as well as px4 workspace. According to the documentation, we only need to add receive: true to the two mentioned topics here: https://docs.px4.io/master/en/ros/ros2_offboard_control.html 

However, on the most current PX4 version, i also had to add receive: true to the topic VehicleCommand, i think the ID was in the 90's. Look at the yaml file located in the PX4 workspace and add receive: true to the topics mentioned as well as VehicleCommand, then rebuild the px4 firmware (should be something like build px4_fmu-v5_rtps) then upload the new firmware to the flight controller using QGroundControl. Let me know if this helps, as the documentation surely needs updating to make mention of this.

LeQuangHien, 2021-03-08:
> > Hi, I have also faced this problem. Any update on this?
> 
> Yes! The documentation is just missing a key step, which after a few hours of frustration i realized. Hopefully I can save you that frustration. First of all, make sure you have edited the .yaml files on both the ros2 workspace, as well as px4 workspace. According to the documentation, we only need to add receive: true to the two mentioned topics here: https://docs.px4.io/master/en/ros/ros2_offboard_control.html
> 
> However, on the most current PX4 version, i also had to add receive: true to the topic VehicleCommand, i think the ID was in the 90's. Look at the yaml file located in the PX4 workspace and add receive: true to the topics mentioned as well as VehicleCommand, then rebuild the px4 firmware (should be something like build px4_fmu-v5_rtps) then upload the new firmware to the flight controller using QGroundControl. Let me know if this helps, as the documentation surely needs updating to make mention of this.
> 
> If this does not work, make sure you are using the correct baud rate on both sides, and that mavlink is disabled on the port you are using for ROS 2. Then make sure that port is set to use RTPS. All of this can be done in QGroundControl => parameters.

Thank you for your quick reply!

My case is a bit different to you. I have followed these steps: https://docs.px4.io/master/en/ros/ros2_comm.html#quick-start-testing-the-bridge-with-px4-sitl that use only for ROS2. In step 3, I start the micrortps_client daemon also in UDP in SITL windows, I received the message like you right away: `WARN [micrortps_client] Unexpected topic ID `. I have not even start offboard_control node. But this command: `ros2 launch px4_ros_com sensor_combined_listener.launch.py` is working.

> If this does not work, make sure you are using the correct baud rate on both sides, and that mavlink is disabled on the port you are using for ROS 2. Then make sure that port is set to use RTPS. All of this can be done in QGroundControl => parameters.

I have followed your instruction and come to here and don't know how to set the port of Mavlink to use RTPS. In QGroundControl => parameters, I cannot find any information about the port. In QGroundControl, I can control the px4_sitl_rtps as normal.

LeQuangHien, 2021-05-07:
This issue has gone with the latest commit.

closed 2021-05-07 by TSC21