PX4/px4_ros_com #73: Unable to use UDP/TCP for RTPS client and agent communication with pixhawkV2

edd-911, 2021-03-19:
i am trying to establish RTPS connection between companion computer(RPI) and pixhawk. I was able to run px4_sitl_rtps with gazebo simulation, passing the data on to micrortps_agent and read the sensor data using SensorCombined_PubSub topic.
When i replaced the whole scenario with pixhawk and RPi4B for hardware simulation, i am not getting any sensor data on agent side using UDP connection. Pixhawk is connected with RPI4b using USB cable at port '/dev/ttyACM0'
i am following this issue "https://github.com/PX4/px4_ros_com/issues/38" and replaced UART with UDP in client and agent command.

TSC21, 2021-03-19:
So what exactly is the problem here? If you are using a serial connection, it is normal that UDP will not work, right?

edd-911, 2021-03-19:
i am running both client and agent in different terminals of RPI only, but i couldn't able to get sensor data from pixhawk using RTPS

TSC21, 2021-03-19:
> i am running both client and agent in different terminals of RPI only, but i couldn't able to get sensor data from pixhawk using RTPS

But are you starting both in UART and in the right port?

edd-911, 2021-03-19:
i am using UDP in place of UART like  'micrortps_client start -t UDP' and 'micrortps_agent -t UDP'

TSC21, 2021-03-19:
> i am using UDP in place of UART like 'micrortps_client start -t UDP' and 'micrortps_agent -t UDP'

But how exactly do you want to make UDP running in a serial link? That doesn't work.

closed 2021-03-20 by edd-911