PX4/px4_ros_com #75: [micrortps_agent] Unexpected topic ID 'xxx' to publish ....

nuitvolgit, 2021-04-02:
-OS: Ubuntu 20.04
-Env: Gazebo simulator with ROS2
I was following this tutorial [http://docs.px4.io/master/en/ros/ros2_comm.html] and all the build processes went well.
But the "Sanitiy Check the Installation" from the tutorial did not work.

What I did was followings
1) run  "$ make px4_sitl_rtps gazebo" from terminal A

2) run  "$ source ~/px4_ros_com_ros2/install/setup.bash" and "$ micrortps_agent -t UDP" from terminal B

3) run "pxh> micrortps_client start -t UDP" from terminal A

4) Then from the micrortps_agent side (terminal B), I got infinite following messages:
[   micrortps_agent   ] Unexpected topic ID '87' to publish Please make sure the agent is capable of parsing the message associated to the topic ID '87'
[   micrortps_agent   ] Unexpected topic ID '96' to publish Please make sure the agent is capable of parsing the message associated to the topic ID '96'
.......
and so on with different topic IDs...

It seems like the micrortps_agent is trying to publish topics to the micrortps_client but the topicIDs are not acceptable...

I also tested from the other computer by installing px4 and px4_ros_com.
But the result was exactly the same.

Any ideas???

TSC21, 2021-04-02:
What branch of PX4 are you cloning? Are you using the master of all the repos (PX4-Autopilot, px4_ros_com, px4_msgs)?

nuitvolgit, 2021-04-02:
Yes, all masters downloaded today. Tried two different PCs and the same results.

TSC21, 2021-04-02:
> Yes, all masters downloaded today. Tried two different PCs and the same results.

Did you change anything on the yaml files where the RTPS ids are set? Either in the PX4 side or in this repo?

nuitvolgit, 2021-04-02:
Did not change anything at all.

TSC21, 2021-04-02:
Ok I just now saw that for some reason the PX4 Jenkins CI is not depliying the changes on the uORB messages. I will check what's happening.

nuitvolgit, 2021-04-02:
I really appreciate it!

nuitvolgit, 2021-04-06:
Do we have any update about this?