PX4/px4_ros_com #76: px4_ros_com compiling error

Boshen97, 2021-04-02:
When compiling with the "colcon build" command, the compilation of [px4_ros_com] fails and the following message is displayed:

Starting >>> px4_ros_com
--- stderr: px4_ros_com                              
make[2]: *** [/home/amov/ros2_ws/src/px4_ros_com/src/micrortps_agent/microRTPS_agent.cpp] Error 1
make[1]: *** [CMakeFiles/micrortps_agent.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< px4_ros_com [0.35s, exited with code 2]

How can I solve this problem? Please

TSC21, 2021-04-02:
Did you follow the documentation in http://docs.px4.io/master/en/ros/ros2_comm.html? The above error doesn't tell us anything that allows its debugging.

Boshen97, 2021-04-02:
I skipped the fastrtps-related operations. I think that the fastrtps-related content will be downloaded in the ROS2 download, but it seems that I am doing it wrong. I will try to install the FASTRTPS-related content.

TSC21, 2021-04-02:
> I skipped the fastrtps-related operations. I think that the fastrtps-related content will be downloaded in the ROS2 download, but it seems that I am doing it wrong. I will try to install the FASTRTPS-related content.

You need at least to install fastrtpsgen 1.0.4. Otherwise it will not be able to generate the micrortps_agent code.

closed 2021-04-02 by TSC21