PX4/px4_ros_com #79: Time synchronization of timestamp_sample

dsix-ls2n, 2021-04-11:
While trying px4_ros_com, I found some issues with the synchronization of the timestamp_samples items in messages. While timestamp items are well synchronized, the timestamps_samples are not for two reasons according to me.

1- The detection of the existence of the member fails (in RtpsTopics.h generated by [RtpsTopic.h.em](https://github.com/PX4/px4_ros_com/blob/master/templates/RtpsTopics.h.em)). Probably because the timestamp_sample member is overloaded in px4_msgs::msg::_A message with timestamp_sample_
```
    // SFINAE
    template<typename T> struct hasTimestampSample{
    private:
      static void detect(...);
      template<typename U> static decltype(std::declval<U>().timestamp_sample()) detect(const U&);
    public:
      static constexpr bool value = std::is_same<uint64_t, decltype(detect(std::declval<T>()))>::value;
    };
```
A working solution for me was
```
    // SFINAE
    template<typename T> struct hasTimestampSample{
    private:
        template<typename U,
                typename = decltype(std::declval<U>().timestamp_sample(int64_t()))>
        static std::true_type detect(int);
        template<typename U>
        static std::false_type detect(...);
    public:
        static constexpr bool value = decltype(detect<T>(0))::value;
    };
```
But there might be a more elegant solution, I am not very comfortable with this kind of detection.

2- Another more easy and obvious issue is the missing appropriate lines in [RtpsTopics.cpp.em](https://github.com/PX4/px4_ros_com/blob/master/templates/RtpsTopics.cpp.em) in the template of 
void RtpsTopics::publish(uint8_t topic_ID, char data_buffer[], size_t len)
```
            uint64_t timestamp_sample = getMsgTimestampSample(&st);
            _timesync->subtractOffset(timestamp_sample);
            setMsgTimestampSample(&st, timestamp_sample);
```

Sorry if it is not the appropriate way to provide my feedback, I am not completely accustomed to Github.

TSC21, 2021-04-12:
Hi @dsix-ls2n! Thanks for reporting! What's the specific message you are not getting the samples synced? Btw, 2. can never be a solution (unless I don't understand why you state as a solution), because not all messages have sample timestamps. So you need to use SNIFAE to understand if a `timestamp_sample` exist. Regarding 1. I am not entirely sure I understand your problem. Are you saying that the `timestamp_sample` had different types for different messages?

dsix-ls2n, 2021-04-12:
Hi, 
I discovered the issue on VehicleOdometry message for example. The timestamp was synchronized but not the timestamp_sample. 

For 2., it was already implemented for timestamp_sample in RtpsTopics::getMsg but not in RtpsTopics::publish. So the timestamp_sample was supposedly synchronized in the way ROS2->vehicle but not the way vehicle->ROS2. Those functions definition uses the SFINAE to detect if the member exists. So adding the lines mentioned doesn't break the compilation. **But** it does not work either.

The reason was that the SFINAE, as currently implemented always outputs **false**, whatever the message type. I checked with debugging printing. I am unsure of the core reason, but I think it is related to the fact that the member function that we try to detect using SFINAE is overloaded (see also [this](https://stackoverflow.com/questions/28309164/checking-for-existence-of-an-overloaded-member-function)). So I tried the way proposed in the mentioned discussion and it worked. The SFINAE was detecting properly the member and the synchronization of the sample stamps work in this case.

the SNIFAE is already there, it was just not working properly (always detecting that the member timestamp_sample does no exist). So there is no compilation error with 2. By the way,

TSC21, 2021-05-07:
@dsix-ls2n can you issue a PR?

dsix-ls2n, 2021-05-07:
Yes, I can and will, in a few days.

TSC21, 2021-05-12:
> @TSC21 I have a question though. Since the code for the micrortps agent is present in px4_ros_com but also in px4-autopilot. Should I make the merge request on both repositories?

@dsix-ls2n answered here: https://github.com/PX4/px4_ros_com/pull/82#pullrequestreview-657590769.

TSC21, 2021-07-01:
Fixed.

closed 2021-07-01 by TSC21