PX4/px4_ros_com #81: micrortps_agent uses excessive resources on Mac

smnogar, 2021-05-05:
Is anyone else running macOS? When I run micro rtps agent it routinely uses 300-400% CPU for a single vehicle. This is for software in the loop simulation, so using UDP locally. 

When killing micrortps_agent, I routinely see many many errors like: 

```
[micrortps_agent-6] 2021-05-03 21:00:53.549 [RTPS_TRANSPORT_SHM Error] Failed init_port fastrtps_port7432: mutex lock failed: Invalid argument -> Function open_port_internal
[micrortps_agent-6] 2021-05-03 21:00:53.551 [RTPS_TRANSPORT_SHM Error] Failed init_port fastrtps_port7420: mutex lock failed: Invalid argument -> Function open_port_internal
[micrortps_agent-6] 2021-05-03 21:00:53.553 [RTPS_TRANSPORT_SHM Error] Failed init_port fastrtps_port7430: mutex lock failed: Invalid argument -> Function open_port_internal
[micrortps_agent-6] 2021-05-03 21:00:53.556 [RTPS_TRANSPORT_SHM Error] Failed init_port fastrtps_port7424: mutex lock failed: Invalid argument -> Function open_port_internal
[micrortps_agent-6] 2021-05-03 21:00:53.561 [RTPS_TRANSPORT_SHM Error] Failed init_port fastrtps_port7424: mutex lock failed: Invalid argument -> Function open_port_internal
[micrortps_agent-6] 2021-05-03 21:00:53.561 [RTPS_TRANSPORT_SHM Error] Failed init_port fastrtps_port7420: mutex lock failed: Invalid argument -> Function open_port_internal
[micrortps_agent-6] 2021-05-03 21:00:53.564 [RTPS_TRANSPORT_SHM Error] Failed init_port fastrtps_port7420: mutex lock failed: Invalid argument -> Function open_port_internal
[micrortps_agent-6] 2021-05-03 21:00:53.565 [RTPS_TRANSPORT_SHM Error] Failed init_port fastrtps_port7432: mutex lock failed: Invalid argument -> Function open_port_internal
[micrortps_agent-6] 2021-05-03 21:00:53.567 [RTPS_TRANSPORT_SHM Error] Failed init_port fastrtps_port7424: mutex lock failed: Invalid argument -> Function open_port_internal
[micrortps_agent-6] 2021-05-03 21:00:53.567 [RTPS_TRANSPORT_SHM Error] Failed init_port fastrtps_port7430: mutex lock failed: Invalid argument -> Function open_port_internal
[micrortps_agent-6] 2021-05-03 21:00:53.570 [RTPS_TRANSPORT_SHM Error] Failed init_port fastrtps_port7430: mutex lock failed: Invalid argument -> Function open_port_internal
```

I've tried updating fastrtps to the newest version, but to no avail. Any other ideas. Is opening >200 ports on macOS just a bad idea?

TSC21, 2021-07-01:
Can you retry latest master of PX4 Firmware and px4_ros_com and check if this is still an issue?

smnogar, 2021-07-01:
Yes, its still an issue. Restricting RTPS traffic to localhost does not help either. Still seeing usage between 200-400%.

TSC21, 2021-07-01:
> Yes, its still an issue. Restricting RTPS traffic to localhost does not help either. Still seeing usage between 200-400%. Maybe this is strictly a FastRTPS issue. Is there a performance test I can run to compare it to Ubuntu?

I don't have a Mac to test it and we don't officially support it, so I can't really help you debug it. There's no performance test for CPU/RAM usage. I would assume this might though be an issue with FastDDS or ROS2 (rmw_fastrtps) so I would try to report it on the respective repos.