PX4/px4_ros_com #83: No ROS2 topic publication

nuitvolgit, 2021-05-17:
-OS: Ubuntu 20.04
-Env: Gazebo11 simulator with ROS2
I was following this tutorial (http://docs.px4.io/master/en/ros/ros2_comm.html) and all the build processes went well.
But the "Sanitiy Check the Installation(https://docs.px4.io/master/en/ros/ros2_comm.html)" from the tutorial did not work.
I followed step 1 ~ 5 without any error. (Gazebo was launched, PX4 started, micrortps_agent worked)
However, at step 5, I was supposed to see "the data beging printed on the terminal" but I did not receive anything as follows:
**dh@dhpc:~/dev_ws$ ros2 launch px4_ros_com sensor_combined_listener.launch.py 
[INFO] [launch]: All log files can be found below /home/dh/.ros/log/2021-05-17-15-04-11-527403-dhpc-60616
[INFO] [launch]: Default logging verbosity is set to INFO
/home/dh/dev_ws/install/px4_ros_com/share/px4_ros_com/launch/sensor_combined_listener.launch.py:51: UserWarning: The parameter 'node_executable' is deprecated, use 'executable' instead
  launch_ros.actions.Node(
[INFO] [sensor_combined_listener-1]: process started with pid [60618]
[sensor_combined_listener-1] Starting sensor_combined listener node...**

I also checked "ros2 topic list" and found out that there was now publisher as follows:
**dh@dhpc:~$ ros2 topic info /SensorCombined_PubSubTopic  -v
Type: px4_msgs/msg/SensorCombined
Publisher count: 0
Subscription count: 1
Node name: sensor_combined_listener
Node namespace: /
Topic type: px4_msgs/msg/SensorCombined
Endpoint type: SUBSCRIPTION
GID: 01.0f.d1.a6.ca.ec.00.00.01.00.00.00.00.00.12.04.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RMW_QOS_POLICY_RELIABILITY_RELIABLE
  Durability: RMW_QOS_POLICY_DURABILITY_VOLATILE
  Lifespan: 2147483651294967295 nanoseconds
  Deadline: 2147483651294967295 nanoseconds
  Liveliness: RMW_QOS_POLICY_LIVELINESS_AUTOMATIC
  Liveliness lease duration: 2147483651294967295 nanoseconds**

Any ideas?

nuitvolgit, 2021-05-17:
By the way, I changed 'node_executable' to 'executable' in the sensor_combined_listener.launch.py since 'node_executable' is depricated. But both didn't change the result.

TSC21, 2021-05-17:
At a first glance it seems that the discovery mechanism is not finding the topics from the agent. Did you setup the ROS2 network to localhost only? If yes, it will not work as you are seeing but I will be bringing a patch soon to have it working. Rather than this, I don't see what can be wrong.

nuitvolgit, 2021-05-17:
I checked "echo $ROS_LOCAL_HOST_ONY " and it returns 0.

TSC21, 2021-05-17:
> I checked "echo $ROS_LOCAL_HOST_ONY " and it returns 0.

Ok then as I said I honestly don't have an idea of the cause. Can you check you are able to list topics from other nodes you start? The above to me looks like a discovery problem and not something related with the micro-RTPS agent considering that you are running it.

nuitvolgit, 2021-05-17:
I can see other ros2 topics from other nodes, which seems like that only the sensor_combined topic is not published.

TSC21, 2021-05-17:
> I can see other ros2 topics from other nodes, which seems like that only the sensor_combined topic is not published.

Being published and being advertised are different things. When a node starts, it advertises that it has publishers. The discovery mechanism will inform other nodes that that node has those topics to publish. In this case, if you are completely sure you are starting the agent, then something is wrong with the discovery. I would argue it might be related with the rtps profile being used, but I am not sure. I cannot reproduce this issue.

nuitvolgit, 2021-05-17:
I just solved the problem by commenting followings in .bashrc file.

#export ROS_DOMAIN_ID=7
#export ROS_NAMESPACE=robot1
#export RMW_IMPLEMENTATION=rmw_fastrtps_cpp
#export RCUTILS_CONSOLE_OUTPUT_FORMAT='[{severity}]: {message}'
#export RCUTILS_COLORIZED_OUTPUT=1
#export RCUTILS_LOGGING_USE_STDOUT=0
#export RCUTILS_LOGGING_BUFFERED_STREAM=1

I first set the env parameters by following someone's tutorial.
But I haven't get enough knowledge about how each of the parameters work.
By commenting them, I was able to receive the messages!
Could you figure out what parameters caused the problem?

TSC21, 2021-05-17:
Probably these two:
```
#export ROS_DOMAIN_ID=7
#export ROS_NAMESPACE=robot1
```
Most certainly the first one. That's usually the problem of following custom tutorials. You should follow the official ones from OSRF. Closing this as this is not related with `px4_ros_com`.

closed 2021-05-17 by TSC21

nuitvolgit, 2021-05-17:
I really appreciate it! Your comments gave me a clue to solve the problem. I will keep your advice in mind.

spes2357, 2021-11-06:
@TSC21 Hi Nuno.  I found this solution very useful. I had the same problem. If I follow the ros2 tutorial. The ros2 tutorial instructs user to set ROS_DOMAIN_ID. I wonder if we can add info(tips) to the document for new users from ros2. 

https://docs.px4.io/master/en/ros/ros2_comm.html#build-ros-2-workspace
https://docs.ros.org/en/foxy/Tutorials/Configuring-ROS2-Environment.html