PX4/px4_ros_com #86: Compilation Errors -- Ubuntu 20/Focal + Ros2/Galactic + colcon build

teyrana, 2021-05-30:
```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.2 LTS
Release:	20.04
Codename:	focal

$ rosversion -d
galactic

$ gcc --version
gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0

$ fastddsgen -version
openjdk version "14.0.2" 2020-07-14
OpenJDK Runtime Environment (build 14.0.2+12-Ubuntu-120.04)
OpenJDK 64-Bit Server VM (build 14.0.2+12-Ubuntu-120.04, mixed mode, sharing)
fastddsgen version 2.0.1

```

TSC21, 2021-05-30:
@teyrana sorry but what exactly is the error?

teyrana, 2021-05-30:
standby, still writing the report

TSC21, 2021-05-30:
Also why are you using FastDDSGen 2.0.1 when it is clearly stated in the docs that FastRTPSGen 1.0.4 is the version to use?

TSC21, 2021-05-30:
It's not clear to me what's causing the error above. Are you sure you are using the latest `px4_ros_com` source code? You can see that the current code in this repo builds correctly according to CI under a similar environment to yours: https://github.com/PX4/px4_ros_com/runs/2702198010?check_suite_focus=true.

teyrana, 2021-05-30:
the relevant sections [of the docs](https://docs.px4.io/master/en/ros/ros2_comm.html):

>  Install Fast DDS
> 
> Follow the Fast DDS Installation Guide to install Fast RTPS(DDS) 2.0.0 (or later) and Fast-RTPS-Gen 1.0.4 (or later) and their dependencies.
> 
> Note
> 
> Check the guide to confirm the latest dependencies! You won't be able to continue with this guide until the correct versions of Fast RTPS(DDS) and **Fast-RTPS-Gen have been installed.

For anyone else running into this issue, there is currently a _maximum_  working version of "fast-rtps-gen".   Newer versions do not work, per TSC21's comment, above.   It is also unclear which exact versions work... although the above comment suggests version 1.04 should work.

TSC21, 2021-05-30:
> It is also unclear which exact versions work... although the above comment suggests version 1.04 should work.

If the user is told to install a specific version, there's a reason for that.

TSC21, 2021-05-30:
@teyrana I just made it more clear here: https://github.com/PX4/PX4-user_guide/pull/1287. Sorry for any inconsistency.

teyrana, 2021-05-30:
@TSC21 
Thanks!

TSC21, 2021-05-30:
@teyrana do you still have the second error or this can be closed?

teyrana, 2021-05-31:
Your choice.
The error is definitely happening, but it's not a configuration the project officially supports.

TSC21, 2021-05-31:
> Your choice.
> The error is definitely happening, but it's not a configuration the project officially supports.

Can you be more explicitly what configuration is generating this error? Is it still related with FastDDSGen? Because if yes, then no, it's not going to be fixed

teyrana, 2021-06-01:
> ... not a configuration the project officially supports.

Specifically, you have attributed it to usage of `rtiddsgen` and I believe you. 

I hove not dug any deeper.

TSC21, 2021-06-01:
> > ... not a configuration the project officially supports.
> 
> Specifically, you have attributed it to usage of `rtiddsgen` and I believe you.
> 
> I hove not dug any deeper.

`rtiddsgen`? Sorry but now I am even more confused now. Error 2 I am mostly sure it is because you are not using the latest code on this repo. Support for galactic was added here: https://github.com/PX4/px4_ros_com/commit/bce18bc11f1e924b76e71ab35adb7687871804d1. Please check of your CMakeLists.txt contain these changes.
Closing this as this has become a non-issue.

closed 2021-06-01 by TSC21

teyrana, 2021-06-02:
*`fastddsgen`  as notes in the issue report.