PX4/px4_ros_com #90: Some uORB messages are not published

nuitvolgit, 2021-06-01:
Hello,

I am currently using gazebo 11 with ROS2 foxy for px4 sitl.
I could subscribe some messages such as 'vehicle_local_position' from ROS2 side.
Next, I tried to receive 'vehicle_global_position' message (just like the same way as receiving 'vehicle_local_position').
From px4_sitl_rtps console, I was able to check 'vehicle_global_position message' 
by using 'pxh> listener vehicle_global_position'.
But I was not able to subscribe 'vehicle_global_position' uORB message from ROS2 side.
Of course I ran 'micrortps_agent -t UDP' in ROS2 side and 'micrortps_client start -t UDP' in px4 side.

Is it like only some of the uORB messages can be transferred from micrortps_client to micrortps_agent?
If it is, can I select the messages that I would like to transfer?

Thanks.

nuitvolgit, 2021-06-02:
I found out the solution by changing "uorb_rtps_message_ids.yaml" file (send, receive)
https://docs.px4.io/master/en/middleware/micrortps.html#supported-uorb-messages

closed 2021-06-02 by nuitvolgit

JohnnyLiu2886, 2022-01-19:
> Hello,
>
> I am currently using gazebo 11 with ROS2 foxy for px4 sitl. I could subscribe some messages such as 'vehicle_local_position' from ROS2 side. Next, I tried to receive [...]

I can subscribe to vehicle_global_position  after adding it to the uorb_rtps_message_ids.yaml.

However, I cannot subscribe to the "vehicle_local_position" even adding it to the .yaml file.
I can detect it using "ros2 topic list" and listen to its data using "listener vehicle_local_position" in the nsh terminal.

However, the subscription function for "vehicle_local_position" did not execute. I got nothing even using "ros2 topic echo /fmu/vehicle_local_position/out".

Other uorb messages run properly in ros2 such as "vehicle_gps_position", "battery_status" and "vehicle_status".

I am using ros2 ,holybro pixhawk4, jetson Xavier and the latest commit of the PX4-Autopilot. I am currently able to setpoint, disarming, arming, switch offboard mode, switch manual mode and receiving gps position using ros2. 

However, I don't why the "vehicle_local_position" is not working. I just added it to the .yaml as usual. Other topics are working, only "vehicle_local_position" is not working.

I need the reference global position of "vehicle_local_position".

JohnnyLiu2886, 2022-01-20:
> I can subscribe to vehicle_global_position after adding it to the uorb_rtps_message_ids.yaml.
>
> However, I cannot subscribe to the "vehicle_local_position" even adding it to the .yaml file. I can [...]

It is working on the gazebo simulator now but it is still not working in the holybro pixhawk 4.