PX4/px4_ros_com #93: Build ros2 workspace fails in microRTPS_transport.cpp

RodBelaFarin, 2021-06-21:
The build_ros2_workspace.bash script fails due to errors in microRTPS_transport.cpp

Following the instructions from here

http://docs.px4.io/master/en/ros/ros2_comm.html

an error occurs during building:

```
/home/user/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/microRTPS_transport.cpp: In member function ‘ssize_t Transport_node::read(uint8_t*, char*, size_t)’:
/home/user/px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent/microRTPS_transport.cpp:153:99: error: expected ‘)’ before ‘PRIu32’
  153 |   if (debug) printf("\033[1;33m[ micrortps_transport ]\t                                (↓↓ %" PRIu32 ")\033[0m\n", msg_start_pos);
      |                    ~                                                                              ^~~~~~~
      |                                                                                                   )
```
Same error in line 181 and 197.

TSC21, 2021-06-21:
The fix is bring brought in https://github.com/PX4/PX4-Autopilot/pull/17782.

RodBelaFarin, 2021-06-21:
In PX4-Autopilot i switched to the branch pr-ros2_bridge_improvements and i ran build_ros2_workspace.bash again.
Going back to my px4_ros_com_ros2 workspace running the build_ros2_workspace.bash again it results in the same error.

Any steps I am missing?

TSC21, 2021-06-21:
> In PX4-Autopilot i switched to the branch pr-ros2_bridge_improvements and i ran build_ros2_workspace.bash again.
> Going back to my px4_ros_com_ros2 workspace running the build_ros2_workspace.bash again it results in the same error.
> 
> Any steps I am missing?

You are not supposed to switch branches. The generator code and templates from this repo come from that repo: they get deployed every time a merge to the master branch with changes to them happen. So you rather wait for that PR to get merged, and so, the deploy to this branch happens, or you manually copy the files from there to the respective folders. I would rather wait because that PR should be merged today or tomorrow.

RodBelaFarin, 2021-06-21:
Yeah, sorry. I was a bit confused. I added the missing inttypes header to the microRTPS_transport.cpp and now it builds fine. Thanks!

closed 2021-06-21 by RodBelaFarin