PX4/px4_ros_com #94: Custom px4_msg not received on client side

GuillermoHerreraF, 2021-06-30:
Hello @TSC21. Hope you are doing well. 
I'm following up on an issue discussed [here](https://github.com/PX4/px4_msgs/issues/7).

I'm trying to add a custom topic to px4_msgs based on a uORB topic that I created in the PX4 firmware. I know the uORB topic on the PX4 firmware works fine because I manage to use it with MAVLink. However, I would like to add some autonomous control capability with ROS2 running on a companion computer.
The client is running on physical hardware (CUAV V5 nano) and I'm running the agent on a Linux machine (Ubuntu 20 and ROS Foxy). I'm running the bridge in serial configuration (UART).

Following the steps you described [here](https://github.com/PX4/px4_msgs/issues/7#issuecomment-859605810), I added the custom message on the agent side. When I run the agent the name appears on the list of available topics to publish. And when I publish to the topic (via terminal), I get the following message: 

`[micrortps_agent] CustomCmd subscriber matched`

 So apparently, on the agent side it appears to be working. 

Nevertheless, on the client side I'm not receiving the corresponding uORB messages. 

To answer your [questions](https://github.com/PX4/px4_msgs/issues/7#issuecomment-871483468):

> Hard to tell without more context. Do you have the `micrortps_client` running and setup properly? Do you have that same topic being published on the `micrortps_client`? You need to make sure that your client is able to receive the message ID 141.

- Yes, I am running the `micrortps_client` properly, I even subscribe to the topics `/VehicleAttitude_PubSubTopic` and `/SensorCombined_PubSubTopic` and perform some computation with that data in another node I created.
- What do you mean when you ask me if I have that same topic being published on the `micrortps_client`?
- When you say that I need to make sure that my client is able to receive the message ID 141, what do you mean specifically? I added the following in the `uorb_rtps_message_ids.yaml` file. 
```
  - msg: custom_cmd
    id: 141
    receive: true
```
As the uORB message already exists, I do not know if there is something else to do regarding the client side. I'm assuming there is not.
Any clue what may be happening here? Thank you in advance!

TSC21, 2021-06-30:
What's the output of `listener custom_cmd` in the PX4 nsh?

TSC21, 2021-06-30:
Regarding the client side: to make sure that your client code already contains the uORB publisher for your custom_cmd, check the resulting microRTPS_client.cpp file in the build folder generated from https://github.com/PX4/PX4-Autopilot/blob/master/msg/templates/uorb_microcdr/microRTPS_client.cpp.em. if it's there, then the only thing I can say is that the message is probably not being sent from the agent to the client, or it's not even arriving to the agent in the first place (on the ROS side). You will have to debug each stage of the communication pipeline to see where the message arrives and where it doesn't.

GuillermoHerreraF, 2021-06-30:
Thank you for your quick response @TSC21.
I checked the `microRTPS_client.cpp`, and my custom message is there:
```
...
  case 141:
  {
      crane_cmd_s crane_cmd_data;
      deserialize_crane_cmd(&reader, &crane_cmd_data, data_buffer);
      pubs->crane_cmd_pub.publish(crane_cmd_data);
      ++received;
  }
  break;
...
```
When the agent displays `[micrortps_agent] CustomCmd subscriber matched` after I start publishing on the ROS topic, doesn't this mean that it is receiving the messages? 

Regarding the `listener` command, I'll test tomorrow and come back to you because I do not have access to the hardware right now. 

However, as I'm running the bridge via serial I cannot run the command at the same I'm running the bridge.
What I do is check the topic afterwards and verify the last published data. 

I tried doing the same with the `debug_value` uORB topic. I published several times on the /DebugValue_PubSubTopic on the agent side, and then ran `listener debug_value` on the client side (without the bridge running), and I got the message: `topic never published`. Which I found strange.

TSC21, 2021-06-30:
> Thank you for your quick response @TSC21.
> I checked the `microRTPS_client.cpp`, and my custom message is there:
>
> ```
> ...
>   case 141:
>   {
>       crane_cmd_s crane_cmd_data; [...]

No. It means that the DDS participant found a matching subscriber to the publisher.

> Regarding the `listener` command, I'll test tomorrow and come back to you because I do not have access to the hardware right now.
>
> However, as I'm running the bridge via serial I cannot run the [...]

My advice is that you connect also the USB cable and use the mavlink_shell to access the FMU side, besides you connection through UART.

> I tried doing the same with the `debug_value` uORB topic. I published several times on the /DebugValue_PubSubTopic on the agent side, and then ran `listener debug_value` on the client side (without the bridge running), and I got the message: `topic never published`. Which I found strange.

Well it's not strange if it's not working :) that poitns to something wrong on your TX or RX lines (depending on the direction), so HW related.

TSC21, 2021-06-30:
What commands are you using to start the client and the agent? Can you please share them here? Did you adjusted the baud rate, i.e. increased it? Are they matching on both sides of the bridge?

ow18600, 2021-07-21:
Hello @TSC21. I hope all you are doing well.
I'm not the writer of this issue, but I have a similar problem. So I leave my issue here.

I'm using CUAV V5 nano (PX4 v1.12.0 & running micrortps_client) and Raspberry Pi 4 (Ubuntu 20.04 & running ROS2 foxy, micrortps_agent). And the micrortps_bridge is connected via UART.

I made custom ROS2 launch file for the communication test, and I checked the companion computer can receive the topics (e.g vehicle status, vehicle local position, etc.). But the PX4 has problem to receive the topics from companion computer. After launching the test file, it works intermittently.
My custom launch file send VehicleCommand topic for flight mode change repeatedly, and I can check the topic message in the terminal on companion computer by using topic echo command. The topic transmission frequency is about 100hz. But sometimes it works and sometimes it doesn't.

I tried to figure out the reason, but I couldn't.
Is there any solution I can check? Thanks in advance :)

TSC21, 2021-07-21:
@ow18600 please check my last comment. Thank you

ow18600, 2021-07-21:
Thank you for the response @TSC21.

The commands I using are "micrortps_client start -d /dev/ttyS2 -b 921600" on PX4 and "micrortps_agent -d /dev/serial0 -b 921600" on companion computer.
And I set PX4 parameters related with communication. The parameters are below,
 - MAV_1_CONFIG = TELEM 2
 - MAV_1_FORWARD = 1
 - MAV_1_MODE = Onboard
 - MAV_1_RADIO_CTL = 1
 - MAV_1_RATE = 0 B/s
 - SER_TEL2_BAUD = 921600 8N1 <-- Baud rate setting

TSC21, 2021-07-21:
> Thank you for the response @TSC21.
> 
> The commands I using are "micrortps_client start -d /dev/ttyS2 -b 921600" on PX4 and "micrortps_agent -d /dev/serial0 -b 921600" on companion computer.
> And I set PX4 parameters related with communication. The parameters are as below,
> 
> * MAV_1_CONFIG = TELEM 2
> * MAV_1_FORWARD = 1
> * MAV_1_MODE = Onboard
> * MAV_1_RADIO_CTL = 1
> * MAV_1_RATE = 0 B/s
> * SER_TEL2_BAUD = 921600 8N1 <-- Baud rate setting
> 
> And the vehicle command topic's parameters that my test launch file publishes are as below,
> 
> * command =176 <-- flight mode change command
> * param1 = 1
> * param2 = 2 <-- for changing to Altitude mode
> * target_system = 1
> * target_component = 1
> * confirmation = 1
> * from_external = true

ok. Question: why are you sending the command multiple times? that might introduce some unwanted behaviors. Also in the nsh shell, can you see the command is sent on the output of `listener vehicle_command`? I would also make sure that you completely trust the cabling of the connection to the RPi, and if possible, use SW or HW flow control (which can also be set on the client and in the agent). Try to also increase the baud rate for something like 1500000.

ow18600, 2021-07-21:
Thanks for your help @TSC21 :)

I wanted to check the PX4-ROS2 bridge works well, so I published the vehicle command topic repeatedly.
Referring to your advice, I published the topic once every 10 seconds. But the vehicle command didn't work.
So I published the topic repeatedly again. Then I enabled SW flow control, and changed baud rate to 1500000 by changing the micrortps command & PX4 parameter.
But PX4 still received the topic intermittently.
After vehicle command worked, I checked the output of "listener vehicle_command" in the nsh shell.
And the outputs are as below,
----------------------------------------------------------
listener vehicle_command

TOPIC: vehicle_command
vehicle_command_s
timestamp: 65318138 (12.727702 seconds ago)
param5: 0.000000
param6: 0.000000
param1: 1.0000
param2: 2.0000
param3: 0.0000
param4: 0.0000
param7: 0.0000
command: 176
target_system: 1
target_component: 1
source_system: 0
source_component: 250
confirmation: 1
from_external: True
nsh>
----------------------------------------------------------
The outputs were same with the topic parameter that I published in companion computer.

I re-checked the cabling condition and I couldn't find any problem in there.
And the vehicle command sometimes works. So I think there is no problem in cabling.

I tested my launch file in gazebo simulation with "px4_sitl_rtps" build. And there was no problem.
Vehicle command worked very well with no delay.
The gazebo simulation was done by desktop and used UDP communication.
I checked the topic transmission rate. And it was about 3000hz.
This is 30 times the companion computer's transmission rate (about 100hz). Can it make problem?

TSC21, 2021-07-21:
> I checked the topic transmission rate. And it was about 3000hz.

which topic? What publishes that? That's actually incorrect and no topic should publish at that rate.

ow18600, 2021-07-21:
> which topic? What publishes that? That's actually incorrect and no topic should publish at that rate.

That was vehicle command topic (VehicleCommand_PubSubTopic) from agent to client. The 3000hz rate was checked using "ros2 topic hz" command in terminal.

TSC21, 2021-07-21:
> The 3000hz rate was checked using "ros2 topic hz" command in terminal.

Then that's a problem on the node you have that's publishing that command.

TSC21, 2021-07-21:
@GuillermoHerreraF do you still have issues on your side? Otherwise we can close this topic. @ow18600 this discussion has progressed far beyond what was the matter of this topic, so let's please continue it on the PX4 forum or in another issue. Thank you

ow18600, 2021-07-21:
Okay. I will re-check my code. Thank you for the advice :)

GuillermoHerreraF, 2021-07-21:
Hello @TSC21. I haven't had the time to continue working on it. I will try this weekend and get back to you.

TSC21, 2021-08-20:
Closing this as stale. Reopen if not solved.

closed 2021-08-20 by TSC21