PX4/px4_ros_com #95: Question: running MicroRTPS_Bridge alongwith MAVROS

karishmapatnaik, 2021-06-30:
hello @TSC21,
I am trying to run the ROS2-PX4 bridge along with mavros for ROS2. When I start the agent and then start the mavros_node, I sometimes see a time_synchronization issue. Specifically, the console shows: 
`Time jump detected. Resetting time synchronizer . `
Is this a warning or something of concern? I would appreciate your help understanding this.

TSC21, 2021-07-01:
What's the rationale for running one along side the other?

karishmapatnaik, 2021-07-01:
I have the entire code based setup successfully in mavros for multiple drones but I wanted to explore publishing directly to the controller and bypass the estimator altogether with my custom one. To keep the entire pipeline intact while getting this benefit is something that I am exploring with the rtps bridge. Would you suggest against employing both together?

TSC21, 2021-07-01:
> I have the entire code based setup successfully in mavros for multiple drones but I wanted to explore publishing directly to the controller and bypass the estimator altogether with my custom one. To [...]

I don't see any benefit on using them together. If you want to keep the previous code base, I would suggest to just use what MAVROS provides only. If you want to migrate to ROS2, you should migrate the code base to use px4_ros_com + px4_msgs through ROS2 instead. One major problem I might see happen is that on the uORB side you might have both px4_ros_com (natively) and MAVROS (through MAVLink) interfacing with the same topic simultaneously, which is obviously a problem. My advice is to migrate to this bridge, as this is anyway the correct and recommended way to interface a companion computer with PX4 through ROS.
Regarding the timesync warning you see: that's something related with MAVROS, and not with px4_ros_com.

karishmapatnaik, 2021-07-01:
That is insightful, thank you for such a prompt response. So as long as any uORB topics do not interfere, this setup might work.  I should probably think over this approach then. Thank you indeed!

closed 2021-07-01 by karishmapatnaik