PX4/px4_ros_com #96: Build failed: listening to actuator_controls uORB topic over the RTPS bridge

karishmapatnaik, 2021-07-02:
Hello @TSC21 
I am trying to listen to the uORB actuator_controls topic via the rtps bridge. I have added the msg id for it in the `msg/tools/uorb_rtps_message_ids.yaml` file. Specifically, I added
```
    - msg: actuator_controls
      id: 1
      send: true
```
and added the corresponding ids as well from actuator_controls_0 to actuator_controls_3. 
However, whenever I build the firmware, I run into the following issue. 
```
fatal error: uORB_microcdr/topics/actuator_controls.h: No such file or directory
   81 | #include <uORB_microcdr/topics/actuator_controls.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```
I am assuming this is the controller output for thrust and torques and I want to loopback this information to the companion computer, any help with this is appreciated, thank you.

TSC21, 2021-07-03:
Two things:
1. Did you build the _rtps target before adding the msg to the yaml file? If yes, then you need to `make clean` and the build again.
2. It's not enough to modify the yaml file on the PX4 side, you also need to update the one here in this repo (under the templates dir). Make also sure that if you built it before adding it to the yaml file, that you first delete the generated micrortps_agent under the px4_ros_com/src dir and then cleanup your build and install folders in your colcon workspace.

TSC21, 2021-07-03:
Also I am not entirely sure what you mean with:

> and added the corresponding ids as well from actuator_controls_0 to actuator_controls_3.

Does that mean you also added the `send: true` to these?

karishmapatnaik, 2021-07-03:
I hear you. I had cleaned the ros2_ws and even did a `make clean` before building the rtps target. But I was missing that I was always skipping the colcon build for the firmware using `--packages-skip`. Since many a time I run rosdep and rosdep throws an error for px4 package, I ran into the habit of skipping building this package after the initial build. Thank you for being so patient with me, I am new to all this. In short, the following is the process I followed to be able to subscribe to the topic:

- Edit the yaml file in `px4/msg/tools` and the yaml file in `px4_ros_com/templates` to set the respective `send or receive` values
- Clean the `ros2_ws`, delete the `micrortps_agent` folder under `px4_ros_com/src` , and build the `ros2_ws`
- Build the px4 firmware 

To your second question, yes I added the `send: true` to all the actuator_controls_0 to actuator_controls_3 topics. `actuator_output` looks like the mixer out and yes correct, I should be able to reconstruct the control thrust and torques. I will explore that possibility too. But anyways this works, so thank you!

closed 2021-07-03 by karishmapatnaik