PX4/px4_ros_com #98: An error occurred when connecting to PX4 using ROS2

Boshen97, 2021-07-19:
Ubuntu: 20.04
ROS2: galactic

I am trying to use ROS2 and PX4, refer to （http://docs.px4.io/master/en/ros/ros2_comm.html） tutorial,At present, it is the fourth step. In the first three steps, some errors occurred during compilation, which may be due to the ROS version. I modified the source code of px4_ros_com.

In the Sanity Check the Installation part, there are problems in the second and third steps,
When I execute the in the third step, the following error appears
$ micrortps_agent -t UDP 

Error message： 
[   micrortps_agent   ]	Unexpected topic ID '87' to publish Please make sure the agent is capable of parsing the message associated to the topic ID '87'
[   micrortps_agent   ]	Unexpected topic ID '96' to publish Please make sure the agent is capable of parsing the message associated to the topic ID '96'
[   micrortps_agent   ]	Unexpected topic ID '112' to publish Please make sure the agent is capable of parsing the message associated to the topic ID '112'

in the fourth step:
$ micrortps_client start -t UDP

Error message：
Command 'micrortps_client' failed, returned -1.

Please tell me how to solve this problem.

Thank you in advance!

TSC21, 2021-07-26:
Are you using master of all repos? (PX4-Autopilot - you can rebase your code in it -, px4_ros_com and px4_msgs)?

TSC21, 2021-08-20:
@Boshen97 any updates?

closed 2021-09-02 by TSC21