PX4/px4_ros_com #99: What is the procedure on marking uORB messages to be PubSub'd by px4_ros_com?

iwishiwasaneagle, 2021-07-19:
https://docs.px4.io/master/en/middleware/micrortps.html#supported-uorb-messages implies that it's a matter of editing `templates/uorb_rtps_message_ids.yaml`, however I have had no success with this. The messages currently marked as `send` and/or `receive` (`VehicleAttitude` for example) in the source code work perfectly fine. Adding new ones to be published (for example `Rpm`) does not yield any data.

```bash
/src/px4_ros_com_ros2/src# ros2 topic hz /Rpm_PubSubTopic
WARNING: topic [/Rpm_PubSubTopic] does not appear to be published yet
/src/px4_ros_com_ros2/src# ros2 topic hz /VehicleAttitude_PubSubTopic
average rate: 212.172
        min: 0.004s max: 0.008s std dev: 0.00152s window: 213
```

Adding some documentation to clarify this would be massively useful. The current trial-and-error approach to figuring this out requires 5+ minutes due to `px4_msgs` needing to be built each time.

TSC21, 2021-07-19:
You have to update the files on both sides of the bridge:
1. Create the uORB file and add it to the PX4 build and build the code
2. Generate the ROS2 message from it (we have a [script in PX4](https://github.com/PX4/PX4-Autopilot/blob/master/msg/tools/uorb_to_ros_msgs.py) that does that) and copy the message to px4_msgs/msg
3. Modify the yaml file on both sides ([PX4](https://github.com/PX4/PX4-Autopilot/blob/master/msg/tools/uorb_rtps_message_ids.yaml) and [px4_ros_com](https://github.com/PX4/px4_ros_com/blob/master/templates/uorb_rtps_message_ids.yaml)). Make sure that you add a matching ID and set send or receive accordingly (needs to be the same on both files)
4. Build the colcon ws where the _px4_ros_com_ and _px4_msgs_ are
5. Then you can write a node to interface with the newly create topic for that message

> Adding some documentation to clarify this would be massively useful. The current trial-and-error approach to figuring this out requires 5+ minutes due to px4_msgs needing to be built each time.

It is work in progress.

TSC21, 2021-07-21:
@iwishiwasaneagle is this solved to you?

iwishiwasaneagle, 2021-07-21:
Hi, I'm away this week and will test it when I return home.

iwishiwasaneagle, 2021-07-28:
@TSC21 That seems to be working. That part of the docs is very hidden honestly [ros/ros2_offboard_control.html#requirements](https://docs.px4.io/master/en/ros/ros2_offboard_control.html#requirements) . Would it be worth putting a note into the `ROS2 User Guide` section under [ros/ros2_comm.html#build-ros-2-workspace](https://docs.px4.io/master/en/ros/ros2_comm.html#build-ros-2-workspace) ?

TSC21, 2021-07-28:
> @TSC21 That seems to be working. That part of the docs is very hidden honestly [...]

I am not following your suggestion. What you are pointing at is an offboard control example to use the ROS2 bridge.

iwishiwasaneagle, 2021-07-28:
> I am not following your suggestion. What you are pointing at is an offboard control example to use the ROS2 bridge.

Well the only mention of the configuration steps needed to add custom messages to the ROS2 node from your link https://docs.px4.io/master/en/ros/ros2.html is found under https://docs.px4.io/master/en/ros/ros2_offboard_control.html#requirements which, in my opinion, is very hidden. I'd assume people wanting to use ROS2 with PX4  would be interested in using the messages they're interested in. If they weren't interested in offboard control then they might never go onto those part of the docs and therefore miss the documentation about how to add more uORB messages to the node. Therefore I think it might be beneficial to add a note like 
[image]
with text like "To add more messages blah blah" along with a link to the relevant docs. 

I personally spent quite some time searching for this info and it never crossed my mind to search the offboard control docs. That's why I started this issue.

TSC21, 2021-07-28:
@iwishiwasaneagle did you read this comment: https://github.com/PX4/px4_ros_com/issues/99#issuecomment-882701326?

iwishiwasaneagle, 2021-07-28:
Apologies @TSC21, seems I skipped your final sentence in it.

closed 2021-07-28 by iwishiwasaneagle