PX4/px4_ros_com #102: Some messages transmission doesn't work

ow18600, 2021-08-02:
Hello,

I'm a beginner to use micrortps_bridge, and I have a problem to use it. I installed microRTPS_bridge related ROS2 packages following the instruction (https://docs.px4.io/master/en/ros/ros2_comm.html), and it worked well in gazebo simulation with UDP.
But when I use it with UART, some messages transmission doesn't work. The examples are as below,

     uasg@uasg-desktop:~$ ros2 topic echo /Timesync_PubSubTopic 
     timestamp: 2666727088
     sys_id: 0
     seq: 121
     tc1: 0
     ts1: 2666727088728
     ---

     uasg@uasg-desktop:~$ ros2 topic echo /VehicleStatus_PubSubTopic 
     timestamp: 96750532
     nav_state: 0
     nav_state_timestamp: 541285
     arming_state: 1
     hil_state:` 0
     failsafe: false
     failsafe_timestamp: 0
     system_type: 10
     system_id: 1
     component_id: 1
     vehicle_type: 3
     is_vtol: false
     is_vtol_tailsitter: false
     vtol_fw_permanent_stab: false
     in_transition_mode: false
     in_transition_to_fw: false
     rc_signal_lost: true
     rc_input_mode: 0
     data_link_lost: false
     data_link_lost_counter: 0
     high_latency_data_link_lost: false
     engine_failure: false
     mission_failure: false
     geofence_violated: false
     failure_detector_status: 0
     onboard_control_sensors_present: 304119852
     onboard_control_sensors_enabled: 304185356
     onboard_control_sensors_health: 2523151
     latest_arming_reason: 0
     latest_disarming_reason: 0
     armed_time: 0
     takeoff_time: 0
     ---

The agent can receive vehicle status topic from the client, but cannot receive timesync topic (sys_id, tc1). And the client cannot receive any data from the agent.

I used the latest version of all the packages (PX4-Autopilot, px4_ros_com and px4_msgs), and my ROS2 version is foxy. And RTPS related parameters in PX4 were set.

 - RTPS_CONFIG: TELEM 2
 - MAV_1_CONFIG TELEM 2
 - MAV_1_MODE: Onboard
 - SER_TEL2_BAUD: 921600 8N1

The commands I used were `micrortps_agent -d /dev/ttyUSB0 -b 921600 -v` for agent and `micrortps_client -d /dev/ttyS2 -b 921600` for client. After micrortps_bridge established, I got warning messages and sometimes "Bad CRC" message appeared.

     uasg@uasg-desktop:~$ micrortps_agent -d /dev/ttyUSB0 -b 921600 -v
     --- MicroRTPS Agent ---
     [   micrortps_agent   ]	Starting link...
     [   micrortps_agent   ]	UART transport: device: /dev/ttyUSB0; baudrate: 921600; sleep: 1us; poll: 1ms; flow_control: No
     [ micrortps_transport ]	UART transport: No flush
     ---   Subscribers   ---
     - DebugArray subscriber started
     - DebugKeyValue subscriber started
     - DebugValue subscriber started
     - DebugVect subscriber started
     - OffboardControlMode subscriber started
     - OpticalFlow subscriber started
     - PositionSetpoint subscriber started
     - PositionSetpointTriplet subscriber started
     - TelemetryStatus subscriber started
     - Timesync subscriber started
     - VehicleCommand subscriber started
     - VehicleLocalPositionSetpoint subscriber started
     - VehicleTrajectoryWaypoint subscriber started
     - OnboardComputerStatus subscriber started
     - TrajectoryBezier subscriber started
     - VehicleTrajectoryBezier subscriber started
     - VehicleMocapOdometry subscriber started
     - VehicleVisualOdometry subscriber started
     - TrajectorySetpoint subscriber started
     -----------------------

     ----   Publishers  ----
     - InputRc publisher started
     - SatelliteInfo publisher started
     - SensorCombined publisher started
     - Timesync publisher started
     - TrajectoryWaypoint publisher started
     - VehicleAttitude publisher started
     - VehicleControlMode publisher started
     - VehicleLocalPosition publisher started
     - VehicleOdometry publisher started
     - VehicleStatus publisher started
     - CollisionConstraints publisher started
     - VehicleAngularVelocity publisher started
     - VehicleTrajectoryWaypointDesired publisher started
     -----------------------
     [ micrortps_transport ]	                                (↓↓ 496)
     [ micrortps_transport ]	                                (↓↓ 44)
     [ micrortps_transport ]	                                (↓↓ 122)
     [ micrortps_transport ]	                                (↓↓ 235)
     [ micrortps_transport ]	                                (↓ 10)
     [ micrortps_transport ]	                                (↓↓ 225)
     [ micrortps_transport ]	                                (↓↓ 48)
     [ micrortps_transport ]	                                (↓↓ 232)
     [ micrortps_transport ]	                                (↓ 36)
     [ micrortps_transport ]	                                (↓ 36)
     [ micrortps_transport ]	                                (↓↓ 141)
     [ micrortps_transport ]	                                (↓↓ 34)
     [ micrortps_transport ]	                                (↓ 36)
     [ micrortps_transport ]	                                (↓ 36)
     [ micrortps_transport ]	                                (↓ 28)
     [ micrortps_transport ]	                                (↓↓ 9)
     [ micrortps_transport ]	Bad CRC 30977 != 55379		(↓ 54)
     [ micrortps_transport ]	                                (↓↓ 85)
     [ micrortps_transport ]	                                (↓ 36)
     [ micrortps_transport ]	                                (↓↓ 25)
     [ micrortps_transport ]	                                (↓ 22)
     [ micrortps_transport ]	                                (↓↓ 2)
     [ micrortps_transport ]	                                (↓↓ 11)
     [ micrortps_transport ]	                                (↓ 110)
     [ micrortps_transport ]	                                (↓ 280)

Does anyone know if I did something wrong? or solution to solve this problem?
Thanks in advance :)

TSC21, 2021-08-02:
You seem to have a bad link established. Try to check the wiring or replace the cabling. What seems obvious is that the TX line from the agent to the client is down (you don't get any data from the agent on the client). And since that you don't get timesync msgs on the client, it never answers to the agent, so that's why you don't get timesync msgs on the agent.
I would also try to increase the baud rate to something equal or higher than 1000000 (1Mbaud), and if that still doesn't do it, try to remove some of the topics you don't need from the yaml files and rebuild both PX4 and px4_ros_com.

ow18600, 2021-08-02:
Thanks for the quick response.

> You seem to have a bad link established. Try to check the wiring or replace the cabling. What seems obvious is that the TX line from the agent to the client is down (you don't get any data from the agent on the client).

I checked my wiring, and I couldn't find any problem there. And when I increased published rate of topic on the agent, the client received the msgs intermittently. I used Raspberry Pi 4 for companion board, and I tried 2 ways GPIO UART port and USB port. The same problem occurred in both methods.

> And since that you don't get timesync msgs on the client, it never answers to the agent, so that's why you don't get timesync msgs on the agent.

I didn't block timesync msgs. In gazebo simulation with UDP, the agent can get the timesync msgs. But when I use UART with the companion board, this problem occurred.

> I would also try to increase the baud rate to something equal or higher than 1000000 (1Mbaud) and use flow control, and if that still doesn't do it, try to remove some of the topics you don't need from the yaml files and rebuild both PX4 and px4_ros_com.

Actually, I saw other issues here for solve this problem and tried it. But the increasing baud-rate, enabling flow control and decreasing topic list couldn't solve my problem.

ow18600, 2021-08-03:
I tried the test following your recommend @TSC21.

I opened cts, rts port on GPIO of Raspberry Pi and tried hw flow control. And I increased baud-rate to 3000000, removed unnecessary topic from yaml files in PX4-Autopilot and px4_ros_com package (only Timesync, TelemetryStatus, VehicleAttitude left).
The command I used were `micrortps_agent -d /dev/serial0 -b 3000000 -h -v` and `micrortps_client start -d /dev/ttyS2 -b 3000000 -h`.

I made launch file for test, and what the test file do is publishing TelemetryStatus topic periodically. If I set the published rate to 1hz, the client didn't receive the topic. But when I increased it to abut 800hz, the client received the topic intermittently. I think the msgs are sometimes matched by chance as published rate increases.
I tested it with another companion computer, NUC. It could increase the published rate up to 8000hz, and the client received more msgs than Raspberry Pi, but not all the msgs.

TSC21, 2021-08-03:
> I tried the test following your recommend @TSC21.
>
> I opened cts, rts port on GPIO of Raspberry Pi and tried hw flow control. And I increased baud-rate to 3000000, removed unnecessary topic from yaml [...]

Have you tried a different cable/converter?

TSC21, 2021-08-03:
One thing to understand is the following: the code that is used to read and write from the serial port is exactly the same, i.e., the transport layer translation is the same. So if you can get data from the FMU properly, but not the other way around, that points to an HW problem (besides the flow control SW setup, which I am bringing a fix soon, but that doesn't affect the transmission at the level you are seeing).

ow18600, 2021-08-03:
> Have you tried a different cable/converter?

I tried the test using the 2 kind of ports GPIO and USB of Raspberry Pi with different set of cable. In case GPIO port, I connected it to TELEM2 directly. And in case USB port, I connected it using USB-to-serial converter. The same problem occurred in both cases.

> One thing to understand is the following: the code that is used to read and write from the serial port is exactly the same in the client and the agent, i.e., the transport layer translation is the same. So if you can get data from the FMU properly, but not the other way around, that points to an HW problem (besides the flow control SW setup, which I am bringing a fix soon, but that doesn't affect the transmission at the level you are seeing).

I think I didn't explain my problem accurately. This problem occurred on the agent side too. Some msgs like VehicleStatus, VehicleAttitude were received well on the agent, but Timesync msg (sys_id and tc1) weren't received in the same test. I left about this in detail in my first comment of this issue.

TSC21, 2021-08-03:
And as I have explained, the Timesync messages are only sent from the client to the agent as a reponse to the Timesync message sent from the agent to the client. If the  Timesync messages don't arrive to the client, then it never sends a response back.

ow18600, 2021-08-04:
> And as I have explained, the Timesync messages are only sent from the client to the agent as a reponse to the Timesync message sent from the agent to the client. If the Timesync messages don't arrive to the client, then it never sends a response back.

Now I understand what you explained. The cause of the problem about Timesync msg is also receiving problem in the client side. I will continue more test to solve my problem, and let you know after the problem solved. Please let me know if you find a solution :)

ow18600, 2021-08-04:
I solved my problem. The cause of the problem was parameter setting in PX4.
I set **MAV_1_CONFIG** to **TELEM 2**. It caused communication error.
I changed the value to **Disabled**, and the microRTPS bridge worked well.
Now I can check the received data on the client side.
It seems that the setting caused the communication to be messed up.

Thanks for all your help @TSC21 :)

TSC21, 2021-08-04:
Great! Thanks for the feedback. Closing as solved

closed 2021-08-04 by TSC21