PX4/px4_ros_com #105: micrortps_agent doesn't seem to publish anything since recent update

vranki, 2021-08-18:
I  had a working PX4/ros2/gazebo setup on Ubuntu 20.04 a couple of days ago so I suppose this is a regression.

Debian packages were updated and things stopped working. I updated all related git repos and rebuilt everything. Still, I cant get any data from gazebo simulated PX4.

* gazebo is running and QGroundControl can be used to control it.
* micrortps_client is running in gazebo.

micrortps_agent output:
```bash
micrortps_agent -t UDP
--- MicroRTPS Agent ---
[   micrortps_agent   ]	Starting link...
[   micrortps_agent   ]	UDP transport: ip address: 127.0.0.1; recv port: 2020; send port: 2019
[ micrortps_transport ]	UDP transport: Trying to connect...
[ micrortps_transport ]	UDP transport: Connected to server!

---   Subscribers   ---
- DebugArray subscriber started
- DebugKeyValue subscriber started
- DebugValue subscriber started
- DebugVect subscriber started
- OffboardControlMode subscriber started
- OpticalFlow subscriber started
- PositionSetpoint subscriber started
- PositionSetpointTriplet subscriber started
- TelemetryStatus subscriber started
- Timesync subscriber started
- VehicleCommand subscriber started
- VehicleLocalPositionSetpoint subscriber started
- VehicleTrajectoryWaypoint subscriber started
- OnboardComputerStatus subscriber started
- TrajectoryBezier subscriber started
- VehicleTrajectoryBezier subscriber started
- TrajectorySetpoint subscriber started
- VehicleMocapOdometry subscriber started
- VehicleVisualOdometry subscriber started
-----------------------

----   Publishers  ----
- Timesync publishers started
- TrajectoryWaypoint publisher started
- VehicleControlMode publisher started
- VehicleOdometry publisher started
- VehicleStatus publisher started
- CollisionConstraints publisher started
- TimesyncStatus publisher started
- VehicleTrajectoryWaypointDesired publisher started
-----------------------
```
ros2 topics:
```bash
 ros2 topic list
/fmu/collision_constraints/out
/fmu/debug_array/in
/fmu/debug_key_value/in
/fmu/debug_value/in
/fmu/debug_vect/in
/fmu/offboard_control_mode/in
/fmu/onboard_computer_status/in
/fmu/optical_flow/in
/fmu/position_setpoint/in
/fmu/position_setpoint_triplet/in
/fmu/telemetry_status/in
/fmu/timesync/in
/fmu/timesync/out
/fmu/trajectory_bezier/in
/fmu/trajectory_setpoint/in
/fmu/trajectory_waypoint/out
/fmu/vehicle_command/in
/fmu/vehicle_control_mode/out
/fmu/vehicle_local_position_setpoint/in
/fmu/vehicle_mocap_odometry/in
/fmu/vehicle_odometry/out
/fmu/vehicle_status/out
/fmu/vehicle_trajectory_bezier/in
/fmu/vehicle_trajectory_waypoint/in
/fmu/vehicle_trajectory_waypoint_desired/out
/fmu/vehicle_visual_odometry/in
/parameter_events
/rosout
/timesync_status
```

Running the example outputs no data:
```bash
$ ros2 launch px4_ros_com sensor_combined_listener.launch.py
[INFO] [launch]: All log files can be found below /home/ville/.ros/log/2021-08-18-13-43-37-334800-zbook-93389
[INFO] [launch]: Default logging verbosity is set to INFO
/home/ville/px4_ros_com_ros2/install/px4_ros_com/share/px4_ros_com/launch/sensor_combined_listener.launch.py:50: UserWarning: The parameter 'node_executable' is deprecated, use 'executable' instead
  launch_ros.actions.Node(
[INFO] [sensor_combined_listener-1]: process started with pid [93396]
[sensor_combined_listener-1] Starting sensor_combined listener node...
```
These used to work a couple of days ago. Did I mess up the update or is there something really broken?

matheus-ps, 2021-08-18:
Hello everyone! 

I have the same setup as @vranki, and I am getting the same output of publishers and subscribers from the micrortps_agent and no data when I run the example (that is supposed to subscribe to the "fmu/sensor_combined/out" topic).

Appreciate any help regarding this issue!

TSC21, 2021-08-18:
It's not working because on the transition, sensor_combined was removed from the yaml list. @vranki @matheus-ps can any of you add it back to the list on a PR? I am currently out and with no access to a laptop. Thanks

TSC21, 2021-08-18:
Please do apply the change on the yaml in PX4-Autopilot, since that's the source file of the one deploy here. Thanks in advance!

llanesc, 2021-08-18:
I was able to get sensor_combined to work yesterday by just adding the message to the yaml, but the offboard example did not work. It would create a new uorb instance with id 1 and the one used by px4 internal was setup to subscribe from instance 0 not 1. I solved this by changing the micrortps_client.cpp.em on the px4 autopilot side so that in the structure RcvTopicsPubs the uorb publications use the class uORB::Publication and not uORB::PublicationMulti. This must be a temporary fix and there has to be something changed or maybe I'm doing something wrong because uORB::PublicationMulti has not been changed in the past month and I recall getting this to work straight out of a clone last week.

TSC21, 2021-08-18:
> I was able to get sensor_combined to work yesterday by just adding the message to the yaml, but the offboard example did not work. It would create a new uorb instance (i.e. vehicle_command, trajectory_setpoint, vehicle_control_mode, etc.) with id 1 and the one used by px4 internal was setup to subscribe from instance 0 not 1. I solved this by changing the micrortps_client.cpp.em on the px4 autopilot side so that in the structure RcvTopicsPubs the uorb publications use the class uORB::Publication and not uORB::PublicationMulti. This must be a temporary fix and there has to be something changed or maybe I'm doing something wrong because uORB::PublicationMulti has not been changed in the past month and I recall getting this to work straight out of a clone last week.

The fix is not reverting to single topic prublication, but rather multi subscription to the controllers. Of you have the time to that I would appreciate it since I am currently on holiday. Thank you!

TSC21, 2021-08-18:
Let's keep track of the offboard control issue in another topic though. Please reopen another thread. Closimg this as solved.

closed 2021-08-18 by TSC21