PX4/px4_ros_com #106: Offboard control example issue

llanesc, 2021-08-19:
This issue was first mentioned in a prior issue: https://github.com/PX4/px4_ros_com/issues/105#issuecomment-901454672.

The problem is that when you start the offboard_control example the vehicle does not respond to any arming or control mode changes. I noticed when running "uorb top -a" there were two instances of vehicle_command and trajectory_setpoint after running the offboard_control example on ros2. I fixed this by changing the micrortps_client.cpp.em on the px4 autopilot side so that in the structure RcvTopicsPubs the uorb publications use the class uORB::Publication and not uORB::PublicationMulti. This must be a temporary fix and a proposed solution was presented here: https://github.com/PX4/px4_ros_com/issues/105#issuecomment-901490119.

vranki, 2021-08-19:
Looks like the sensor_combined fix was added in 6423ea50e62ded71afd96f8d98d8152356833dba

I also changed PublicationMulti -> Publication. 

Now micrortps_agent is flooding these:
```bash
[   micrortps_agent   ]	Unexpected topic ID '27' to publish. Please make sure the agent is capable of parsing the message associated to the topic ID '27'
```
Any ideas? Would it be possible to use some tagged stable version instead of master? At least the user guides instruct to use master but this is way too unstable.

vranki, 2021-08-19:
Disregard that - I rebuilt everything and now the error flood stopped. I got the topics published by agent, but 
they don't seem to publish any data:

```bash
ros2 topic echo /VehicleAttitude_PubSubTopic
```
outputs nothing. micrortps_client is running in gazebo.

TSC21, 2021-08-19:
> Disregard that - I rebuilt everything and now the error flood stopped. I got the topics published by agent, but
> they don't seem to publish any data:
> 
> ```shell
> ros2 topic echo /VehicleAttitude_PubSubTopic
> ```
> 
> outputs nothing. micrortps_client is running in gazebo.

`vehicle_attitude` is not published by default on the bridge. You need to add it yourself to the yaml files. Also this is unrelated to this specific issue.

TSC21, 2021-08-19:
> Looks like the sensor_combined fix was added in 6423ea50e62ded71afd96f8d98d8152356833dba
> 
> I also changed PublicationMulti -> Publication.
> 
> Now micrortps_agent is flooding these:
> 
> ```shell
> [   micrortps_agent   ]	Unexpected topic ID '27' to publish. Please make sure the agent is capable of parsing the message associated to the topic ID '27'
> ```
> 
> Any ideas? Would it be possible to use some tagged stable version instead of master? At least the user guides instruct to use master but this is way too unstable.

We make a disclosure on the documentation that we don't have specific topic definitions per stable releases yet.

llanesc, 2021-08-19:
@TSC21 So i've been studying the uORB Pub/Sub API and I can digest a bit of what's going on. From what I gathered PublicationMulti::publish() tries to advertise to instance 0 but it already exists so the uORB::DeviceMaster::advertise() changes the instance requested to 1. I thought about how you said to make the subscribers multi but there are quite a lot of subscribers for vehicle_command and I don't see a reason for having multiple instances for this topic along with others like trajectory_setpoint. I also don't see how making subscribers multi would make the PublicationMulti only advertise to instance 0.

TSC21, 2021-08-20:
> @TSC21 So i've been studying the uORB Pub/Sub API and I can digest a bit of what's going on. From what I gathered PublicationMulti::publish() tries to advertise to instance 0 but it already exists so the uORB::DeviceMaster::advertise() changes the instance requested to 1. I thought about how you said to make the subscribers multi but there are quite a lot of subscribers for vehicle_command and I don't see a reason for having multiple instances for this topic along with others like trajectory_setpoint. I also don't see how making subscribers multi would make the PublicationMulti not create new instances every initialization which is the main issue here.

Yes for things like vehicle_command it doesn't make sense no. This will have to be changed at the template level so one can select which topics should use multi-publishers instead. Meanwhile, can you please add a PR in PX4-Autopilot to revert this back to a simple Publisher while this feature is not updated? Thanks!

TSC21, 2021-08-20:
@bkueng regarding the above: I thought that if a topic is not a multi-topic instance, the extra instance wouldn't be created in the first place. Looks like we cannot default for multi-publishers, but instead we need to choose which ones should be multi-instanced.

llanesc, 2021-08-20:
> Yes for things like vehicle_command it doesn't make sense no. This will have to be changed at the template level so one can select which topics should use multi-publishers instead. Meanwhile, can [...]

Sounds good and here is the pull request https://github.com/PX4/PX4-Autopilot/pull/18116.

TSC21, 2021-08-20:
@llanesc thanks! @vranki do you mind testing the offboard example? (after pulling latest px4_ros_com, cleaning the workspace and rebuilding). Should work fine now. If yes, please let me know so we can close the issue.

closed 2021-09-02 by TSC21